raspberrypi/firmware #1279: VSync not working properly for non-Xorg context on RPI3B with firmware KMS overlay

psyke83, 2019-10-29:
**Describe the bug**
When running software in a KMS context outside of Xorg on RPI 3B with the FKMS overlay active, vsync does not work correctly, resulting in visible tearing and jerkiness in displayed OpenGL(ES) content.

**To reproduce**
A good testcase can be reproduced purely with SDL2 and its included `testgl2` test.

It's necessary to build SDL2 with the `kmsdrm` driver enabled and `rpi` driver disabled (Raspbian's SDL2 packages only has `rpi` enabled). Here are the steps to reproduce on latest upstream SDL2 (with the build modified to install to a temporary directory so as not to interfere with the system packages):
```
# SDL2 tests will not build correctly with libunwind-dev installed on latest master, so for brevity's sake, just uninstall the package for the moment
sudo apt-get build-dep libsdl2-dev && sudo apt-get remove libunwind-dev
git clone https://github.com/spurious/SDL-mirror
cd SDL-mirror
./configure --disable-video-rpi --enable-video-kmsdrm --prefix=$(readlink -m tempinstall) && make -j4 install
cd test
./configure --with-sdl-prefix=$(readlink -m ../tempinstall) && make testgl2
```

To demonstrate the issue, you should test the resulting `testgl2` binary twice, first while running in an Xorg session (so that it will use the SDL2 x11 video driver), and then after exiting Xorg and dropping to a VT (so it will use the SDL2 `kmsdrm` video driver):

Xorg:
```
DISPLAY=:0 ./testgl2 --fullscreen-desktop --vsync --fsaa 4 --video x11
```
Bare KMS:
```
./testgl2 --fullscreen-desktop --vsync --fsaa 4 --video kmsdrm
```

**Expected behaviour**
The cube should spin at 60fps (or lower, due to the FSAA4x overhead) without any visible tearing.

**Actual behaviour**
The cube spins smoothly while running in an Xorg session (at ~30fps), whereas there is very noticeable tearing and jerkiness outside of Xorg using SDL2's `kmsdrm` driver (but runs faster, at ~40-50fps).

**System**
* Pi3B+ (revision a02082).
* Raspbian buster with latest updates installed
* Firmware: `version 6820edeee4ef3891b95fc01cf02a7abd7ca52f17 (clean) (release) (start)`
* Kernel: `Linux rpi3 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux`

**Logs**
dmesg shows no output during testing, so this is probably unnecessary. Furthermore, I see nothing suspicious in the `testgl2` output.

**Additional context**
Several items to note regarding the issue:
1. The issue is more noticeable when the GPU (or CPU?) is more heavily loaded. Running the testcase without `--fsaa 4` will result in a smooth 60fps with no tearing observed for either testing context. This same effect can be seen in games or emulators - making sure that vsync is enabled - where more intensive 3D scenes (that would run at sub-60fps framerate) causes the same stutter and tearing.
3. If we switch from the FKMS to KMS overlay, the tearing problem goes away - but the KMS overlay is not a viable solution, as I need to use firmware features such as hardware decoding. Nevertheless, it may be a useful observation and indicates that this is *not* a bug with SDL2's `kmsdrm` driver.
4. A workaround for the issue is to set the environment variable `VC4_DEBUG=always_sync`. This will result in both test cases running at ~30fps with no tearing/jerkiness.
5. My RPI 4B appears to be immune to this problem (at least with this exact testcase), but what's notable is that for this board, both test cases will run at ~30fps, which gives me the impression that the V3D driver seems to run similarly to the VC4 driver when the `always_sync` debug option is activated.
6. The tearing/jerkiness can also be observed on RPI3B with several cores of the retroarch emulator running in its own KMS context driver (that *doesn't* use SDL2). This is not convenient for presenting a testcase due to the need to use copyright ROMs etc., but it's useful to keep in mind, as it's another indication that this problem isn't a bug in SDL2's internal `kmsdrm` driver.

popcornmix, 2019-10-29:
Can reproduce the tearing/flickering issue described.

I'm aware of a gui tearing issue with kodi on Pi4 with fkms (that doesn't occur with kms).
It depends on resolution and complexity of current menu, so may be the same root cause
(even though you couldn't reproduce on Pi4).

This may be a simpler test case.
@6by9 had considered making the page flipping logic for fkms match kms more closely (but currently this is just a guess that it may solve the issue).

psyke83, 2019-10-29:
FWIW, I've tested the Kodi packages distributed via the pipplware repository, which are primarily targeted for Pi4B (built against your `leia_pi4` branch, I think), but they do happen to work on 3B as well. I do see tearing on both board revisions, but I'm hesitant to point that out as a valid testcase due to the fact that I haven't looked at the code or tried compiling it myself. 

While I can't speak for Kodi, I have tested SDL2 programs' behaviour against my laptop's i965 driver in a KMS context to rule out these tearing issues as being a generic bug in SDL2's kmsdrm driver's page flip handing, and it does seem isolated to the RPI3B's fkms overlay. What's strange is that vsync *does* work properly via Xorg, but I've yet to determine what it does differently regarding page flipping etc. to account for the difference with SDL2's driver, retroarch etc.