raspberrypi/firmware #1281: I2C1 clock frequency is wrong

AlanStern, 2019-11-01:
The clock frequency for I2C1 (that is, the I2C controller for physical GPIO pins 3 and 5) is wrong; it is too low by a factor of 5/8.

The default setting is supposed to be 100 KHz but the actual clock frequency, measured on an oscilloscope, is 62.5 KHz.  Since I wanted to use Fast Mode I2C at 400 KHz, I added the following line to config.txt:

dtparam=i2c_arm=on,i2c_arm_baudrate=400000

but then the measured frequency was 250 KHz,  Changing the baudrate value in the config line to 640000 causes the I2C clock to run at the desired speed.

Tests were made using the 2019-07-10-raspbian-buster system image.

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 3 Model B V1.2

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage4

* Which firmware version (`vcgencmd version`)?
Jul  9 2019 14:40:53 
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux honeypitest 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux

pelwell, 2019-11-05:
This is caused by the fact that the source clock that drives the I2C interfaces is the same as the VPU "core" clock. The core clock normally switches between 250MHz at idle and 400MHz under load. The I2C clock divider has to be calculated for the maximum clock speed so the bus clock never exceeds the requested value, hence the 62.5% ratio.

If you require a constant and accurate I2C clock, set core_freq and core_freq_min to the same value. I suggest you add the following to config.txt:
```
core_freq=250
core_freq_min=250
```

pelwell, 2020-01-10:
See https://github.com/raspberrypi/firmware/issues/1308.