raspberrypi/firmware #1282: USB webcam Raspberry Pi 4

SalvatoreAD, 2019-11-07:
I have a problem with the USB webcam. I installed cheese. as soon as the application starts, the webcam appears jerky and it blocks mouse and keyboard. 

I have installed the latest firmware available via rpi-update.

The problem also presents itself by going to a site via chromium to test the webcam.

Can you help me?

Thanks.

Raspberry Pi 4 - 4gb ram
Raspian Buster
Usb HD120p webcam (trust)

timg236, 2019-11-07:
USB webcams aren't part of the firmware. Try posting on the Raspberry Pi General forums to see if other users have encountered similar issues

 https://www.raspberrypi.org/forums/viewforum.php?f=63&sid=2f85f3a3ecaf1299cd940524a61679e2

SalvatoreAD, 2019-11-08:
it could be a firmware problem because a friend of mine has the same problem.  as soon as the operating system was installed it worked, after updating the firmware it no longer works

P33M, 2019-11-08:
How did you "update the firmware"?

SalvatoreAD, 2019-11-08:
yes both commands

Engineer2Designer, 2019-11-28:
is YUYv422 bug. on V4l2 
you can try  qv4l
sudo apt install qv4l2

gam3t3chelectronicshobbyhouse, 2019-12-01:
Receiving the same issue was able to use the camera prior to my firmware update but now doing the same thing.  Attempted the suggestion above but still not resolved.   Will play around more with it but if anyone has resolved this that wants to drop a few cents in would be great.

gam3t3chelectronicshobbyhouse, 2019-12-01:
today I tried multiple cameras ran it through opencv and web browser (web streaming worked but not ideal) VLC and a few other programs for fun some lag on opencv  but not the worst.   Web browser was fine running a webcam server so with multiple cameras and brands being used and software / options done.   Wiped the pi and started fresh a few times testing before upgrades and after still no go.

ilyabelkin, 2019-12-01:
This may be related to a USB controller firmware issue; please see: https://www.raspberrypi.org/forums/viewtopic.php?p=1490914

Try running the setpci command and if it works, keep the new firmware and add to rc.local. The board temperature is lower and USB devices like Webcams are happier.
```
#install PCI utilities
sudo apt-get install pciutils
#run on every boot
sudo nano /etc/rc.local
#add the command before exit 0
sudo setpci -s 01:00.0 0xD4.B=0x41
#reboot
sudo reboot
```

gam3t3chelectronicshobbyhouse, 2019-12-01:
> This may be related to a USB controller firmware issue; please see: https://www.raspberrypi.org/forums/viewtopic.php?p=1490914
>
> Try running the setpci command and if it works, keep the new firmware [...]

Funny story I had tried that....So Just now I tried it again.....seems I forgot to to put it before the exit 0 which I thought I had and low and behold this time it all settled down.   All is good now thank you  [ilyabelkin ](https://github.com/ilyabelkin)