raspberrypi/firmware #1283: Can't enable the serial5 and serial enumeration is weird

patrickelectric, 2019-11-08:
**Describe the bug**
Can't enable the serial5 with Raspberry Pi 4 kernel 4.19.81

**To reproduce**
Enable uarts config.txt
```
dtoverlay=uart0
dtoverlay=uart1
dtoverlay=uart2
dtoverlay=uart3
dtoverlay=uart4
dtoverlay=uart5
```
Note: Check more **Additional context** for more information.

**Expected behaviour**
Show all ttyAMA0-5 under /dev 

**Actual behaviour**
It'll only show ttyAMA0-4

**System**

* Which model of Raspberry Pi? Raspberry Pi 4
* Which OS and version (`cat /etc/rpi-issue`)? Alarm Arch
* Which firmware version (`vcgencmd version`)? Nops
* Which kernel version (`uname -a`)? Linux alarmpi 4.19.81-1-ARCH #1 SMP PREEMPT Sun Nov 3 16:09:27 UTC 2019 armv7l GNU/Linux

**Logs**
https://pastebin.com/ux4vi0xD

**Additional context**
Is it possible to enable uart5 ? This is really possible or it needs a special configuration ?

The funny thing is that if you disable uart0 and uart1 via dtparam and enable the others:
```
dtparam=uart0=off
dtparam=uart1=off

dtoverlay=uart0
dtoverlay=uart1
dtoverlay=uart2
dtoverlay=uart3
dtoverlay=uart4
dtoverlay=uart5
```
ttyAMA0-4 will appear, but if you disable uart0 or uart1 via dtoverlay the output will be ttyAMA0-3. It appears that there is something wrong.

patrickelectric, 2019-11-11:
@popcornmix any tip ?

pelwell, 2019-11-11:
You won't see `ttyAMA5` on a Pi. Any `ttyAMAn` is a device object for an instance of a PL011 UART. BCM2835, 6 & 7 all have one PL011 (`ttyAMA0`) and one 8250-clone (`ttyS0`). BCM2711 has an additional 4  PL011s. `n` is an index into the set of UARTs of each type, not into the set of all UARTs, which means that the highest number you will see on a Pi 4 is 4.

patrickelectric, 2019-11-11:
Perfect, thank you for the explanation!

closed 2019-11-11 by patrickelectric

pelwell, 2019-11-11:
One thing to beware of when looking at overlays is that they don't actually get applied until the next `dtoverlay=` line is found, or the end of config.txt is reached. The reason for this is that a single `dtoverlay=` line can be split across multiple lines, with any extra parameters passed via subsequent `dtparam=` lines. This can lead to some very subtle bugs, e.g.:
```
dtoverlay=disable-bt # 1. Swap the UARTs back on BT-equipped Pi
dtparam=uart0=off    # 2. Disable UART0
dtoverlay=uart2      # 3. Enable UART2
```
Reading this you might reasonably expect the `uart0=off` to override `disable-bt` (which enables UART0) and leave it disabled, but actually the `disable-bt` overlay doesn't get applied until line 3 is reached, so the effective order is:
```
dtparam=uart0=off    # 1. Disable UART0
dtoverlay=disable-bt # 2. Swap the UARTs back on BT-equipped Pi
dtoverlay=uart2      # 3. Enable UART2
```
This leaves UART0 enabled - the opposite of what wanted.

One could argue that this shouldn't happen, and I'll consider some way of improving the situation, but there is a workaround. In this case you can move the `uart0=off` after `dtoverlay=uart2`, since the uart2 overlay doesn't interfere with UART0, but a more general fix is to add an explicit flush:
```
dtoverlay=disable-bt # 1. Swap the UARTs back on BT-equipped Pi
dtoverlay=           # 2. Apply the previous overlay now!
dtparam=uart0=off    # 3. Disable UART0
dtoverlay=uart2      # 4. Enable UART2
```