raspberrypi/firmware #1284: This repository should contain the kernel headers

Shachar, 2019-11-12:
**Describe the bug**
Installing a new kernel using `rpi-update` does not install the kernel headers, making it impossible to keep out-of-tree modules up to date (via dkms, for example).
 
**To reproduce**
Install raspbian on a Pie 4
upgrade it to latest kernel with rpi-update
Install the source package from https://apt.driveu.dev.liveu.tv/i2c-mcp2221-dkms_0.1-1_all.deb

**Expected behaviour**
The package installs the kernel module

**Actual behaviour**
The compilation fails, because it does not find the header files for the kernel

**System**
```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

Linux connectador 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
Revision        : c03111
Serial          : 1000000043132ee0
Model           : Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Sep 24 2019 17:34:59 
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start_cd)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30186616 1808156  26821984   7% /
devtmpfs         1893160       0   1893160   0% /dev
tmpfs            2025256       0   2025256   0% /dev/shm
tmpfs            2025256    8960   2016296   1% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs            2025256       0   2025256   0% /sys/fs/cgroup
/dev/mmcblk0p6    258094   53033    205061  21% /boot
tmpfs             405048       0    405048   0% /run/user/1000

Filename                                Type            Size    Used    Priority
/var/swap                               file            102396  0       -2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20191105
openbox:
  Installed: (none)
lxpanel:
[... 594 lines cut ...]
[ 2535.477353] i2c-mcp2221 1-1.2.5:1.2: mcp2221_init: Success
[ 2535.477801] i2c-mcp2221 1-1.2.5:1.2: mcp2221_probe() -> chip connected -> Success
[ 2535.586430] i2c-mcp2221(disconnect) -> chip disconnected
[ 2535.586563] i2c-mcp2221 1-1.2.5:1.2: MCP2221 at USB bus 001 address 009 -- mcp2221_init()
[ 2535.586571] i2c-mcp2221 1-1.2.5:1.2: mcp2221_init: Success
[ 2535.586635] i2c-mcp2221 1-1.2.5:1.2: mcp2221_probe() -> chip connected -> Success
[ 2580.096966] usb 1-1.2: USB disconnect, device number 8
[ 2580.096985] usb 1-1.2.5: USB disconnect, device number 9
```

* Raspberry PI 4 with 4GB
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

* Which firmware version (`vcgencmd version`)?
```
Sep 24 2019 17:34:59 
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start_cd)
```

* Which kernel version (`uname -a`)?
`Linux connectador 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux`
(before upgrade)

pelwell, 2019-11-12:
`rpi-update` is not meant for regular use. Use a Raspbian kernel and the `linux-headers` package.

closed 2019-11-12 by pelwell

satmandu, 2020-04-15:
Apologies for hijacking a closed thread, but the question seems relevant to this discussion.

Is there a reproducible kernel tag that corresponds to the rpi-update kernels such that we can compile modules for those kernels such that we can install them from here?

(Troubleshooting a firmware bug for a machine which needs to have zfs-dkms compiled for it, and it would be easiest to try with a rpi-update kernel which corresponds to the firmware.)

mwoolweaver, 2020-04-15:
> Is there a reproducible kernel tag that corresponds to the rpi-update kernels such that we can compile modules for those kernels such that we can install them from here?

I have been wondering this for a long time myself because I can use `rpi-update` once every 6 months which is not what @pelwell describes and I would still need headers to compile a wireguard module via dkms which I can't find via Raspbian APT repositories.

satmandu, 2020-04-16:
Is this process sufficient? https://www.ploxiln.net/rpi_kernel_modules.html