raspberrypi/firmware #1285: PI4: Overscan doesnt work on composite output while dtoverlay=vc4-fkms-v3d is in use

OlehNazar, 2019-11-14:
Hi!
I have faced with problem : I use composite video output. When parameter dtoverlay=vc4-fkms-v3d in config.txt file is active then overscan configuration doesnt take an affect. My screen is cutted and some piece of interface is out of screen. If I comment parameter #dtoverlay=vc4-fkms-v3d overscan configuration will work. "rpi-update" and update&upgrade were already performed.
I have no other possibility to use RPI4 except for composite TV out. Necessary application for me demand dtoverlay=vc4-fkms-v3d overscan so just turn off this parameter is not solution for me. If you need additional info, please contact me.
Thank you in advance!

**Info:**
* PI 4 Model B 4GB
* cat /etc/rpi-issue 
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage5

* uname -a
Linux raspberrypi 4.19.83-v7l+ #1277 SMP Mon Nov 11 16:41:20 GMT 2019 armv7l GNU/Linux

* vcgencmd version
Nov  8 2019 18:10:14 
Copyright (c) 2012 Broadcom
version e3ba08775a44334ecfda5e37fafad3fb07a3dc7e (clean) (release) (start)

* cat /boot/.firmware_revision 
d86586926cfca6aa158cab13888194d53ae23d1b

* tvservice --status
state 0x40000 [NTSC 14:9], 720x480 @ 60.00Hz, interlaced

* 
[config.txt](https://github.com/raspberrypi/firmware/files/3848960/config.txt)
My config.txt in attachments

BR,
Oleh

6by9, 2019-11-15:
FKMS recreates the display properties. Overscan settings can get passed through to the kernel, but I may have only done it for the HDMI displays (they have to be done by name).
I'll have a look later, but if you could post the contents of /proc/cmdline it would give the answer. Overscan would be signalled via an entry starting ```video=Composite-1:720x576,top_margin=X,bottom_margin=X...```.

6by9, 2019-11-15:
OK, the firmware changes to add the relevant entries were trivial.

Annoyingly there's a bug in the kernel that if you insert any of the options after the refresh rate (eg 'i' for interlaced), then it incorrectly assigns all of the rest of the string as options, and then very quickly fails when it sees ','.

I've a quick fix for that, but there were a couple of cmdline parsing patches that went past on dri-devel in the last couple of days that may be a better fix. It may be safer to just push my patch anyway, and drop it again when upstream fix things.

OlehNazar, 2019-11-15:
@6by9 

Thanks for your investigation. 
output:
pi@raspberrypi:~ $ cat /proc/cmdline
coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M  smsc95xx.macaddr=DC:A6:32:37:4E:83 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=5e3da3da-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

If its needed I can perform tests of your patch or provide additional info

ashwinterminator, 2019-11-18:
> OK, the firmware changes to add the relevant entries were trivial.
>
> Annoyingly there's a bug in the kernel that if you insert any of the options after the refresh rate (eg 'i' for interlaced), then [...]

I presume this is not yet on rpi-update?

6by9, 2019-11-18:
> I presume this is not yet on rpi-update?

That would be a touch optimistic seeing as I posted at 18:45 on Friday and you're querying it 09:56 on the Monday after.

You can insert the relevant text for yourself into /boot/cmdline.txt if you're that desperate for it
```video=Composite-1:720x480@60,margin_left=32,margin_right=32,margin_top=32,margin_bottom=32```
Replace the 720x480@60 with 720x576@50 if you're using PAL, and the margin values as you see fit.
It appears that the kernel doesn't check the interlaced flag when comparing the defined modes against the command line mode, so the kernel issue doesn't matter if you're being lazy over your mode definition. (There should be an i after the refresh rate to denote interlaced).

ashwinterminator, 2019-11-18:
> 
> 
> > I presume this is not yet on rpi-update?
> 
> That would be a touch optimistic seeing as I posted at 18:45 on Friday and you're querying it 09:56 on the Monday after.

Apologies if it came off that way. I was a little confused because of this: "It may be safer to just push my patch anyway, and drop it again when upstream fix things." So I wasn't sure if you had pushed it yet. Hence the clarification. :)

JamesH65, 2019-11-18:
Just a quick comment on process. Although we push stuff to our internal repo as a PR, it's still takes a day or two, or longer, for change in there to be accepted to master, and then subsequently moved in to the rpi-update system.

ashwinterminator, 2019-11-18:
> 
> 
> Just a quick comment on process. Although we push stuff to our internal repo as a PR, it's still takes a day or two, or longer, for change in there to be accepted to master, and then subsequently moved in to the rpi-update system.

Ah. Thank you, James. That makes the situation a lot more clear :)

peetereczek, 2019-11-27:
I recorded the same issue on RPi 3B+ Hopefuly I'll test latest fix soon

popcornmix, 2019-11-29:
Can you try with latest rpi-update firmware?

ashwinterminator, 2019-11-29:
> Can you try with latest rpi-update firmware?
> 
> edit: I think the fix is only currently on Pi4 - we'll port to Pi3 if firmware if tests are positive

Just updated my Pi4

Can confirm it works! Thank you! :)

OlehNazar, 2019-12-01:
> Can you try with latest rpi-update firmware?
> 
> edit: I think the fix is only currently on Pi4 - we'll port to Pi3 if firmware if tests are positive

confirm, it works on pi4

peetereczek, 2019-12-09:
> Can you try with latest rpi-update firmware?
> 
> edit: I think the fix is only currently on Pi4 - we'll port to Pi3 if firmware if tests are positive

I was facing exactly the same issue with Pi 3B+
On latest rpi-update release all is good :)

alanbork, 2021-05-14:
seems like this has been fixed and can be closed?

closed 2021-05-17 by popcornmix