raspberrypi/firmware #1287: USB boot fails

netwrkx, 2019-11-16:
Hello, I can't get my RP3b+ to boot from usb hdd, whenever I boot with sdcard, my RP3B+ can detect my hdd and access it. I made use of Etcher to make my usb hdd bootable

i also tried the following steps
Format an SD card as FAT32
Copy the current next branch bootcode.bin from GitHub onto the SD card
Connect my hdd to my RPI3B+
Plug it into the Pi and try again

i also add rootdelay=60 to the cmdline.txt

No screen output when trying to boot from hdd (without SDcard) and It doesn't look like any activity takes place without sdcard, just static red led light on. Nothing else

i've tried two different external hard drive

```
pi@raspberrypi:~ $ sudo sfdisk -l /dev/sda | grep Disklabel
Disklabel type: dos
pi@raspberrypi:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 298.1G  0 disk
├─sda1        8:1    0  42.9M  0 part
└─sda2        8:2    0   1.6G  0 part
mmcblk0     179:0    0    30G  0 disk
├─mmcblk0p1 179:1    0  42.9M  0 part /boot
└─mmcblk0p2 179:2    0    30G  0 part /
pi@raspberrypi:~ $ sudo fsck -f /dev/sda1
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
/dev/sda1: 188 files, 44075/86467 clusters
pi@raspberrypi:~ $ sudo fsck -f /dev/sda2
fsck from util-linux 2.29.2
e2fsck 1.43.4 (31-Jan-2017)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
rootfs: 39794/107072 files (0.1% non-contiguous), 239385/428032 blocks
pi@raspberrypi:~ $ lsusb
Bus 001 Device 005: ID 0bc2:2101 Seagate RSS LLC
Bus 001 Device 004: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

how do i fix this problem

JamesH65, 2019-11-16:
You will probably get a better response by posting on the forums.

netwrkx, 2019-11-16:
@JamesH65 

There are unresolved similar cases as mine on the forum with little or no solution. 
Usb boot is meant to work directly without any hack

pelwell, 2019-11-16:
> Usb boot is meant to work directly without any hack

That's a pointless sentence. Everything is supposed to work, nothing is meant to be broken, but this stuff is hard. Most people seem to understand that we are a limited number of developers doing what we can, and that a large support team would mean a higher end-user price and less money to the Foundation.

netwrkx, 2019-11-16:
> That's a pointless sentence. Everything is supposed to work, nothing is meant to be broken, but this stuff is hard. Most people seem to understand that we are a limited number of developers doing [...]

I meant no harm & kudos to you guys.  Great👍work.

ghollingworth, 2019-11-17:
Unfortunately there are some hard drives that won't work with the bootrom.  The reason is usually because the drive doesn't register with the USB system until after it has spun up the disk.  We can only detect drives that spin up quickly and respond to the host within two seconds (or five seconds if you enable the timeout option)

What you can do is put in an SD card with _only_ bootcode.bin on it.  If it then works it shows that the problem isn't your programming of the harddrive, but the bootrom.  Unfortunately that's an unfixable problem...  The bootcode.bin solution would than have to be used.

Gordon

netwrkx, 2019-11-23:
I've finally gotten to boot from hdd.
I added an empty file named "timeout" with the bootcode.bin on my SD card to increase USB wait time. Thanks to Raspberry Forum

closed 2019-11-23 by netwrkx