raspberrypi/firmware #1289: A lot of random errors from brcmfmac

tomty89, 2019-11-21:
**Describe the bug**
Samples:
```
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: Failed access turning clock off: -110
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Nov 21 03:51:23 alarmpi kernel: brcmfmac: brcmf_sdio_rxfail: count never zeroed: last 0xffff
Nov 21 03:51:23 alarmpi kernel: brcmfmac: brcmf_sdio_hostmail: unexpected NAKHANDLED!
Nov 21 03:51:23 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: Failed access turning clock off: -110
Nov 21 03:51:23 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
--- repeat many times ---
```
```
Nov 21 03:56:52 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
Nov 21 03:56:52 alarmpi kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
Nov 21 03:56:52 alarmpi kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Nov 21 03:56:52 alarmpi kernel: brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
Nov 21 03:56:52 alarmpi kernel: brcmfmac: brcmf_sdio_hostmail: Version mismatch, dongle reports 255, expecting 4
Nov 21 03:56:52 alarmpi kernel: brcmfmac: brcmf_sdio_hostmail: Unknown mailbox data content: 0xffffffff
Nov 21 03:56:52 alarmpi kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
Nov 21 03:56:Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: Failed access turning clock off: -110
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Nov 21 03:51:23 alarmpi kernel: brcmfmac: brcmf_sdio_rxfail: count never zeroed: last 0xffff
Nov 21 03:51:23 alarmpi kernel: brcmfmac: brcmf_sdio_hostmail: unexpected NAKHANDLED!
Nov 21 03:51:23 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: Failed access turning clock off: -110
Nov 21 03:51:23 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
52 alarmpi kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Nov 21 03:56:55 alarmpi kernel: brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Nov 21 03:56:55 alaNov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: Failed access turning clock off: -110
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
Nov 21 03:51:21 alarmpi kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Nov 21 03:51:23 alarmpi kernel: brcmfmac: brcmf_sdio_rxfail: count never zeroed: last 0xffff
Nov 21 03:51:23 alarmpi kernel: brcmfmac: brcmf_sdio_hostmail: unexpected NAKHANDLED!
Nov 21 03:51:23 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: Failed access turning clock off: -110
Nov 21 03:51:23 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
rmpi kernel: brcmfmac: brcmf_sdio_isr: failed backplane access
Nov 21 03:56:57 alarmpi kernel: brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Nov 21 03:56:57 alarmpi kernel: brcmfmac: brcmf_set_mpc: fail to set mpc
Nov 21 03:56:57 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
Nov 21 03:57:00 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Nov 21 03:57:02 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Nov 21 03:57:05 alarmpi kernel: brcmfmac: brcmf_configure_wpaie: wme_bss_disable error -110
Nov 21 03:57:05 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
Nov 21 03:57:07 alarmpi kernel: brcmfmac: brcmf_set_mpc: fail to set mpc
Nov 21 03:57:07 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
Nov 21 03:57:10 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Nov 21 03:57:12 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Nov 21 03:57:15 alarmpi kernel: brcmfmac: brcmf_cfg80211_del_station: SCB_DEAUTHENTICATE_FOR_REASON failed -110
Nov 21 03:57:15 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
Nov 21 03:57:18 alarmpi kernel: brcmfmac: brcmf_cfg80211_del_station: SCB_DEAUTHENTICATE_FOR_REASON failed -110
Nov 21 03:57:43 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Nov 21 03:57:45 alarmpi kernel: brcmfmac: brcmf_cfg80211_change_iface: WLC_SET_INFRA error (-110)
Nov 21 03:57:45 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Nov 21 03:57:48 alarmpi kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Nov 21 03:57:48 alarmpi kernel: brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
Nov 21 03:57:48 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Nov 21 03:57:50 alarmpi kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Nov 21 03:57:50 alarmpi kernel: brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
Nov 21 03:57:50 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Nov 21 03:57:53 alarmpi kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Nov 21 03:57:53 alarmpi kernel: brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
Nov 21 03:57:53 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Nov 21 03:57:55 alarmpi kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Nov 21 03:57:55 alarmpi kernel: brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
Nov 21 03:57:55 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Nov 21 03:57:58 alarmpi kernel: brcmfmac: _brcmf_set_multicast_list: Setting mcast_list failed, -110
Nov 21 03:57:58 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Nov 21 03:58:00 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
Nov 21 03:58:03 alarmpi kernel: brcmfmac: _brcmf_set_multicast_list: Setting allmulti failed, -110
Nov 21 03:58:03 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Nov 21 03:58:06 alarmpi kernel: brcmfmac: brcmf_cfg80211_change_iface: WLC_SET_INFRA error (-110)
Nov 21 03:58:06 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -5
Nov 21 03:58:08 alarmpi kernel: brcmfmac: _brcmf_set_multicast_list: Setting BRCMF_C_SET_PROMISC failed, -110
Nov 21 03:58:08 alarmpi kernel: brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
Nov 21 03:58:11 alarmpi kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
```

**To reproduce**
Once I update the bootloader to 6e4a7232ca12ed4cb42326a25e2ceb13c22c06d6, there are errors flooding (in an occasion the wlan interface won't even show up). Downgrading to b9f83b14a8f8ea11e4cc509da037d05836efa00f eliminates them. I am not sure if it has anything to do with the kernel update, but downgrading to 4.19.83 instead doesn't help.

**System**
OS is Arch Linux ARM
Device is Raspberry Pi 4 4GB
`uname -a`:
```
Linux alarmpi 4.19.84-1-ARCH #1 SMP PREEMPT Thu Nov 21 00:52:06 UTC 2019 armv7l GNU/Linux
```

tomty89, 2019-11-21:
The WLAN interface is being used as an AP by the way. Here's the diff of the shipped configuration of hostapd and the one I am using:
```
[root@alarmpi ~]# diff /etc/hostapd/hostapd.conf{.orig,}
88c88
< ssid=test
---
> ssid=redacted
155c155
< hw_mode=g
---
> hw_mode=a
165c165
< channel=1
---
> channel=153
560c560
< #ieee80211n=1
---
> ieee80211n=1
595c595
< #ht_capab=[HT40-][SHORT-GI-20][SHORT-GI-40]
---
> ht_capab=[HT40-][SHORT-GI-20][SHORT-GI-40]
598c598
< #require_ht=1
---
> require_ht=0
1467c1467
< #wpa=2
---
> wpa=2
1476c1476
< #wpa_passphrase=secret passphrase
---
> wpa_passphrase=redacted
1531c1531
< #wpa_pairwise=TKIP CCMP
---
> wpa_pairwise=CCMP
```

pelwell, 2019-11-21:
Can you confirm that, starting with a working system using the kernel and firmware from https://github.com/raspberrypi/firmware/commit/b9f83b14a8f8ea11e4cc509da037d05836efa00f, only updating start4.elf and fixup4.dat from https://github.com/raspberrypi/firmware/commit/6e4a7232ca12ed4cb42326a25e2ceb13c22c06d6 causes the errors you've reported? Have you tried the intermediate commit - https://github.com/raspberrypi/firmware/commit/68ec48189c5256b365c9232909180cb56c8820bc?

tomty89, 2019-11-21:
So I downgraded to 4.19.83 again, have `start4.elf` and `fixup4.dat` from https://github.com/raspberrypi/firmware/commit/68ec48189c5256b365c9232909180cb56c8820bc as the only `.elf` and `.dat` in `/boot`. Similar issue occurs, and upgrading to 4.19.84 doesn't eliminate it, but downgrading `start4.elf` and `fixup4.dat` to those from https://github.com/raspberrypi/firmware/commit/b9f83b14a8f8ea11e4cc509da037d05836efa00f does (they work for both 4.19.83 and 4.19.84).

For the record the errors that shows up with .83 and .84 seem to differ. Also the interface seems to have a higher chance of disappearing with .83 (I don't exactly remember if it is ever gone with .84 / always gone with .83).

pelwell, 2019-11-21:
Thanks - that narrows it down a bit. Unfortunately there are a large number of internal commits in that one revision, and I've not seen the error yet.

1) How long do you typically have to wait after booting to see a problem?
2) Assuming the errors aren't immediate, can you try keeping the system moderately busy with a background thread or two running `while true; do true; done &`?

pelwell, 2019-11-21:
3. If the Pi is busy, try keeping it idle instead.

timg236, 2019-11-21:
It would be useful to know if the errors only occur during throttled so running 
vcgencmd get_throttled and vcgencmd measure_temp and vcgencmd measure_clock emmc

Another thing to try is setting this on config.txt
core_freq=500
core_freq_min=500 

The HDMI display resolution will also alter the voltage and clock requirements so it would be useful to know that. If you are running 4Kp60 try 4kp30

tomty89, 2019-11-21:
Right after boot. And although the errors differ a bit every boot, so far they always show up. I have systemd-networkd brings up the wlan interface on boot (hostapd is started on boot as well). Was going to see if the errors show up if it is not brought up but a miss in file copying stops the Pi from booting now, so I can't test further (maybe until next month).

For the record, the linux-raspberrypi4 package in Arch Linux ARM pulls in [some extra brcmfmac firmware of unknown origin](https://archlinuxarm.org/packages/any/firmware-raspberrypi/files/PKGBUILD). Not sure if it is relevant (was going to delete them and see if it helps, but again I can boot it anymore, so)

P.S. It's a headless server. Neither of the HDMI ports is being used.

pelwell, 2019-11-21:
systemd has been described as "aggressively parallel", starting all the things as concurrently as possible. Current draw is at its maximum during boot, and the WiFi chip has been seen to suffer when the power supply is marginal, so that's something else to bear in mind along with @timg236's suggestions.

tomty89, 2019-11-21:
Well, I will test with another power supply (even though the current one should be 5V/3A), and see if unplugging the usb 3.0 thumb drive helps (it was being used for the root filesystem), when I can.

timg236, 2019-11-22:
I was able to reproduce the issue and resolve it by setting over_voltage=1. If that fails try over_voltage=2

Please could you try just that change in your original config and report the values for 'vcgencmd measure voltage' with and without the config change.

N.B. The voltage varies with clock speed and higher resolutions require higher clock speeds so it's important to be consistent there i.e. if my theory is correct then a higher resolution might appear to make WiFi work as a side effect of increasing clocks / voltage.

timg236, 2019-11-27:
@tomty89 Please could you try the over_voltage=6240 setting to see if that resolves the WiFi issue?

tomty89, 2019-11-28:
While I couldn't test it yet anyway, why `6240`?

From https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md:
```
Values above 6 are only allowed when force_turbo is specified
```
So should I set `force_turbo` as well? And what does this drastically large value do anyway?

timg236, 2019-11-28:
Sorry that should be 6250. It’s possibly board specific but large numbers translate to microvolts and that’s the smallest step that makes sense on your board.
If you are not comfortable with that then testing with over_voltage=1 would also be useful

pelwell, 2019-11-28:
FYI values over 500,000 (used to be 1,000,000) are treated as absolute values in microvolts, e.g. 880000 is 0.88V.

JamesH65, 2019-11-28:
I had this error a couple of days ago, completely filled up the SD card with syslog/kernel errors. If it reoccurs will take a closer look. Looks like the error report, if nothing else, needs to be rate limited.

popcornmix, 2019-11-29:
Can you try with latest rpi-update firmware?

tomty89, 2019-12-16:
601d36d fixed it and no more error with current master either. Thanks!

closed 2019-12-16 by tomty89

timg236, 2019-12-16:
Great, thanks for confirming the fix :)