raspberrypi/firmware #1292: improved debug logging from BOOT_UART

cleverca22, 2019-11-27:
**Describe the bug**
https://gist.github.com/cleverca22/f78553a139241f7f916c65932502426f

when trying to network boot my rpi3, it just fails with a rainbow screen and 7 blinks on the status led
the wiki says that kernel.img is mising, but the logs on the tftp server show no attempt to load a kernel
and the debug over the uart (enabled via `BOOT_UART=1`) claims nothing is wrong

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
the `BOOT_UART=1` debug should clearly say what is wrong

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B v1.2
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)? `bootcode.bin` from `2019-09-26-raspbian-buster-lite.img`
`start.elf` from 2016, unknown version
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

pelwell, 2019-11-27:
Please add "uart_2ndstage=1" to config.txt and post the UART output.

cleverca22, 2019-11-27:
updated to the latest `start.elf` and added that
```
Raspberry Pi Bootcode
Read File: config.txt, 96
Read File: start.elf, 2880356 (bytes)
Read File: fixup.dat, 6744 (bytes)
MESS:00:00:07.609166:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.613264:0: brfs: File read: 96 bytes
MESS:00:00:12.708897:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:12.910292:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:12.917724:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:12.923747:0: *** Restart logging
MESS:00:00:12.947205:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:12.975751:0: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
MESS:00:00:12.980409:0: Loading 'bcm2710-rpi-3-b.dtb' to 0x100 size 0x6743
MESS:00:00:13.130983:0: brfs: File read: 26435 bytes
MESS:00:00:13.137163:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:13.141075:0: brfs: File read: 96 bytes
MESS:00:00:13.148210:0: dterror: not a valid FDT - err -9
MESS:00:00:13.151908:0: Failed to load overlay 'pi3-disable-bt'
MESS:00:00:13.157904:0: brfs: File read: /mfs/sd/overlays/pi3-disable-bt.dtbo
MESS:00:00:13.165530:0: brfs: File read: 954 bytes
MESS:00:00:13.178548:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:13.182568:0: Read command line from file 'cmdline.txt':
MESS:00:00:13.188422:0: 'console=ttyS0,115200 pi3-disable-bt kgdboc=ttyS0,115200 systemConfig=/nix/store/bvzwk35xk7smagklbwpmv3bfih0kcfpf-not-os netroot=192.168.2.1=9080d9b6/root.squashfs quiet splash plymouth.ignore-serial-consoles plymouth.ignore-ude
MESS:00:00:15.378374:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:15.491919:0: brfs: File read: 243 bytes
MESS:00:00:34.444674:0: brfs: File read: /mfs/sd/kernel7.img
MESS:00:00:34.448633:0: Loading 'kernel7.img' to 0x8000 size 0x3711d0
MESS:00:00:34.454800:0: Device tree loaded to 0x2eff9400 (size 0x6bc6)
MESS:00:00:34.462522:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:34.468811:0: uart: Baud rate change done...
```

cleverca22, 2019-11-27:
this time, it does read a kernel (rng in the boot process??) but hangs upon executing it

cleverca22, 2019-11-27:
if i reset the rpi using the run header, it fails to boot and gets into a weird state
the status light is on, and blinks OFF 4 times
rather then the usual blinking ON and idling at off

it also doesnt show any `MESS:` debug output

pelwell, 2019-11-27:
```
MESS:00:00:13.148210:0: dterror: not a valid FDT - err -9
MESS:00:00:13.151908:0: Failed to load overlay 'pi3-disable-bt'
```
Either the overlay is corrupted or the load is failing. Can you verify (md5sum) the overlay? It's probably worth doing the same for the base .dtb and the kernel as well.

P.S. That particular overlay should now be called `disable-bt`, since it also applies to Pi Zero W and Pi 4.

cleverca22, 2019-11-27:
```
[root@router:~]# md5sum /tftproot/9080d9b6/overlays/disable-bt.dtbo /tftproot/9080d9b6/overlays/pi3-disable-bt.dtbo
f0cdebcafcb09d542ef98db3787a38d2  /tftproot/9080d9b6/overlays/disable-bt.dtbo
f0cdebcafcb09d542ef98db3787a38d2  /tftproot/9080d9b6/overlays/pi3-disable-bt.dtbo
```
both overlays are present, and are identical
```
MESS:00:00:14.246261:0: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
MESS:00:00:14.250917:0: Loading 'bcm2710-rpi-3-b.dtb' to 0x100 size 0x6743
MESS:00:00:14.402353:0: brfs: File read: 26435 bytes
MESS:00:00:14.408425:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:14.412335:0: brfs: File read: 92 bytes
MESS:00:00:14.419596:0: dterror: not a valid FDT - err -9
MESS:00:00:14.423315:0: Failed to load overlay 'disable-bt'
MESS:00:00:14.428915:0: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
MESS:00:00:14.436222:0: brfs: File read: 954 bytes
MESS:00:00:14.442310:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:14.446330:0: Read command line from file 'cmdline.txt':
MESS:00:00:14.452184:0: 'console=ttyS0,115200 pi3-disable-bt kgdboc=ttyS0,115200 systemConfig=/nix/store/bvzwk35xk7smagklbwpmv3bfih0kcfpf-not-os netroot=192.168.2.1=9080d9b6/root.squashfs quiet splash plymouth.ignore-serial-consoles plymouth.ignore-ude
MESS:00:00:16.661446:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:16.783203:0: brfs: File read: 243 bytes
MESS:00:00:34.933829:0: brfs: File read: /mfs/sd/kernel7.img
MESS:00:00:34.937789:0: Loading 'kernel7.img' to 0x8000 size 0x3711d0
MESS:00:00:34.943956:0: Device tree loaded to 0x2edf5300 (size 0x6c06)
MESS:00:00:34.951422:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:34.957909:0: uart: Baud rate change done...
```

my `config.txt` is currently:
```
initramfs initrd followkernel
dtoverlay=disable-bt
enable_uart=1
gpu_mem=32
uart_2ndstage=1
```

cleverca22, 2019-11-27:
```
MESS:00:00:27.948340:0: Loading 'kernel7.img' to 0x8000 size 0x3711d0
MESS:00:00:27.954509:0: Device tree loaded to 0x2eff9400 (size 0x6bc6)
MESS:00:00:27.961973:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:27.968458:0: uart: Baud rate change done...
MESS:00:00:27.971892:0: uart: Baud rate
```
it also seems to hang in the middle of this line often

pelwell, 2019-11-27:
That's not a real hang - it's the firmware switching to the other UART.

Please also verify the content of the .dtb file and kernel.

cleverca22, 2019-11-27:
```
[nix-shell:/tftproot]# dtc -I dtb -O dts 9080d9b6/overlays/disable-bt.dtbo
/dts-v1/;

/ {
        compatible = "brcm,bcm2835";

        fragment@0 {
                target = <0xffffffff>;

                __overlay__ {
                        status = "disabled";
                };
        };

        fragment@1 {
                target = <0xffffffff>;

                __overlay__ {
                        pinctrl-names = "default";
                        pinctrl-0 = <0xffffffff>;
                        status = "okay";
                };
        };

        fragment@2 {
                target = <0xffffffff>;

                __overlay__ {
                        brcm,pins;
                        brcm,function;
                        brcm,pull;
                };
        };

        fragment@3 {
                target = <0xffffffff>;

                __overlay__ {
                        brcm,pins;
                        brcm,function;
                        brcm,pull;
                };
        };

        fragment@4 {
                target-path = "/aliases";

                __overlay__ {
                        serial0 = "/soc/serial@7e201000";
                        serial1 = "/soc/serial@7e215040";
                };
        };

        __fixups__ {
                uart1 = "/fragment@0:target:0";
                uart0 = "/fragment@1:target:0";
                uart0_pins = "/fragment@1/__overlay__:pinctrl-0:0\0/fragment@2:target:0";
                bt_pins = "/fragment@3:target:0";
        };
};
[nix-shell:/tftproot]# file 9080d9b6/kernel7.img -L
9080d9b6/kernel7.img: Linux kernel ARM boot executable zImage (little-endian)
```
and i can confirm that the config for the kernel was set to an `armv7l`

cleverca22, 2019-11-27:
is there a config flag to keep the firmware uart on the gpio 14/15 pins? either via the mux or keeping the same uart module

pelwell, 2019-11-27:
No, I mean `md5sum bcm2710-rpi-3-b.dtb kernel7.img` on the TFTP server and in your build, although it seems unlikely that it would be corrupted.

It would help if you took known-good firmware builds (e.g. from our firmware repo) and used them instead.

> is there a config flag to keep the firmware uart on the gpio 14/15 pins? either via the mux or keeping the same uart module

Yes, its `dtoverlay=disable-bt`, but I think that particular bucket has a hole. The sequencing of log messages is a bit odd, so try adding `dtdebug=1` to help narrow down what is causing the error.

cleverca22, 2019-11-27:
```
Raspberry Pi Bootcode
Read File: config.txt, 102
Read File: start.elf, 2880356 (bytes)
Read File: fixup.dat, 6744 (bytes)
MESS:00:00:07.464096:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.468207:0: brfs: File read: 102 bytes
MESS:00:00:07.529732:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.533965:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:07.738250:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:07.745682:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:07.751704:0: *** Restart logging
MESS:00:00:07.755592:0: brfs: File read: 102 bytes
MESS:00:00:07.779344:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:08.912276:0: brfs: File read: /mfs/sd/initrd
MESS:00:00:08.915811:0: Loading 'initrd' to 0x0 size 0x203181
MESS:00:00:09.169311:0: initramfs loaded to 0x2edfc000 (size 0x203181)
MESS:00:00:09.178703:0: Trying Device Tree file 'bcm2710-rpi-3-b.dtb'
MESS:00:00:09.183449:0: brfs: File read: 2109825 bytes
MESS:00:00:09.204563:0: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
MESS:00:00:09.209222:0: Loading 'bcm2710-rpi-3-b.dtb' to 0x100 size 0x6743
MESS:00:00:09.234480:0: dtdebug: delete_node(/__local_fixups__)
MESS:00:00:09.249133:0: dtdebug: /aliases:i2c_vc=i2c0
MESS:00:00:09.262943:0: dtdebug: /__symbols__:i2c_vc=i2c0
MESS:00:00:09.274308:0: dtdebug: /__overrides__:i2c_vc=i2c0
MESS:00:00:09.293342:0: dtdebug: /__overrides__:i2c_vc_baudrate=i2c0_baudrate
MESS:00:00:09.303525:0: dtdebug: /aliases:i2c=i2c1
MESS:00:00:09.317291:0: dtdebug: /__symbols__:i2c=i2c1
MESS:00:00:09.328495:0: dtdebug: /__overrides__:i2c=i2c1
MESS:00:00:09.336890:0: dtdebug: /aliases:i2c_arm=i2c1
MESS:00:00:09.351069:0: dtdebug: /__symbols__:i2c_arm=i2c1
MESS:00:00:09.362706:0: dtdebug: /__overrides__:i2c_arm=i2c1
MESS:00:00:09.381828:0: dtdebug: /__overrides__:i2c_baudrate=i2c1_baudrate
MESS:00:00:09.402390:0: dtdebug: /__overrides__:i2c_arm_baudrate=i2c1_baudrate
MESS:00:00:09.407949:0: dtparam: uart0_clkrate=48000000
MESS:00:00:09.418401:0: dtdebug: /__overrides__ has no uart0_clkrate property
MESS:00:00:09.423834:0: Unknown dtparam 'uart0_clkrate' - ignored
MESS:00:00:09.429659:0: brfs: File read: 26435 bytes
MESS:00:00:09.437236:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:09.441150:0: brfs: File read: 102 bytes
MESS:00:00:09.448354:0: dtdebug: Opened overlay file 'overlays/disable-bt.dtbo'
MESS:00:00:09.454015:0: dterror: not a valid FDT - err -9
MESS:00:00:09.459100:0: Failed to load overlay 'disable-bt'
MESS:00:00:09.464727:0: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
MESS:00:00:09.472016:0: brfs: File read: 954 bytes
MESS:00:00:09.478125:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:09.482140:0: Read command line from file 'cmdline.txt':
MESS:00:00:09.487994:0: 'console=ttyS0,115200 pi3-disable-bt kgdboc=ttyS0,115200 systemConfig=/nix/store/bvzwk35xk7smagklbwpmv3bfih0kcfpf-not-os netroot=192.168.2.1=9080d9b6/root.squashfs quiet splash plymouth.ignore-serial-consoles plymouth.ignore-ude
MESS:00:00:11.684181:0: dtparam: cam0-pwdn-ctrl=22
MESS:00:00:11.692565:0: dtdebug: Found override cam0-pwdn-ctrl
MESS:00:00:11.696711:0: dtparam: cam0-pwdn=5
MESS:00:00:11.705969:0: dtdebug: Found override cam0-pwdn
MESS:00:00:11.721605:0: dtparam: cam0-led-ctrl=22
MESS:00:00:11.729945:0: dtdebug: Found override cam0-led-ctrl
MESS:00:00:11.734005:0: dtparam: cam0-led=6
MESS:00:00:11.743194:0: dtdebug: Found override cam0-led
MESS:00:00:11.746834:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:11.754034:0: dtparam: arm_freq=1200000000
MESS:00:00:11.764055:0: dtdebug: Found override arm_freq
MESS:00:00:11.767714:0: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
MESS:00:00:11.792691:0: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
MESS:00:00:11.816202:0: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
MESS:00:00:11.839762:0: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
MESS:00:00:11.863390:0: dtparam: core_freq=250000000
MESS:00:00:11.872277:0: dtdebug: /__overrides__ has no core_freq property
MESS:00:00:11.877380:0: Unknown dtparam 'core_freq' - ignored
MESS:00:00:11.911632:0: dtdebug: delete_node(/hat)
MESS:00:00:11.927167:0: brfs: File read: 243 bytes
```

pelwell, 2019-11-27:
Is your `bootcode.bin` the latest? It is looking as though it is somehow failing to load the overlay even though you have verified its contents.

cleverca22, 2019-11-27:
the `bootcode.bin` came from a recent `2019-09-26-raspbian-buster-lite.img` i had downloaded
i'll try getting an even more recent one from github...

pelwell, 2019-11-27:
That should be fine.

I'm struggling to see how this could be failing for you and only you. How about loading some other overlays before and after `disable-bt` to see if others are affected and whether doing so mysteriously fixes it?

cleverca22, 2019-11-27:
`cat ${firmware}/boot/bootcode.bin | sed -e "s/BOOT_UART=0/BOOT_UART=1/" >bootcode.bin`
```
[root@system76:~]# md5sum /mnt/bootcode.bin
088e3137bf0fa3dbcaaecb794ea6cfbe  /mnt/bootcode.bin
```
similar behavior after updating to the latest bootcode from github
and something broke the run button i added when swapping SD cards, so debug is slower, let me fix that...

cleverca22, 2019-11-27:
need to redo that switch later, just wiggling the wires fixed it!

```
# cat ../9080d9b6/config.txt
initramfs initrd followkernel
dtoverlay=act-led
dtoverlay=disable-bt
enable_uart=1
gpu_mem=32
uart_2ndstage=1
dtdebug=1
```
leads to:
```
MESS:00:00:06.425535:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:06.429668:0: brfs: File read: 120 bytes
MESS:00:00:12.142225:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:12.343665:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:12.351093:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:12.357115:0: *** Restart logging
MESS:00:00:12.380418:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:12.409203:0: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
MESS:00:00:12.413865:0: Loading 'bcm2710-rpi-3-b.dtb' to 0x100 size 0x6743
MESS:00:00:12.564907:0: brfs: File read: 26435 bytes
MESS:00:00:12.571117:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:12.575028:0: brfs: File read: 120 bytes
MESS:00:00:12.582288:0: dterror: not a valid FDT - err -9
MESS:00:00:12.585985:0: Failed to load overlay 'act-led'
MESS:00:00:12.594161:0: dterror: not a valid FDT - err -9
MESS:00:00:12.597882:0: Failed to load overlay 'disable-bt'
MESS:00:00:12.603483:0: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
MESS:00:00:12.610792:0: brfs: File read: 954 bytes
MESS:00:00:12.617003:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:12.621020:0: Read command line from file 'cmdline.txt':
MESS:00:00:12.626874:0: 'console=ttyS0,115200 pi3-disable-bt kgdboc=ttyS0,115200 systemConfig=/nix/store/bvzwk35xk7smagklbwpmv3bfih0kcfpf-not-os netroot=192.168.2.1=9080d9b6/root.squashfs quiet splash plymouth.ignore-serial-consoles plymouth.ignore-ude
MESS:00:00:14.843750:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:14.957555:0: brfs: File read: 243 bytes
MESS:00:00:28.793561:0: brfs: File read: /mfs/sd/kernel7.img
MESS:00:00:28.797524:0: Loading 'kernel7.img' to 0x8000 size 0x3711d0
MESS:00:00:28.803693:0: Device tree loaded to 0x2eff9400 (size 0x6bc6)
MESS:00:00:28.811156:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:28.817648:0: uart: Baud rate change done...
MESS:00:00:28.821077:0: uart: Baud rate
```

should `dtoverlay` just be the name of an overlay, or a full path with a `.dtbo` ?

cleverca22, 2019-11-27:
```
Nov 27 15:19:09 router tftpd[31186]: tftpd: trying to get file: 9080d9b6/overlays/act-led.dtbo
Nov 27 15:19:09 router tftpd[31188]: tftpd: trying to get file: 9080d9b6/overlays/disable-bt.dtbo
```
tftp logs show it did `overlays/%s.dtbo` on its own, but the logs from `start.elf` say it failed to find it...

pelwell, 2019-11-27:
Just the basename. It's clearly locating the file OK because the size of the overlay is correct.

> but the logs from start.elf say it failed to find it...

Not quite - the errors say it failed to load it, which isn't quite the same:
```
MESS:00:00:12.582288:0: dterror: not a valid FDT - err -9
MESS:00:00:12.585985:0: Failed to load overlay 'act-led'
MESS:00:00:12.594161:0: dterror: not a valid FDT - err -9
MESS:00:00:12.597882:0: Failed to load overlay 'disable-bt'
```

cleverca22, 2019-11-27:
i copied the boot files to the SD card:
```
Raspberry Pi Bootcode
Read File: config.txt, 125
Read File: start.elf, 2880356 (bytes)
Read File: fixup.dat, 6744 (bytes)
MESS:00:00:01.493907:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.498155:0: brfs: File read: 125 bytes
MESS:00:00:01.554077:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.558342:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:01.762888:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:01.770321:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:01.776342:0: *** Restart logging
MESS:00:00:01.780230:0: brfs: File read: 125 bytes
MESS:00:00:01.801772:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:01.954494:0: brfs: File read: /mfs/sd/initrd
MESS:00:00:01.958022:0: Loading 'initrd' to 0x0 size 0x203181
MESS:00:00:02.206371:0: initramfs loaded to 0x2edfc000 (size 0x203181)
MESS:00:00:02.215545:0: Trying Device Tree file 'bcm2710-rpi-3-b.dtb'
MESS:00:00:02.220290:0: brfs: File read: 2109825 bytes
MESS:00:00:02.230430:0: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
MESS:00:00:02.235091:0: Loading 'bcm2710-rpi-3-b.dtb' to 0x100 size 0x6743
MESS:00:00:02.260245:0: dtdebug: delete_node(/__local_fixups__)
MESS:00:00:02.274842:0: dtdebug: /aliases:i2c_vc=i2c0
MESS:00:00:02.288633:0: dtdebug: /__symbols__:i2c_vc=i2c0
MESS:00:00:02.299974:0: dtdebug: /__overrides__:i2c_vc=i2c0
MESS:00:00:02.318666:0: dtdebug: /__overrides__:i2c_vc_baudrate=i2c0_baudrate
MESS:00:00:02.329019:0: dtdebug: /aliases:i2c=i2c1
MESS:00:00:02.342779:0: dtdebug: /__symbols__:i2c=i2c1
MESS:00:00:02.353955:0: dtdebug: /__overrides__:i2c=i2c1
MESS:00:00:02.362257:0: dtdebug: /aliases:i2c_arm=i2c1
MESS:00:00:02.376395:0: dtdebug: /__symbols__:i2c_arm=i2c1
MESS:00:00:02.387973:0: dtdebug: /__overrides__:i2c_arm=i2c1
MESS:00:00:02.406941:0: dtdebug: /__overrides__:i2c_baudrate=i2c1_baudrate
MESS:00:00:02.427230:0: dtdebug: /__overrides__:i2c_arm_baudrate=i2c1_baudrate
MESS:00:00:02.433024:0: dtparam: uart0_clkrate=48000000
MESS:00:00:02.443225:0: dtdebug: /__overrides__ has no uart0_clkrate property
MESS:00:00:02.448659:0: Unknown dtparam 'uart0_clkrate' - ignored
MESS:00:00:02.454484:0: brfs: File read: 26435 bytes
MESS:00:00:02.461461:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.465376:0: brfs: File read: 125 bytes
MESS:00:00:02.475187:0: dtdebug: Opened overlay file 'overlays/act-led.dtbo'
MESS:00:00:02.481379:0: brfs: File read: /mfs/sd/overlays/act-led.dtbo
MESS:00:00:02.501080:0: Loaded overlay 'act-led.dtbo'
MESS:00:00:02.504465:0: dtdebug: Found fragment 0 (offset 36)
MESS:00:00:02.532689:0: dtdebug: merge_fragment(/leds/act,/fragment@0/__overlay__)
MESS:00:00:02.538577:0: dtdebug:   +prop(gpios)
MESS:00:00:02.543784:0: dtdebug: merge_fragment() end
MESS:00:00:02.547751:0: brfs: File read: 569 bytes
MESS:00:00:02.556114:0: dtdebug: Opened overlay file 'overlays/disable-bt.dtbo'
MESS:00:00:02.562359:0: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
MESS:00:00:02.590103:0: Loaded overlay 'disable-bt'
MESS:00:00:02.593349:0: dtdebug: Found fragment 0 (offset 36)
MESS:00:00:02.611861:0: dtdebug: merge_fragment(/soc/serial@7e215040,/fragment@0/__overlay__)
MESS:00:00:02.618703:0: dtdebug:   +prop(status)
MESS:00:00:02.624994:0: dtdebug: merge_fragment() end
MESS:00:00:02.628384:0: dtdebug: Found fragment 1 (offset 116)
MESS:00:00:02.644490:0: dtdebug: merge_fragment(/soc/serial@7e201000,/fragment@1/__overlay__)
MESS:00:00:02.651333:0: dtdebug:   +prop(pinctrl-names)
MESS:00:00:02.658546:0: dtdebug:   +prop(pinctrl-0)
MESS:00:00:02.664002:0: dtdebug:   +prop(status)
MESS:00:00:02.669490:0: dtdebug: merge_fragment() end
MESS:00:00:02.672884:0: dtdebug: Found fragment 2 (offset 228)
MESS:00:00:02.688514:0: dtdebug: merge_fragment(/soc/gpio@7e200000/uart0_pins,/fragment@2/__overlay__)
MESS:00:00:02.696126:0: dtdebug:   +prop(brcm,pins)
MESS:00:00:02.702985:0: dtdebug:   +prop(brcm,function)
MESS:00:00:02.708763:0: dtdebug:   +prop(brcm,pull)
MESS:00:00:02.714207:0: dtdebug: merge_fragment() end
MESS:00:00:02.717612:0: dtdebug: Found fragment 3 (offset 320)
MESS:00:00:02.733187:0: dtdebug: merge_fragment(/soc/gpio@7e200000/bt_pins,/fragment@3/__overlay__)
[... 16 lines cut ...]
MESS:00:00:04.993122:0: dtdebug: Found override cam0-pwdn
MESS:00:00:05.008728:0: dtparam: cam0-led-ctrl=22
MESS:00:00:05.017079:0: dtdebug: Found override cam0-led-ctrl
MESS:00:00:05.021141:0: dtparam: cam0-led=6
MESS:00:00:05.030365:0: dtdebug: Found override cam0-led
MESS:00:00:05.033992:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:05.041211:0: dtparam: arm_freq=1200000000
MESS:00:00:05.051231:0: dtdebug: Found override arm_freq
MESS:00:00:05.054871:0: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
MESS:00:00:05.079831:0: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
MESS:00:00:05.103224:0: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
MESS:00:00:05.126723:0: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
MESS:00:00:05.150302:0: dtparam: core_freq=400000000
MESS:00:00:05.159175:0: dtdebug: /__overrides__ has no core_freq property
MESS:00:00:05.164255:0: Unknown dtparam 'core_freq' - ignored
MESS:00:00:05.198490:0: dtdebug: delete_node(/hat)
MESS:00:00:05.213866:0: brfs: File read: 243 bytes
MESS:00:00:05.467538:0: brfs: File read: /mfs/sd/kernel7.img
MESS:00:00:05.471500:0: Loading 'kernel7.img' to 0x8000 size 0x3711d0
MESS:00:00:05.477672:0: Device tree loaded to 0x2edf5400 (size 0x6be6)
MESS:00:00:05.485157:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:05.491620:0: uart: Baud rate change done...
MESS:00:00:05.495054:0: uart: Baud rate change done...
MESS:00:00:05.500705:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
```

cleverca22, 2019-11-27:
still hangs at the rainbow when booting like that

cleverca22, 2019-11-27:
if i swap the uSD card over th an rpi2 v1.1:
```
Raspberry Pi Bootcode
Read File: config.txt, 125
Read File: start.elf, 2880356 (bytes)
Read File: fixup.dat, 6744 (bytes)
MESS:00:00:01.364226:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.368475:0: brfs: File read: 125 bytes
MESS:00:00:01.423606:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.427858:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:01.444658:0: gpioman: gpioman_get_pin_num: pin WL_LPO_CLK not defined
MESS:00:00:01.450365:0: gpioman: gpioman_get_pin_num: pin BT_ON not defined
MESS:00:00:01.457045:0: gpioman: gpioman_get_pin_num: pin WL_ON not defined
MESS:00:00:01.638300:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:01.645733:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:01.651754:0: *** Restart logging
MESS:00:00:01.655642:0: brfs: File read: 125 bytes
MESS:00:00:01.676938:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:01.829264:0: brfs: File read: /mfs/sd/initrd
MESS:00:00:01.832792:0: Loading 'initrd' to 0x0 size 0x203181
MESS:00:00:02.080641:0: initramfs loaded to 0x2edfc000 (size 0x203181)
MESS:00:00:02.089816:0: Trying Device Tree file 'bcm2709-rpi-2-b.dtb'
MESS:00:00:02.094559:0: brfs: File read: 2109825 bytes
MESS:00:00:02.103271:0: Failed to load Device Tree file '?'
MESS:00:00:02.107592:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:02.112588:0: Read command line from file 'cmdline.txt':
MESS:00:00:02.118440:0: 'console=ttyS0,115200 pi3-disable-bt kgdboc=ttyS0,115200 systemConfig=/nix/store/bvzwk35xk7smagklbwpmv3bfih0kcfpf-not-os netroot=192.168.2.1=9080d9b6/root.squashfs quiet splash plymouth.ignore-serial-consoles plymouth.ignore-ude
MESS:00:00:02.141116:0: brfs: File read: 243 bytes
MESS:00:00:02.390000:0: brfs: File read: /mfs/sd/kernel7.img
MESS:00:00:02.393963:0: Loading 'kernel7.img' to 0x8000 size 0x3711d0
MESS:00:00:02.400162:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:02.408213:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:02.415042:0: uart: Baud rate change done...
MESS:00:00:02.418472:0: uart: Baud rate change done...
MESS:00:00:02.424150:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
```
it also hangs on the rainbow

pelwell, 2019-11-27:
The load errors are gone, so that's one thing. I've just noticed that some of logs show that an initrd is being used, but not others. Can you account for the variability? Have you rebuilt your initrd to match the kernel?

I strongly advise you to use an off-the-shelf kernel and .dtb (such as those from https://github.com/raspberrypi/firmware/tree/master/boot) to pipeclean the network booting process before using your own builds and advanced features such as initrd. Also change your cmdline.txt to something simple like:
```
console=serial0,115200 console=tty1 <insert-your-root> elevator=deadline fsck.repair=yes rootwait
```

cleverca22, 2019-11-27:
this kernel/initrd/root.squashfs have booted before in the past

and the network boot has always been randomly failing for me

i can try the kernel from the rasbian image, and see if things are any different

cleverca22, 2019-11-27:
that also fails, let me undo all changes to the uSD and see what the .img does on its own
then progressively switch back to my config one thing at a time...

cleverca22, 2019-11-27:
unmodified `.img` file boots, resizes its own rootfs, then boots more, so the hardware is fine

cleverca22, 2019-11-27:
for cross-reference later, https://gist.github.com/cleverca22/5024fdc6f14b56408de0257de379b767 is a normal boot without `BOOT_UART` but with `uart_2ndstage` and `dtdebug`

cleverca22, 2019-11-27:
```
[    4.779682] systemd[1]: Listening on Journal Socket.
[    4.813973] systemd[1]: Created slice system-getty.slice.
[    4.849482] systemd[1]: Starting Create list of required static device nodes for the current kernel...
MESS:00:00:12.482001:0: brfs: File read: 5310624 bytes
```
is it normal that the VC4 is reading files that late into the linux boot?

cleverca22, 2019-11-27:
aha, i see part of the problem
`console=serial0,115200` works
`console=ttyS0,115200` doesnt work!

i think the new dtb files renamed things, and my older kernel respects those names, then cant find the old serial device

pelwell, 2019-11-27:
> is it normal that the VC4 is reading files that late into the linux boot?

That's the size of the kernel, which has obviously been loaded some time ago. I commented earlier on the difficulty of matching up some of the log messages with the events that triggered them ("The sequencing of log messages is a bit odd..."), so I think this is just an extreme example of that.

`serial0` and `serial1` are aliases that the firmware interprets as meaning the primary (console) UART and the secondary (Bluetooth) UART, respectively. Based on the Device Tree content it replaces them with ttyAMA0 and ttyS0, as appropriate.

cleverca22, 2019-11-27:
```
MESS:00:00:01.743898:0: 'console=serial0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait'
[    0.000000] Kernel command line: coherent_pool=1M bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=1024 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```

aha, yep, the cmdline from the config doesnt match the cmdline linux saw

that might also solve the need to disable bluetooth, let the firmware pass whichever uart is enabled to linux

cleverca22, 2019-11-27:
network booting is still very unstable
```
Nov 27 16:38:04 router tftpd[19891]: tftpd: trying to get file: 1077df95/cmdline.txt
Nov 27 16:38:04 router tftpd[19895]: tftpd: trying to get file: 1077df95/recovery7.img
Nov 27 16:38:04 router tftpd[19901]: tftpd: trying to get file: 1077df95/armstub7.bin
```
and 
```

Raspberry Pi Bootcode
Read File: config.txt, 120
Read File: start.elf, 2880356 (bytes)
Read File: fixup.dat, 6744 (bytes)
MESS:00:00:07.427456:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.431589:0: brfs: File read: 120 bytes
MESS:00:00:12.519051:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:12.534908:0: gpioman: gpioman_get_pin_num: pin WL_LPO_CLK not defined
MESS:00:00:12.540613:0: gpioman: gpioman_get_pin_num: pin BT_ON not defined
MESS:00:00:12.547291:0: gpioman: gpioman_get_pin_num: pin WL_ON not defined
MESS:00:00:12.578534:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:12.585958:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:12.591980:0: *** Restart logging
MESS:00:00:12.615099:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:12.630625:0: Failed to load Device Tree file '?'
MESS:00:00:12.637794:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:12.641811:0: Read command line from file 'cmdline.txt':
MESS:00:00:12.647663:0: 'console=serial0,115200 pi3-disable-bt kgdboc=serial0,115200 systemConfig=/nix/store/bvzwk35xk7smagklbwpmv3bfih0kcfpf-not-os netroot=192.168.2.1=9080d9b6/root.squashfs quiet splash plymouth.ignore-serial-consoles plymouth.ignore
MESS:00:00:12.670408:0: brfs: File read: 247 bytes
MESS:00:00:19.182774:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:23.988633:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
```
without any software or network changes, rebooting the pi randomly gets stuck at different phases

pelwell, 2019-11-27:
I am suspicious that there is some uninitialised variable or other state hanging around that causes this variability.

cleverca22, 2019-11-27:
i have noticed that there are differences between shorting `RUN` vs pulling the plug

cleverca22, 2019-11-27:
something else ive been curious about, but it may belong in a different issue

how complicated would it be for the foundation to modify `start.elf`, such that it can be booted after the ARM is already online?

basically, replace bootcode.bin and start.elf with open-source software, boot linux on the ARM core, then have linux shove a `start.elf` into ram, and the open-source firmware will execute the closed-source `start.elf`, which should then leave the arm core as-is

pelwell, 2019-11-27:
The ARMs stay in reset until the VPU wakes them up - they have no Boot ROM. Some people replace the kernel with U-boot, which then loads the real kernel.

cleverca22, 2019-11-27:
thats where https://github.com/christinaa/rpi-open-firmware/ comes in
it replaces bootcode.bin and start.elf

but then you loose nearly all of the gpu features (h264, hdmi, v3d)

which is why i was thinking of an option where you can load the existing start.elf blob into ram after booting, to re-enable such features

pelwell, 2019-11-27:
It's a massive effort with no obvious benefit to most users, so I can't see it happening.

cleverca22, 2019-11-27:
the benefits i can see:
* the early boot process will be more open, so it would be simpler to debug issues like in this thread
* the closed-source firmware becomes optional, but you still have the choice to switch over, after booting

cleverca22, 2019-11-27:
hmmm, something i could see that would help debug problems i've suspected, what about a debug flag to make the firmware print the hash of every file it loads?

pelwell, 2019-11-27:
That's not a bad idea - I'll add it to the list of possible improvements.

cleverca22, 2019-11-27:
i put the entire `/boot` partition from the `.img` onto the tftp server
and then left only `bootcode.bin` on the SD card

this should allow me to test the whole netboot chain, but leave the rootfs on the SD card
and also swap the kernel out remotely

on the first boot, it hangs at:
```
...
Waiting for dhcp_reply
ARP request: my IP addr 192.168.2.196 my eth_addr b8:27:eb:77:df:95 ARP for addr 192.168.2.1
Done ARP for 192.168.2.1 got 00:30:48:c5:d2:df
Read File: config.txt, 1784 (bytes)

Raspberry Pi Bootcode
Read File: config.txt, 1784
Read File: fixup.dat, 6736 (bytes)
```
and is giving 4 blinks and no rainbow

resetting via `RUN` leads to:
```ARP request: my IP addr 192.168.2.197 my eth_addr b8:27:eb:77:df:95 ARP for addr 192.168.2.1
Done ARP for 192.168.2.1 got 00:30:48:c5:d2:df

Raspberry Pi Bootcode
Read File: start.elf, 2877988 (bytes)
Read File: fixup.dat, 6736 (bytes)
MESS:00:00:19.050536:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:19.055299:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:19.071638:0: gpioman: gpioman_get_pin_num: pin WL_LPO_CLK not defined
MESS:00:00:19.077338:0: gpioman: gpioman_get_pin_num: pin BT_ON not defined
MESS:00:00:19.084018:0: gpioman: gpioman_get_pin_num: pin WL_ON not defined
MESS:00:00:19.115304:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:19.122733:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:19.128754:0: *** Restart logging
MESS:00:00:19.132643:0: brfs: File read: 1784 bytes
MESS:00:00:19.156502:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:19.167201:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:19.171202:0: Read command line from file 'cmdline.txt':
MESS:00:00:19.177068:0: 'console=serial0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet init=/usr/lib/raspi-config/init_resize.sh'
MESS:00:00:19.198654:0: brfs: File read: 169 bytes
MESS:00:00:19.217680:0: brfs: File read: /mfs/sd/bcm2709-rpi-2-b.dtb
MESS:00:00:19.222334:0: Loading 'bcm2709-rpi-2-b.dtb' to 0x100 size 0x62cd
MESS:00:00:19.387091:0: brfs: File read: 25293 bytes
MESS:00:00:19.394331:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:19.398781:0: dtparam: audio=on
MESS:00:00:21.472433:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:21.567823:0: brfs: File read: 1784 bytes
MESS:00:00:24.280866:0: brfs: File read: /mfs/sd/kernel7.img
MESS:00:00:24.284827:0: Loading 'kernel7.img' to 0x8000 size 0x5108a0
MESS:00:00:24.290995:0: Device tree loaded to 0x2eff9900 (size 0x66c0)
MESS:00:00:24.298121:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:24.304950:0: uart: Baud rate change done...
MESS:00:00:24.308382:0: uart: Baud rate change done...
MESS:00:00:25.721682:0: vchiq_core: vchiq_init_state: slot_zero = 0xfad80000, is_master = 1
MESS:00:00:25.732741:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:25.745579:0: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000023
[    2.571532] sysrq: SysRq : Resetting
[    2.575382] CPU2: stopping
[    2.578540] CPU0: stopping
[    2.581638] CPU1: stopping

Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Enabling PORT POWER on port 4
Enabling PORT POWER on port 5
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Trying booting from Ethernet device addr 5
Initialise ethernet with MAC b8:27:eb:77:df:95
Wait for Link up
Link up
Sending DHCP request
Waiting for dhcp_reply
ARP request: my IP addr 192.168.2.198 my eth_addr b8:27:eb:77:df:95 ARP for addr 192.168.2.1
Done ARP for 192.168.2.1 got 00:30:48:c5:d2:df
Read File: config.txt, 1784 (bytes)

Raspberry Pi Bootcode
Read File: config.txt, 1784
Read File: fixup.dat, 6736 (bytes)
```

something on the uart upset linux? and it went back into the firmware, then repeated the 1st failure
i'll unplug the rpi's rx line, so it cant do that...

cleverca22, 2019-11-27:
oh, wait, is it supposed to remove `init=/usr/lib/raspi-config/init_resize.sh` from its own config, on first boot?

pelwell, 2019-11-27:
The mini/AUX UART (ttyS0) has broken stop/break detection, so a line held permanently low upsets it.

> oh, wait, is it supposed to remove init=/usr/lib/raspi-config/init_resize.sh from its own config, on first boot?

Yes, after resizing the main partition.

cleverca22, 2019-11-27:
without any changes to the binary files (just `init=` removed from `cmdline.exe`), on rebooting, it randomly has several outcomes:
* start.elf fails to load (no rainbow, 4 blinks)
* kernel doesnt load (not mentioned in los, 7 blinks i think)
* kernel does load, but it hangs at the rainbow screen
* kernel fully loads (only happened twice, before i removed `init=..`, so it rebooted itself)

cleverca22, 2019-11-27:
with `init=` removed, and rebooting repeatedly, it does eventually land in a regular Raspbian shell

so it looks like my network boot setup is "working" and just as unstable as it was a few years ago

where shal we go next to debug why its so unstable?

pelwell, 2019-11-27:
An md5 hash patch, at least for a test firmware, sounds like the way to go. I'll (try to) get back to you tomorrow.

cleverca22, 2019-11-27:
```
pi@raspberrypi:~$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether b8:27:eb:77:df:95 brd ff:ff:ff:ff:ff:ff
```
and the network isnt even working after it booted!

the link LED is on, yet linux claims its down

pelwell, 2019-11-28:
There's a test firmware (just start.elf and fixup.dat, so not suitable for a Pi 4) here: https://drive.google.com/file/d/1f5hTq72vbr4UbsdvSC8pB2kMGrIrI10W/view?usp=sharing
With `md5sum=1` in config.txt you should find it displays the hashes of loaded files.

There are a few caveats:
1. It's not hooked into the lowest level file API, but is should work for OS files and overlays.
2. The overlay API made it awkward to include the filename in the same message as the hash, but it does appear on the next line.

pelwell, 2019-11-28:
3. I've been testing using regular SD card booting, but it should work equally well when network booting.

cleverca22, 2019-11-28:
rpi4 is still in the mail, so that wont be an issue currently

```
MESS:00:00:13.845397:0: Loaded 'bcm2709-rpi-2-b.dtb' to 0x100 size 0x62cd (md5 28fc26c74463137b6be5b69c5055b0ae)
MESS:00:00:24.640164:0: Loaded 'kernel7.img' to 0x8000 size 0x5108a0 (md5 2adb211a405f961abd1ac2438091ff1e)
```

i can confirm the hashes are correct, when things boot fully

once when booting, it hung at:
```
...
Wait for Link up
Link up
Sending DHCP request
Waiting for dhcp_reply
ARP request: my IP addr 192.168.2.116 my eth_addr b8:27:eb:77:df:95 ARP for addr 192.168.2.1
Done ARP for 192.168.2.1 got 00:30:48:c5:d2:df

Raspberry Pi Bootcode
Read File: fixup.dat, 6743 (bytes)
```
and was just giving 4 blinks
so things are also failing at the `bootcode.bin` stage, before this patch can do any help

i'll see if i can reproduce problems at the `start.elf` stage...

cleverca22, 2019-11-28:
```
MESS:00:00:13.583351:0: Loaded 'bcm2709-rpi-2-b.dtb' to 0x100 size 0x62cd (md5 28fc26c74463137b6be5b69c5055b0ae)
MESS:00:00:13.752028:0: brfs: File read: 25293 bytes
MESS:00:00:13.758962:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:13.763427:0: dtparam: audio=on
MESS:00:00:13.786552:0: brfs: File read: 1794 bytes
MESS:00:00:13.792753:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:13.796742:0: Read command line from file 'cmdline.txt':
MESS:00:00:13.802617:0: 'console=serial0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait'
MESS:00:00:15.866022:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:15.958146:0: gpioman: gpioman_get_pin_num: pin WL_ON not defined
MESS:00:00:15.975324:0: brfs: File read: 121 bytes
MESS:00:00:24.169755:0: brfs: File read: /mfs/sd/kernel.img
MESS:00:00:24.511874:0: Loaded 'kernel.img' to 0x8000 size 0x4cbd38 (md5 860365692a3bd4c2425b94d5bb788e4f)
MESS:00:00:24.519832:0: Device tree loaded to 0x2eff9900 (size 0x6690)
MESS:00:00:24.526924:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:24.533790:0: uart: Baud rate change done...
MESS:00:00:24.537219:0: uart: Baud rate change done...
```
and:
```
Nov 28 19:11:33 router tftpd[19380]: tftpd: trying to get file: 1077df95/fixup.dat
Nov 28 19:11:33 router tftpd[19382]: tftpd: trying to get file: 1077df95/recovery.elf
Nov 28 19:11:33 router tftpd[19384]: tftpd: trying to get file: 1077df95/config.txt
Nov 28 19:11:33 router tftpd[19386]: tftpd: trying to get file: 1077df95/dt-blob.bin
Nov 28 19:11:33 router tftpd[19388]: tftpd: trying to get file: 1077df95/recovery.elf
Nov 28 19:11:34 router tftpd[19390]: tftpd: trying to get file: 1077df95/config.txt
Nov 28 19:11:34 router tftpd[19399]: tftpd: trying to get file: 1077df95/bootcfg.txt
Nov 28 19:11:34 router tftpd[19401]: tftpd: trying to get file: 1077df95/bcm2709-rpi-2-b.dtb
Nov 28 19:11:34 router tftpd[19403]: tftpd: trying to get file: 1077df95/config.txt
Nov 28 19:11:34 router tftpd[19405]: tftpd: trying to get file: 1077df95/cmdline.txt
Nov 28 19:11:36 router tftpd[19415]: tftpd: trying to get file: 1077df95/recovery7.img
Nov 28 19:11:36 router tftpd[19417]: tftpd: trying to get file: 1077df95/recovery.img
Nov 28 19:11:42 router tftpd[19444]: tftpd: trying to get file: 1077df95/kernel.img
Nov 28 19:11:42 router tftpd[19446]: tftpd: trying to get file: 1077df95/armstub7.bin
Nov 28 19:11:42 router tftpd[19448]: tftpd: trying to get file: 1077df95/kernel.img
```
and just one reboot later, it entirely skips `kernel7.img`, tries to instead load `kernel.img` on the same hardware, (hash is correct) and hangs (likely due to running a v6 kernel on a rpi2)

cleverca22, 2019-11-28:
my current theory on that specific bug, is that if `kernel7.img` "does not exist" it will silently fall back to `kernel.img`

and if a random network glitch causes the tftp to fail, the firmware may blindly assume the file doesnt exist?

working case:
```
MESS:00:00:13.796742:0: Read command line from file 'cmdline.txt':
MESS:00:00:13.802617:0: 'console=serial0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait'
MESS:00:00:15.866022:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:15.958146:0: gpioman: gpioman_get_pin_num: pin WL_ON not defined
MESS:00:00:15.975324:0: brfs: File read: 121 bytes
MESS:00:00:24.169755:0: brfs: File read: /mfs/sd/kernel.img
```
2.06 second delay between cmdline and kernel

and the failed case above 10.37 seconds!!
```
-rwxr-xr-x 1 root root 5.1M Sep 25 14:28 kernel7.img
-rwxr-xr-x 1 root root 4.8M Sep 25 14:28 kernel.img
```

definitely looks like a timeout caused `start.elf` (and likely `bootcode.bin`) to falsely assume the file doesnt exist, and just move on (or hard fail)

would it be possible for both firmware files to log all file-not-found events? and perhaps improve the debug within the tftp layer?

cleverca22, 2019-12-06:
@pelwell any update?

pelwell, 2019-12-06:
Improved logging isn't going to solve the underlying problem, which is that network packets are either going astray or being ignored. I'm not aware of anyone else in the same situation, suggesting there is an environmental factor.

Any ideas, @ghollingworth ?

cleverca22, 2019-12-06:
there is also a bug in my router's NIC driver, that causes the tx queue to entirely shutdown randomly

but, the tftp protocol should have a proper error code for `file not found` and treating no response as `file not found` feels like a bug that hould still be fixed or at least logged

cleverca22, 2019-12-06:
i have not configmred if the router issue is related though, since some of the boot failures didnt line up with router failures, and logging timeouts would still be of use

cleverca22, 2019-12-13:
@pelwell & @ghollingworth ping?

```
04:25:28.971722 IP 192.168.2.15.52444 > 192.168.2.1.69:  19 RRQ "bar.txt" netascii
E../7.@.@.}............E......bar.txt.netascii.
04:25:28.974440 IP 192.168.2.1.40195 > 192.168.2.15.52444: UDP, length 19
E../=k@.@.w.....................File not found.
04:25:28.974530 IP 192.168.2.15.52444 > 192.168.2.1.40195: UDP, length 4
E.. 7.@.@.}................~....
```
if i try to fetch a file that doesnt exist, the tftp server clearly answers with an error stating such

silently treating timeouts as file-not-found just turns random recoverable errors into fatal errors and weird glitches (loading the wrong kernel for example)