raspberrypi/firmware #1294: No sound on HDMI or headphone jack except hello_audio

jdescelliers, 2019-11-30:
**Is this the right place for my bug report?**
Yes.

**Describe the bug**
No sound on either HDMI or headphone jack.

**To reproduce**
- distro: Raspian running Debian 10.2 (Buster)
- try to play audio or video
- no sound

**Expected behaviour**
Sound should be working.

**Actual behaviour**
No sound.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Pi 4
* Which OS and version (`cat /etc/rpi-issue`)? Debian 10.2 Buster (raspian)  - Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage5
* Which firmware version (`vcgencmd version`)?
Sep 24 2019 17:34:30 
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

**Logs**
dmesg logs:
```console
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.75-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1270 SMP Tue Sep 24 18:51:41 BST 2019
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 816128 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e8 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632
[    0.000000] pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M video=HDMI-A-1:1920x1080M@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0 smsc95xx.macaddr=DC:A6:32:45:F9:1A vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=5e3da3da-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 3735592K/4050944K available (8192K kernel code, 661K rwdata, 2352K rodata, 2048K init, 850K bss, 53208K reserved, 262144K cma-reserved, 3264512K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 662 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 851 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 28124 entries in 83 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[... 335 lines cut ...]
[   14.360683] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.360689] Bluetooth: BNEP filters: protocol multicast
[   14.360700] Bluetooth: BNEP socket layer initialized
[   14.429783] Bluetooth: RFCOMM TTY layer initialized
[   14.429799] Bluetooth: RFCOMM socket layer initialized
[   14.429812] Bluetooth: RFCOMM ver 1.11
```

**Additional context**
- audio is working over headphone jack from the terminal by typing: `/opt/vc/src/hello_pi/hello_audio/hello_audio.bin`
- Audio device settings: 
  - sound card: bcm2835 ALSA (only this entry is visible)
  - PCM sound bar is set to maximum

- Pulseaudio
  - package removed, purged, reboot and reinstalled: same issue
  - output devices: set to Analogue out (100%) - HDMI is NOT in the dropdown menu
  - configuration: internal audio > set to: Mono Analogue Out. No HDMI

VLC:
- playing either a video or audio: no sound
- Pulseaudio shows an audio bar moving while playing sound and this bar is almost at its maximum.

Chromium
- playing video: same issue

andrewpmiller, 2019-12-16:
@jdescelliers I had the same symptoms that you reported (at the time, my firmware version details were 4.19.86-v7l+ <span>#</span>1283). I updated the firmware a few days ago (4.19.88-v7l+ <span>#</span>1284) and the audio is back. 

When I first saw your issue, I ran `dmesg` and noticed a number of warnings about bcm modules that were loading from a staging directory and wondered if that was related to the loss of audio... but those warnings still appear in the output... so :man_shrugging: 

YMMV but it might be worth updating the firmware to see if your audio returns.

JamesH65, 2020-02-24:
Been a few changes in the audio area - can people seeing the issue please update to the latest firmware/Raspbian, retest, and if ok please close the issue.