raspberrypi/firmware #1295: Pi4: crackling sound using newer firmware files

mkreisl, 2019-12-02:
One of our user cant use latest firmware dated from

```
Nov 19 2019 16:42:26 
Copyright (c) 2012 Broadcom
version aeebba4c03968ede49097db077673eadc2888a22 (clean) (release) (start_x)
```

He is getting noise while Kodi is running and crackling sound while playing video

Reverting to an older firmware 

```
Sep 30 2019 14:39:52 
Copyright (c) 2012 Broadcom
version b1477501b8675de389acd184c2b7e7c4b2e39e4a (clean) (release) (start_x)
```

solves the issue for him.

To clearify it, the `*4*` files in /boot folder are meant, **not** the kernel itself

This is reproduceable on his Pi4/4GB version, it does **not** depend on which kernel version is used (4.19.76, 4.19.84 and 4.19.86 has been tried)

Even playing wav files via aplay producing same garbaged sound, so it is **not** Kodi related

Unfortunately I can not reproduce this with my Pi4/4GB version, that makes it a bit more complicated

ghollingworth, 2019-12-02:
Is this audio through HDMI or through analogue output?

mkreisl, 2019-12-02:
Sorry,  HDMI used

popcornmix, 2019-12-02:
Can you confirm if Nov 14 firmware has the issue?

mkreisl, 2019-12-03:
Have to checkout commit first, build package and sent it to my customer.
Will report today evening

mkreisl, 2019-12-03:
Yes, I built 2 versions, one reported Nov 14, the other one Nov 8, both versions having this issue already

popcornmix, 2019-12-03:
It looks like https://github.com/raspberrypi/firmware/commit/d737daa826adaab8679f83018c3cdc8783d49b59 is probably the first commit with the issue (would be useful if you can confirm).

@timg236 - any thoughts?

popcornmix, 2019-12-03:
@mkreisl info on the setup would be useful.
What was hdmi resolution/framerate?
How is audio configured (sample rate, number of channels, passthough)
Can he test analogue audio?
Does changing hdmi resolution affect the issue? Does changing sample rate/number of channels?

mkreisl, 2019-12-03:
As already written, crackling sound is always there, even with using aplay instead of kodi
 No overclock, no hdmi_enable_4kp60

popcornmix, 2019-12-03:
aplay with latest firmware here has no crackle.
Having a file known to produce the crackle would be useful, as would knowing the HDMI mode used.

mkreisl, 2019-12-03:
As alraedy written, I can not reproduce this issue. With my Pi4 everything is well

We used the wav files coming with Kodi package to test sound with aplay

mkreisl, 2019-12-03:
Ah, one thing I forgot: HDMI cable and monitor/tv already changed, no difference

popcornmix, 2019-12-03:
Adding ``core_freq=500`` to config.txt might avoid the issue (but just guessing).

popcornmix, 2019-12-05:
We did find an issue that could affect audio in lower resolution modes (e.g. 720p).
If they were using a lower resolution mode (than 1080p) then try with latest firmware.

mkreisl, 2019-12-05:
Thanks @popcornmix, we will try ASAP. The Pi4 having that sound issue is running 720p resolution

mkreisl, 2019-12-09:
@popcornmix We can confirm, with latest firmware sound issue has been gone
Thank you

pelwell, 2019-12-09:
Please close the issue if you feel it has been resolved.

mkreisl, 2019-12-09:
@pelwell The note is unnecessary. I always do that after a certain time

closed 2019-12-16 by mkreisl