raspberrypi/firmware #1296: Config.txt ignores conditional filters for hdmi settings

colerd24, 2019-12-04:
**Description**
There is a bug with the config.txt parser that ignores conditional filters only for hdmi settings, even when the settings are under the [none] filter.  

**Reproduction Steps**
1. Download fresh Raspbian image
2. Update (apt update && apt upgrade -y)
3. Replace config.txt with version listed below (see *config.txt*)
4. Connect Raspberry Pi to an HDMI display (anything that is not the resolution of hdmi_mode, 640x480 in this example)
5. Boot Raspberry Pi

*config.txt:*
In reality, [none] should be a filter that evaluates to false (none simulates false)
[none]
hdmi_ignore_edid=0xa5000080
hdmi_force_hotplug=1
hdmi_pixel_encoding=2
hdmi_drive=1
hdmi_group=2
hdmi_mode=4
[all]

**Expected Behaviour**
Display resolution should match that of the display as the custom HDMI implementations should not run when they are under the [none] filter (nor any filter that evaluates to false).

**Actual Behaviour**
Custom HDMI options are set and screen has output resolution as specified under the [none] filter (in this example it's 640x480).

**System**
Models Tested: CM 3B+, 3B+
OS: Raspbian Buster Lite (Fresh install, fully updated)
vcgencmd version: Sep 24 2019 6820edeee4ef3891b95fc01cf02a7abd7ca52f17
rpi-issue: 2019-09-26 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2
Kernel Version: 4.19.75-v7+

Also tried this after rpi-update to kernel 4.19.86-v7+

**Logs**
Output of vcdbg log msg:
001288.733: brfs: File read: /mfs/sd/config.txt
001289.187: brfs: File read: 139 bytes
001321.736: brfs: File read: /mfs/sd/config.txt
001542.160: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
001544.267: *** Restart logging
001544.331: brfs: File read: 139 bytes
001551.070: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001555.067: brfs: File read: /mfs/sd/cmdline.txt
001555.157: Read command line from file 'cmdline.txt':
001555.216: 'console=serial0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait'
001559.852: brfs: File read: 121 bytes
001563.592: brfs: File read: /mfs/sd/bcm2710-rpi-3-b-plus.dtb
001563.655: Loading 'bcm2710-rpi-3-b-plus.dtb' to 0x100 size 0x69ca
001717.610: brfs: File read: 27082 bytes
001720.957: brfs: File read: /mfs/sd/config.txt
003433.319: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
003534.620: brfs: File read: 139 bytes
003903.243: brfs: File read: /mfs/sd/kernel7.img
003903.307: Loading 'kernel7.img' to 0x8000 size 0x5108a0
003903.376: Device tree loaded to 0x2eff9200 (size 0x6de5)
003905.944: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
005331.235: vchiq_core: vchiq_init_state: slot_zero = 0xfad80000, is_master = 1
005335.580: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
005340.805: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000004
009791.047: brfs: File read: 5310624 bytes

colerd24, 2019-12-05:
I've identified the line that specifically is being ignored by the filter, causing all of the issues:
hdmi_ignore_edid=0xa5000080

Removing this line under the [none] filter allows the logic to work correctly. It also was not necessary to have for my use case of driving a TFT when a gpio was high and a normal HDMI display when it was low; however, this bug still exists for if people need to use hdmi_ignore_edid with conditional filters.

Example of my implementation:
[gpio#=1]
#TFT Settings here
[all]

[gpio#=0]
#HDMI with EDID settings here
[all]

pelwell, 2019-12-06:
Indeed - the `hdmi*` handling is borked. config.txt is processed in multiple passes, the state being reset between the passes. When the array types were added to handle the multiple HDMI inputs, they were omitted from the reset code, so the state accumulated during the early passes (when the filters are ignored) is preserved when it shouldn't be.

A potential fix is being reviewed internally, and it (or something like it) will appear in a future firmware release.

pelwell, 2019-12-06:
A workaround would be to put `hdmi_ignore_edid=0` in a `[none]` section after the one containing `hdmi_ignore_edid=0xa5000080` (or just comment it out).

pelwell, 2020-01-10:
The fix went into firmware version 0c01dbefba45a08c47f8538d5a071a0fba6b7e83. You can pick up an updated firmware using `sudo rpi-update` (not normally recommended, but it should be safe at the moment).

Please close the issue if you are satisfied with the outcome.

colerd24, 2020-02-10:
Thanks! Closing the issue, looks like it is working great after doing the update.

closed 2020-02-10 by colerd24