raspberrypi/firmware #1298: kernel errors at random with hung_task_timeout_secs on raspberry pi 4

fjleon1980, 2019-12-06:
i am having to reboot my raspberry pi 4b with 4GB ram twice a day because kernel errors keep popping up. It's caused by something related to the X server / VNC / Chromium because that's what i do.

When i get the error, i can't connect via VNC anymore, though the device still responds to SSH and i can run commands normally. I have seen errors regarding audio, video and more. log as follows:

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.75-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1270 SMP Tue Sep 24 18:51:41 BST 2019
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 816128 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e8 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632
[    0.000000] pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M video=HDMI-A-1:640x480M@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0 smsc95xx.macaddr=DC:A6:32:07:92:91 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 3735592K/4050944K available (8192K kernel code, 661K rwdata, 2352K rodata, 2048K init, 850K bss, 53208K reserved, 262144K cma-reserved, 3264512K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 662 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 851 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 28124 entries in 83 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000018] Switching to timer-based delay loop, resolution 18ns
[    0.000243] Console: colour dummy device 80x30
[    0.000268] console [tty1] enabled
[    0.000308] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000329] pid_max: default: 32768 minimum: 301
[    0.000591] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000610] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001360] CPU: Testing write buffer coherency: ok
[    0.001774] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002421] Setting up static identity map for 0x200000 - 0x20003c
[    0.002577] rcu: Hierarchical SRCU implementation.
[    0.003442] smp: Bringing up secondary CPUs ...
[    0.004345] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005381] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006374] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006508] smp: Brought up 1 node, 4 CPUs
[    0.006522] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.006531] CPU: All CPU(s) started in HYP mode.
[    0.006540] CPU: Virtualization extensions available.
[    0.007322] devtmpfs: initialized
[    0.017724] VFP support v0.3: implementor 41 architecture 3 part 40 variant 8 rev 0
[    0.017930] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017951] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.025465] pinctrl core: initialized pinctrl subsystem
[    0.026313] NET: Registered protocol family 16
[... 540 lines cut ...]
[ 7370.897322] [<c0997424>] (__schedule) from [<c0997a94>] (schedule+0x50/0xa8)
[ 7370.897329] [<c0997a94>] (schedule) from [<c099ba7c>] (schedule_timeout+0x200/0x428)
[ 7370.897337] [<c099ba7c>] (schedule_timeout) from [<c0998704>] (wait_for_common+0xd4/0x1b0)
[ 7370.897345] [<c0998704>] (wait_for_common) from [<c0998800>] (wait_for_completion+0x20/0x24)
[ 7370.897352] [<c0998800>] (wait_for_completion) from [<c08109f4>] (rpi_firmware_transaction+0x78/0xd0)
[ 7370.897359] [<c08109f4>] (rpi_firmware_transaction) from [<c0810b8c>] (rpi_firmware_property_list+0x140/0x2b0)
[ 7370.897367] [<c0810b8c>] (rpi_firmware_property_list) from [<c0810d78>] (rpi_firmware_property+0x7c/0xfc)
[ 7370.897375] [<c0810d78>] (rpi_firmware_property) from [<c07e7e74>] (bcm2835_cpufreq_clock_property.constprop.1+0x58/0x80)
[ 7370.897383] [<c07e7e74>] (bcm2835_cpufreq_clock_property.constprop.1) from [<c07e7ef8>] (bcm2835_cpufreq_driver_target_index+0x5c/0xe4)
[ 7370.897391] [<c07e7ef8>] (bcm2835_cpufreq_driver_target_index) from [<c07e2b54>] (__cpufreq_driver_target+0x2d8/0x5a8)
[ 7370.897399] [<c07e2b54>] (__cpufreq_driver_target) from [<c07e65a0>] (od_dbs_update+0xec/0x174)
[ 7370.897407] [<c07e65a0>] (od_dbs_update) from [<c07e71bc>] (dbs_work_handler+0x3c/0x68)
[ 7370.897414] [<c07e71bc>] (dbs_work_handler) from [<c023db40>] (process_one_work+0x170/0x458)
[ 7370.897421] [<c023db40>] (process_one_work) from [<c023de84>] (worker_thread+0x5c/0x5a4)
[ 7370.897427] [<c023de84>] (worker_thread) from [<c0244170>] (kthread+0x138/0x168)
[ 7370.897434] [<c0244170>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[ 7370.897438] Exception stack(0xd4221fb0 to 0xd4221ff8)
[ 7370.897443] 1fa0:                                     00000000 00000000 00000000 00000000
[ 7370.897448] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 7370.897453] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000

i have ran apt update/upgrade and even updated the firmware in the past. 
right now i'm running  4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

Please advise, i'm getting desperate.

pelwell, 2019-12-06:
It looks like the VPU firmware is failing to respond to requests from the kernel.
1. What are you typically doing with the Pi when you lose VNC connectivity?
2. In the dead state does it respond to `vcgencmd version`?
3. Add `start_debug=1` to config.txt and reboot, let it run until it fails, then post the output of `sudo vcdbg log msg` and  `sudo vcdbg log assert` .

fjleon1980, 2019-12-06:
i am always using chromium to watch videos on the hideout.tv site.
another bug i'm experiencing which may or may not be related: sometimes chromium freezes and even if i kill it and start it again, i cannot connect to any website anymore until i reboot. i still have network connectivity, it's just that something happens and chromium continues to freeze, not loading websites until i reboot.

i will try that command next time, but i think everything that doesn't require a GUI will still work in the dead state.

i will do the debug config and report back. meanwhile i blacklisted the audio driver to see if it makes a difference

fjleon1980, 2019-12-08:
this may be a different issue, but chromium just froze while vnc is still up. when i killed the process, i got this:

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.912836] Internal error: Oops: a06 [#1] SMP ARM

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913606] Process Chrome_ChildIOT (pid: 29003, stack limit = 0x253ba83e)

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913619] Stack: (0xda689d70 to 0xda68a000)

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913633] 9d60:                                     e1880000 00000000 00000000 bf15d1d0

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913649] 9d80: e1880000 00000000 d61e5780 d61e5790 ef82ce10 cc1f8990 00000000 00000008

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913665] 9da0: da689dc4 da689db0 bf15d1e4 bf15c414 d671b480 ef6c5b00 da689dec da689dc8

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913680] 9dc0: c07065f4 bf15d19c d8bdec00 ef6c5b00 ef82ce10 cc1f8990 00000000 ef6c5b00

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913696] 9de0: da689e2c da689df0 c03b1508 c07065a0 00000000 00000000 c0242190 d8bdec08

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913711] 9e00: d8214138 00000000 d821411c d8213c00 d8214138 c10a6510 d8bdec00 00000036

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913727] 9e20: da689e3c da689e30 c03b16bc c03b1470 da689e64 da689e40 c02421ac c03b16b0

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913742] 9e40: da689fb0 c1004d88 fffffe00 da689ec4 b55c5518 b55c551c da689ebc da689e68

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913757] 9e60: c02339fc c02420fc 00000000 00000000 6f726843 7542656d 43560066 61204d53

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913773] 9e80: d95c39c0 69616620 d7b81340 da689ec4 00000000 da689fb0 c1004d88 fffffe00

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913788] 9ea0: da689ec4 b55c5518 b55c551c 00000036 da689f8c da689ec0 c020c68c c0233170

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913804] 9ec0: da688000 00000036 da689f6c da689ed8 c03c4620 bf15e528 da689f40 da689f3c

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913819] 9ee0: 00000040 ab3594e8 da689f04 da689ef8 c03d08f4 c03cffc8 da689f14 7073e8a9

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913835] 9f00: c03d0968 c1004d88 d1b661c0 00000040 ab3594e8 c084abd0 d9a16780 c1004d88

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913863] 9f20: da689f94 da689f30 c084abd0 c03d08f4 0540f8f5 ab358c50 00000033 7073e8a9

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913879] 9f40: da689f5c c0e97ad8 c020cc10 c02011c4 da689fb0 c02011c4 da688000 7073e8a9

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913894] 9f60: da689f8c 00000085 da688010 c02011c4 da689fb0 c02011c4 da688000 00000036

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913910] 9f80: da689fac da689f90 c020cc88 c020c614 0540f8f5 00057000 00000033 00000036

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913925] 9fa0: 00000000 da689fb0 c0201034 c020cbb8 00000033 80404a5a ab358c50 4d534356

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913940] 9fc0: 0540f8f5 00057000 00000033 00000036 ab358c50 b6f90968 ab358cec ab358f44

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.913956] 9fe0: b6e98150 ab358c34 b6e841d8 b55c5518 20000010 00000033 00000000 00000000

Message from syslogd@raspbeer at Dec  8 09:45:01 ...
 kernel:[216237.914250] Code: e894000c e5950000 e3a0cc01 e3a01c02 (e5823004)

vcgencmd version works
Sep 24 2019 17:34:30
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start)

fjleon1980, 2019-12-08:
output from dmesg. since i have to reboot i will put the debug line on the config for next time.

[214769.863982] vc_sm_cma_vchi_rx_ack: received response 65536, throw away...
[216237.911365] [vc_sm_cma_ioctl_alloc]: something failed - cleanup. ret -512
[216237.911392] ------------[ cut here ]------------
[216237.911417] WARNING: CPU: 3 PID: 29003 at mm/memory.c:2299 apply_to_page_range+0x324/0x330
[216237.911427] Modules linked in: act_police sch_ingress cls_u32 sch_sfq sch_cbq xt_nat fuse ipt_REJECT nf_reject_ipv4 xt_conntrack rfcomm bnep hci_uart btbcm serdev bluetooth ecdh_generic nft_chain_nat_ipv6 ip6t_MASQUERADE nf_nat_ipv6 tun wireguard(O) ip6_udp_tunnel udp_tunnel xt_tcpudp nft_counter nft_chain_nat_ipv4 ipt_MASQUERADE nf_nat_ipv4 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables nfnetlink bridge stp llc vc4 drm_kms_helper snd_soc_core snd_compress v3d snd_pcm_dmaengine gpu_sched drm brcmfmac brcmutil snd_pcm snd_timer drm_panel_orientation_quirks snd sha256_generic syscopyarea sysfillrect sysimgblt fb_sys_fops cfg80211 raspberrypi_hwmon rfkill hwmon bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) videobuf2_dma_contig bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc
[216237.912116]  videobuf2_memops videobuf2_v4l2 videobuf2_common videodev vc_sm_cma(C) media rpivid_mem uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
[216237.912255] CPU: 3 PID: 29003 Comm: Chrome_ChildIOT Tainted: G         C O      4.19.75-v7l+ #1270
[216237.912263] Hardware name: BCM2835
[216237.912298] [<c0212d10>] (unwind_backtrace) from [<c020d530>] (show_stack+0x20/0x24)
[216237.912317] [<c020d530>] (show_stack) from [<c097fb20>] (dump_stack+0xd4/0x118)
[216237.912336] [<c097fb20>] (dump_stack) from [<c0222330>] (__warn+0x104/0x11c)
[216237.912354] [<c0222330>] (__warn) from [<c0222480>] (warn_slowpath_null+0x50/0x58)
[216237.912370] [<c0222480>] (warn_slowpath_null) from [<c0378418>] (apply_to_page_range+0x324/0x330)
[216237.912389] [<c0378418>] (apply_to_page_range) from [<c021930c>] (cma_allocator_free+0xa4/0xec)
[216237.912405] [<c021930c>] (cma_allocator_free) from [<c02183f8>] (__arm_dma_free+0x16c/0x17c)
[216237.912420] [<c02183f8>] (__arm_dma_free) from [<c0218478>] (arm_dma_free+0x34/0x3c)
[216237.912457] [<c0218478>] (arm_dma_free) from [<bf15c560>] (vc_sm_release_resource+0x158/0x198 [vc_sm_cma])
[216237.912505] [<bf15c560>] (vc_sm_release_resource [vc_sm_cma]) from [<bf15d1e4>] (vc_sm_dma_buf_release+0x54/0xa8 [vc_sm_cma])
[216237.912537] [<bf15d1e4>] (vc_sm_dma_buf_release [vc_sm_cma]) from [<c07065f4>] (dma_buf_release+0x60/0x164)
[216237.912558] [<c07065f4>] (dma_buf_release) from [<c03b1508>] (__fput+0xa4/0x1e8)
[216237.912575] [<c03b1508>] (__fput) from [<c03b16bc>] (____fput+0x18/0x1c)
[216237.912594] [<c03b16bc>] (____fput) from [<c02421ac>] (task_work_run+0xbc/0xe0)
[216237.912619] [<c02421ac>] (task_work_run) from [<c02339fc>] (get_signal+0x898/0x8cc)
[216237.912634] [<c02339fc>] (get_signal) from [<c020c68c>] (do_signal+0x84/0x474)
[216237.912648] [<c020c68c>] (do_signal) from [<c020cc88>] (do_work_pending+0xdc/0xf4)
[216237.912663] [<c020cc88>] (do_work_pending) from [<c0201034>] (slow_work_pending+0xc/0x20)
[216237.912672] Exception stack(0xda689fb0 to 0xda689ff8)
[216237.912683] 9fa0:                                     00000033 80404a5a ab358c50 4d534356
[216237.912696] 9fc0: 0540f8f5 00057000 00000033 00000036 ab358c50 b6f90968 ab358cec ab358f44
[216237.912707] 9fe0: b6e98150 ab358c34 b6e841d8 b55c5518 20000010 00000033
[216237.912718] ---[ end trace 88ec9ef38d6fbba0 ]---
[216237.912751] Unable to handle kernel NULL pointer dereference at virtual address 00000004
[216237.912773] pgd = 9e31e6bc
[216237.912787] [00000004] *pgd=18187003, *pmd=00000000
[216237.912836] Internal error: Oops: a06 [#1] SMP ARM
[216237.912850] Modules linked in: act_police sch_ingress cls_u32 sch_sfq sch_cbq xt_nat fuse ipt_REJECT nf_reject_ipv4 xt_conntrack rfcomm bnep hci_uart btbcm serdev bluetooth ecdh_generic nft_chain_nat_ipv6 ip6t_MASQUERADE nf_nat_ipv6 tun wireguard(O) ip6_udp_tunnel udp_tunnel xt_tcpudp nft_counter nft_chain_nat_ipv4 ipt_MASQUERADE nf_nat_ipv4 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables nfnetlink bridge stp llc vc4 drm_kms_helper snd_soc_core snd_compress v3d snd_pcm_dmaengine gpu_sched drm brcmfmac brcmutil snd_pcm snd_timer drm_panel_orientation_quirks snd sha256_generic syscopyarea sysfillrect sysimgblt fb_sys_fops cfg80211 raspberrypi_hwmon rfkill hwmon bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) videobuf2_dma_contig bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc
[216237.913325]  videobuf2_memops videobuf2_v4l2 videobuf2_common videodev vc_sm_cma(C) media rpivid_mem uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
[216237.913434] CPU: 3 PID: 29003 Comm: Chrome_ChildIOT Tainted: G        WC O      4.19.75-v7l+ #1270
[216237.913446] Hardware name: BCM2835
[216237.913473] PC is at vc_sm_release_resource+0x88/0x198 [vc_sm_cma]
[216237.913500] LR is at _cond_resched+0x30/0x50
[216237.913513] pc : [<bf15c490>]    lr : [<c0997e34>]    psr: 60000013
[216237.913526] sp : da689d70  ip : 00000100  fp : da689dac
[216237.913538] r10: e1880000  r9 : ef6c5b00  r8 : ef2a5410
[216237.913551] r7 : 00000000  r6 : c0a01454  r5 : bf162300  r4 : d61e5780
[216237.913563] r3 : 00000000  r2 : 00000000  r1 : 00000200  r0 : dc58c640
[216237.913578] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[... cut ...]
[216237.914173] [<c020cc88>] (do_work_pending) from [<c0201034>] (slow_work_pending+0xc/0x20)
[216237.914186] Exception stack(0xda689fb0 to 0xda689ff8)
[216237.914201] 9fa0:                                     00000033 80404a5a ab358c50 4d534356
[216237.914217] 9fc0: 0540f8f5 00057000 00000033 00000036 ab358c50 b6f90968 ab358cec ab358f44
[216237.914232] 9fe0: b6e98150 ab358c34 b6e841d8 b55c5518 20000010 00000033
[216237.914250] Code: e894000c e5950000 e3a0cc01 e3a01c02 (e5823004)
[216237.914270] ---[ end trace 88ec9ef38d6fbba1 ]---

fjleon1980, 2019-12-11:
after i rebooted on sunday and put the debug line, the whole device crashed and was unable to boot any longer until i reformatted the SD card. so i'm back at square one. left it playing videos overnight and in the early morning it froze again with the following stack:

[122757.048261] INFO: task kworker/2:4:29548 blocked for more than 120 seconds.
[122757.048275]       Tainted: G         C        4.19.75-v7l+ #1270
[122757.048285] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[122757.048296] kworker/2:4     D    0 29548      2 0x00000000
[122757.048339] Workqueue: events dbs_work_handler
[122757.048376] [<c0997424>] (__schedule) from [<c0997a94>] (schedule+0x50/0xa8)
[122757.048396] [<c0997a94>] (schedule) from [<c0997ef0>] (schedule_preempt_disabled+0x18/0x1c)
[122757.048417] [<c0997ef0>] (schedule_preempt_disabled) from [<c0998fa0>] (__mutex_lock.constprop.5+0x1a8/0x590)
[122757.048438] [<c0998fa0>] (__mutex_lock.constprop.5) from [<c09994a4>] (__mutex_lock_slowpath+0x1c/0x20)
[122757.048458] [<c09994a4>] (__mutex_lock_slowpath) from [<c0999504>] (mutex_lock+0x5c/0x60)
[122757.048478] [<c0999504>] (mutex_lock) from [<c08109d0>] (rpi_firmware_transaction+0x54/0xd0)
[122757.048498] [<c08109d0>] (rpi_firmware_transaction) from [<c0810b8c>] (rpi_firmware_property_list+0x140/0x2b0)
[122757.048514] [<c0810b8c>] (rpi_firmware_property_list) from [<c0810d78>] (rpi_firmware_property+0x7c/0xfc)
[122757.048534] [<c0810d78>] (rpi_firmware_property) from [<c07e7e74>] (bcm2835_cpufreq_clock_property.constprop.1+0x58/0x80)
[122757.048555] [<c07e7e74>] (bcm2835_cpufreq_clock_property.constprop.1) from [<c07e7ef8>] (bcm2835_cpufreq_driver_target_index+0x5c/0xe4)
[122757.048575] [<c07e7ef8>] (bcm2835_cpufreq_driver_target_index) from [<c07e2b54>] (__cpufreq_driver_target+0x2d8/0x5a8)
[122757.048593] [<c07e2b54>] (__cpufreq_driver_target) from [<c07e655c>] (od_dbs_update+0xa8/0x174)
[122757.048611] [<c07e655c>] (od_dbs_update) from [<c07e71bc>] (dbs_work_handler+0x3c/0x68)
[122757.048632] [<c07e71bc>] (dbs_work_handler) from [<c023db40>] (process_one_work+0x170/0x458)
[122757.048652] [<c023db40>] (process_one_work) from [<c023de84>] (worker_thread+0x5c/0x5a4)
[122757.048667] [<c023de84>] (worker_thread) from [<c0244170>] (kthread+0x138/0x168)
[122757.048684] [<c0244170>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[122757.048695] Exception stack(0xd8ec1fb0 to 0xd8ec1ff8)
[122757.048708] 1fa0:                                     00000000 00000000 00000000 00000000
[122757.048721] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[122757.048733] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[122757.048749] INFO: task kworker/1:3:32545 blocked for more than 120 seconds.
[122757.048760]       Tainted: G         C        4.19.75-v7l+ #1270
[122757.048769] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[122757.048778] kworker/1:3     D    0 32545      2 0x00000000
[122757.048814] Workqueue: events get_values_poll [raspberrypi_hwmon]
[122757.048843] [<c0997424>] (__schedule) from [<c0997a94>] (schedule+0x50/0xa8)
[122757.048862] [<c0997a94>] (schedule) from [<c099ba7c>] (schedule_timeout+0x200/0x428)
[122757.048881] [<c099ba7c>] (schedule_timeout) from [<c0998704>] (wait_for_common+0xd4/0x1b0)
[122757.048900] [<c0998704>] (wait_for_common) from [<c0998800>] (wait_for_completion+0x20/0x24)
[122757.048919] [<c0998800>] (wait_for_completion) from [<c08109f4>] (rpi_firmware_transaction+0x78/0xd0)
[122757.048936] [<c08109f4>] (rpi_firmware_transaction) from [<c0810b8c>] (rpi_firmware_property_list+0x140/0x2b0)
[122757.048952] [<c0810b8c>] (rpi_firmware_property_list) from [<c0810d78>] (rpi_firmware_property+0x7c/0xfc)
[122757.048973] [<c0810d78>] (rpi_firmware_property) from [<bf3360c0>] (get_values_poll+0x4c/0x15c [raspberrypi_hwmon])
[122757.049008] [<bf3360c0>] (get_values_poll [raspberrypi_hwmon]) from [<c023db40>] (process_one_work+0x170/0x458)
[122757.049025] [<c023db40>] (process_one_work) from [<c023de84>] (worker_thread+0x5c/0x5a4)
[122757.049041] [<c023de84>] (worker_thread) from [<c0244170>] (kthread+0x138/0x168)
[122757.049055] [<c0244170>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[122757.049066] Exception stack(0xd5e05fb0 to 0xd5e05ff8)
[122757.049077] 5fa0:                                     00000000 00000000 00000000 00000000
[122757.049090] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[122757.049102] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[122879.930212] INFO: task sandbox_ipc_thr:1908 blocked for more than 120 seconds.
[122879.930228]       Tainted: G         C        4.19.75-v7l+ #1270
[122879.930238] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[122879.930249] sandbox_ipc_thr D    0  1908   1718 0x00000000
[122879.930298] [<c0997424>] (__schedule) from [<c0997a94>] (schedule+0x50/0xa8)
[122879.930319] [<c0997a94>] (schedule) from [<c0997ef0>] (schedule_preempt_disabled+0x18/0x1c)
[122879.930341] [<c0997ef0>] (schedule_preempt_disabled) from [<c0998fa0>] (__mutex_lock.constprop.5+0x1a8/0x590)
[122879.930361] [<c0998fa0>] (__mutex_lock.constprop.5) from [<c09994a4>] (__mutex_lock_slowpath+0x1c/0x20)
[122879.930381] [<c09994a4>] (__mutex_lock_slowpath) from [<c0999504>] (mutex_lock+0x5c/0x60)
[122879.930402] [<c0999504>] (mutex_lock) from [<c08109d0>] (rpi_firmware_transaction+0x54/0xd0)
[... cut ...]
[123002.813615] [<c020cc88>] (do_work_pending) from [<c0201034>] (slow_work_pending+0xc/0x20)
[123002.813625] Exception stack(0xd37effb0 to 0xd37efff8)
[123002.813637] ffa0:                                     a7efc8b8 00000080 00000000 a7efc790
[123002.813650] ffc0: a7efc790 657a1ff7 a7efc8b8 000000f0 a7efc890 a7efc7f8 a7efc798 00000000
[123002.813662] ffe0: 00000000 a7efc750 00000000 b6aa9630 80000010 a7efc8b8

error seems completely different from last time. i wonder if i have a device with defective RAM or something?

fjleon1980, 2019-12-15:
@pelwell second time of the day it has crashed, so here's the log with debug enabled. Should i exchange the device? vcgencmd version does hang when ran. This is driving me nuts!

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
[TONS OF TIMES]
003260.282: arasan: arasan_emmc_open
003262.458: arasan: arasan_emmc_set_clock C0: 0x00000000 C1: 0x000ef447 emmc: 200000000 actual: 200000 div: 0x000001f4 target: 200000 min: 100000 max: 400000 delay: 10
003379.211: arasan: arasan_emmc_set_clock C0: 0x00000000 C1: 0x000ef447 emmc: 200000000 actual: 200000 div: 0x000001f4 target: 200000 min: 100000 max: 400000 delay: 10
003386.366: arasan: arasan_emmc_set_clock C0: 0x00000f00 C1: 0x000e01c7 emmc: 200000000 actual: 130039 div: 0x00000301 target: 130000 min: 100000 max: 400000 delay: 15
003429.635: arasan: arasan_emmc_set_clock C0: 0x00000f06 C1: 0x000e0307 emmc: 200000000 actual: 33333333 div: 0x00000003 target: 40000000 min: 0 max: 40000000 delay: 1
003722.175: brfs: File read: /mfs/sd/config.txt
003722.925: brfs: File read: 1750 bytes
003756.080: brfs: File read: /mfs/sd/config.txt
003957.280: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
003959.879: *** Restart logging
003959.922: brfs: File read: 1750 bytes
003963.707: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
003967.069: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
003967.104: HDMI0: hdmi_pixel_encoding: 300000000

003967.118: HDMI1: hdmi_pixel_encoding: 300000000

003968.072: brfs: File read: /mfs/sd/cmdline.txt
003968.149: Read command line from file 'cmdline.txt':
003968.190: 'dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=651219a0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles'
003972.766: brfs: File read: 187 bytes
003978.668: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
003978.709: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0x9e6f
004191.907: brfs: File read: 40559 bytes
004194.004: brfs: File read: /mfs/sd/config.txt
004194.494: dtparam: audio=on
004220.770: brfs: File read: 1750 bytes
004236.059: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d.dtbo
004275.464: Loaded overlay 'vc4-fkms-v3d'
005646.888: brfs: File read: 1255 bytes
006106.445: brfs: File read: /mfs/sd/kernel7l.img
006106.485: Loading 'kernel7l.img' to 0x8000 size 0x55e088
006106.526: Device tree loaded to 0x2eff5c00 (size 0xa3ac)
006108.897: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
008717.237: vchiq_core: vchiq_init_state: slot_zero = 0xded80000, is_master = 1
012040.135: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
012340.948: camsubs: Camera not found
[several times]
758173.060: mmal: mmal_ril_set_port_settings: ril.isp:in:0(OPQV): failed to set port definition (4) buffers 0/0/0
758173.283: mmal: mmal_ril_port_send: format not set on port 3f78fd00, ril.isp:in:0(OPQV)
[several times]

___________________ASSERT_______________
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
[TONS OF TIMES]
3359061.239: assert( module_state[hdmi_core].hdmi.initialised ) failed; ../../../../../middleware/hdmi/hdmi.c::hdmi_audio_supported line 1493 rev cd3add5
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
3403291.188: assert( module_state[hdmi_core].hdmi.initialised ) failed; ../../../../../middleware/hdmi/hdmi.c::hdmi_audio_supported line 1493 rev cd3add5
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
[several times]

pelwell, 2019-12-16:
@6by9 Does that MMAL message mean anything to you?

6by9, 2019-12-16:
AFAIK Not a log level that is enabled by default.
Something has tried to set an invalid port format on the input of an ISP component. In the older mode Chromium does use the ISP for format conversion from YUV to RGBA, and resize to fit the window (X then only has to do a blit rather than resize). I lose track of where John has got to with Chromium and VLC, but I thought he was now rendering through EGL rather than doing the blit. Is this system fully updated?

I think I now have VC offline memory dump analysis working again, so I'll dig out the runes for dumping that.

6by9, 2019-12-16:
Current runes for dumping VideoCore memory for analysis:
- ```cat /proc/cmdline```. The output will include entries ```vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000```. Please make a note of those numbers (you need them in the next step, and we need them for the analysis).
- ```sudo vcdbg save dump.bin 0 <value from vc_mem.mem_size>```. This will create a fill called dump.bin (it will be big!)
- ```gzip dump.bin```
- ```vcgencmd version``` and note the hash and date
- Push dump.bin to a suitable file sharing service. Feel free to make it a private link if there is a chance of sensitive data being in that dump.
- Post a link to the file, the values from /proc/cmdline, and the hash/date from```vcgencmd version```. (We can probably cope with just the file, but having the other bits makes life easier). If there is sensitive information, then please email us (we'll look at creating an alias for this, but otherwise info@raspberrypi.org tagged FAO Phil for now).

We will look at for automating this, but the relevant tools have only just been ressurrected and aren't that polished yet.

pelwell, 2019-12-16:
I can assure you that we have no interest in anything but the state of the firmware (memory allocations, thread states, etc.) at the time the dump was captured, and the entire content will be treated as confidential and deleted once we have extracted that state.

timg236, 2019-12-16:
"after i rebooted on sunday and put the debug line, the whole device crashed and was unable to boot any longer until i reformatted the SD card. so i'm back at square one. left it playing videos overnight and in the early morning it froze again with the following stack:"

There was a fixed related to video-freeze on 1st November. This is in rpi-update but there has not been an update to the apt firmware package since the 20th September.
https://github.com/raspberrypi/firmware/commit/b79618b5db4e4e7c02f9ad9d3ada51713825313e

fjleon1980, 2019-12-17:
@pelwell @6by9 @timg236 

My dump can be downloaded at https://drive.google.com/open?id=12t3fLuDNjduSCrhdN8MQM2c2-INkHQt4

coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M  smsc95xx.macaddr=DC:A6:32:07:92:91 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=651219a0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles

Sep 24 2019 17:35:42
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start_db)

If it helps reproduce this bug, i usually watch videos at https://hideout.tv with chromium. The problem seems to be random, sometimes i can go 1-2 days with a crash, sometimes it's in under an hour

6by9, 2019-12-23:
FYI I have downloaded this, but so far haven't managed to make up the appropriate binary to be able to fully analyse it. I need the firmware file without all the debugging info stripped out, and as we haven't had the tools up and running to do the analysis before, we haven't been routinely keeping those.
We're also about to hit the Christmas period, so it will probably have to wait until people are back in the office.

fjleon1980, 2019-12-23:
in a very weird turn of events, i haven't had this crash since i uploaded the files. the only thing i have changed is that i closed port 1194 on my firewall, which i was using for openvpn. i think it's just a coincidence. i'm still using it exactly the same

smnvhn, 2020-02-01:
Hi All, I am running a network of Raspberries 4 and have similar issue on all devices i.e. devices randomly freezing the system load is becoming 2.00 2.00 2.00. 2 processes are in uninterruptible sleep mode (kworkers). In this state reboot takes around 10 minutes. In this state ssh is working, however video playback is freezed (omxplayer).

Please help to sort out the issue. Here is a portion from my logs:
Dec  2 03:35:19 raspberrypi rngd[433]: stats: bits received from HRNG source: 1040064
Dec  2 03:35:19 raspberrypi rngd[433]: stats: bits sent to kernel pool: 993312
Dec  2 03:35:19 raspberrypi rngd[433]: stats: entropy added to kernel pool: 993312
Dec  2 03:35:19 raspberrypi rngd[433]: stats: FIPS 140-2 successes: 52
Dec  2 03:35:19 raspberrypi rngd[433]: stats: FIPS 140-2 failures: 0
Dec  2 03:35:19 raspberrypi rngd[433]: stats: FIPS 140-2(2001-10-10) Monobit: 0
Dec  2 03:35:19 raspberrypi rngd[433]: stats: FIPS 140-2(2001-10-10) Poker: 0
Dec  2 03:35:19 raspberrypi rngd[433]: stats: FIPS 140-2(2001-10-10) Runs: 0
Dec  2 03:35:19 raspberrypi rngd[433]: stats: FIPS 140-2(2001-10-10) Long run: 0
Dec  2 03:35:19 raspberrypi rngd[433]: stats: FIPS 140-2(2001-10-10) Continuous run: 0
Dec  2 03:35:19 raspberrypi rngd[433]: stats: HRNG source speed: (min=263.459; avg=523.346; max=547.923)Kibits/s
Dec  2 03:35:19 raspberrypi rngd[433]: stats: FIPS tests speed: (min=5.039; avg=12.262; max=31.217)Mibits/s
Dec  2 03:35:19 raspberrypi rngd[433]: stats: Lowest ready-buffers level: 2
Dec  2 03:35:19 raspberrypi rngd[433]: stats: Entropy starvations: 0
Dec  2 03:35:19 raspberrypi rngd[433]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us
Dec  2 03:35:35 raspberrypi kernel: [118825.067254] INFO: task kworker/3:1:15296 blocked for more than 120 seconds.
Dec  2 03:35:35 raspberrypi kernel: [118825.067269]       Tainted: G         C        4.19.75-v7l+ #1270
Dec  2 03:35:35 raspberrypi kernel: [118825.067279] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Dec  2 03:35:35 raspberrypi kernel: [118825.067290] kworker/3:1     D    0 15296      2 0x00000000
Dec  2 03:35:35 raspberrypi kernel: [118825.067336] Workqueue: events dbs_work_handler
Dec  2 03:35:35 raspberrypi kernel: [118825.067376] [<c0997424>] (__schedule) from [<c0997a94>] (schedule+0x50/0xa8)
Dec  2 03:35:35 raspberrypi kernel: [118825.067396] [<c0997a94>] (schedule) from [<c0997ef0>] (schedule_preempt_disabled+0x18/0x1c)
Dec  2 03:35:35 raspberrypi kernel: [118825.067417] [<c0997ef0>] (schedule_preempt_disabled) from [<c0998fa0>] (__mutex_lock.constprop.5+0x1a8/0x590)
Dec  2 03:35:35 raspberrypi kernel: [118825.067437] [<c0998fa0>] (__mutex_lock.constprop.5) from [<c09994a4>] (__mutex_lock_slowpath+0x1c/0x20)
Dec  2 03:35:35 raspberrypi kernel: [118825.067457] [<c09994a4>] (__mutex_lock_slowpath) from [<c0999504>] (mutex_lock+0x5c/0x60)
Dec  2 03:35:35 raspberrypi kernel: [118825.067480] [<c0999504>] (mutex_lock) from [<c08109d0>] (rpi_firmware_transaction+0x54/0xd0)
Dec  2 03:35:35 raspberrypi kernel: [118825.067501] [<c08109d0>] (rpi_firmware_transaction) from [<c0810b8c>] (rpi_firmware_property_list+0x140/0x2b0)
Dec  2 03:35:35 raspberrypi kernel: [118825.067519] [<c0810b8c>] (rpi_firmware_property_list) from [<c0810d78>] (rpi_firmware_property+0x7c/0xfc)
Dec  2 03:35:35 raspberrypi kernel: [118825.067540] [<c0810d78>] (rpi_firmware_property) from [<c07e7e74>] (bcm2835_cpufreq_clock_property.constprop.1+0x58/0x80)
Dec  2 03:35:35 raspberrypi kernel: [118825.067563] [<c07e7e74>] (bcm2835_cpufreq_clock_property.constprop.1) from [<c07e7ef8>] (bcm2835_cpufreq_driver_target_index+0x5c/0xe4)
Dec  2 03:35:35 raspberrypi kernel: [118825.067583] [<c07e7ef8>] (bcm2835_cpufreq_driver_target_index) from [<c07e2b54>] (__cpufreq_driver_target+0x2d8/0x5a8)
Dec  2 03:35:35 raspberrypi kernel: [118825.067602] [<c07e2b54>] (__cpufreq_driver_target) from [<c07e655c>] (od_dbs_update+0xa8/0x174)
Dec  2 03:35:35 raspberrypi kernel: [118825.067620] [<c07e655c>] (od_dbs_update) from [<c07e71bc>] (dbs_work_handler+0x3c/0x68)
Dec  2 03:35:35 raspberrypi kernel: [118825.067643] [<c07e71bc>] (dbs_work_handler) from [<c023db40>] (process_one_work+0x170/0x458)
Dec  2 03:35:35 raspberrypi kernel: [118825.067663] [<c023db40>] (process_one_work) from [<c023de84>] (worker_thread+0x5c/0x5a4)
Dec  2 03:35:35 raspberrypi kernel: [118825.067678] [<c023de84>] (worker_thread) from [<c0244170>] (kthread+0x138/0x168)
Dec  2 03:35:35 raspberrypi kernel: [118825.067696] [<c0244170>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
Dec  2 03:35:35 raspberrypi kernel: [118825.067707] Exception stack(0xef93ffb0 to 0xef93fff8)
Dec  2 03:35:35 raspberrypi kernel: [118825.067720] ffa0:                                     00000000 00000000 00000000 00000000
Dec  2 03:35:35 raspberrypi kernel: [118825.067733] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Dec  2 03:35:35 raspberrypi kernel: [118825.067745] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
Dec  2 03:35:35 raspberrypi kernel: [118825.067762] INFO: task kworker/0:1:15306 blocked for more than 120 seconds.
Dec  2 03:35:35 raspberrypi kernel: [118825.067772]       Tainted: G         C        4.19.75-v7l+ #1270
Dec  2 03:35:35 raspberrypi kernel: [118825.067781] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Dec  2 03:35:35 raspberrypi kernel: [118825.067791] kworker/0:1     D    0 15306      2 0x00000000
[... cut ...]
Dec  2 03:35:35 raspberrypi kernel: [118825.068087] [<c0244170>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
Dec  2 03:35:35 raspberrypi kernel: [118825.068098] Exception stack(0xda033fb0 to 0xda033ff8)
Dec  2 03:35:35 raspberrypi kernel: [118825.068111] 3fa0:                                     00000000 00000000 00000000 00000000
Dec  2 03:35:35 raspberrypi kernel: [118825.068125] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Dec  2 03:35:35 raspberrypi kernel: [118825.068136] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000

fjleon1980, 2020-02-01:
do you by any chance have hostapd (using your rpi as a wifi hotspot) installed? i may have disabled it when my issue disappeared.

timg236, 2020-02-02:
@smnvhn The logs dated Dec2 suggest that this is old firmware+kernel. A video related firmware hang was fixed a while ago (see previous comments). The fixes are in the apt packages now. Please try updating "apt update && apt upgrade" and see if the hang occurs again

fjleon1980, 2020-02-02:
my current kernel is 4.19.75-v8 and i don't want to touch it because it hasn't crashed again. i haven't upgraded any package so i don't truly understand why it hasn't crashed again (except for the two things i have stopped doing/changed)

smnvhn, 2020-02-02:
@timg236 thank you very much I will try.

charlescapon, 2020-03-04:
hi, in a python script we use raspistill to take picture, but  we have several pi 4 which randomly crash, the raspistill command hangs up and we have to reboot (which take 10 min) the PI to use the picamera again
here a portion of my log:
```
INFO: task kworker/1:2:3287 blocked for more than 120 seconds.
Feb 27 09:29:09 raspberrypi kernel: [ 2332.874798]       Tainted: G         C        4.19.93-v7l+ #1290
Feb 27 09:29:09 raspberrypi kernel: [ 2332.874807] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Feb 27 09:29:09 raspberrypi kernel: [ 2332.874816] kworker/1:2     D    0  3287      2 0x00000000
Feb 27 09:29:09 raspberrypi kernel: [ 2332.874849] Workqueue: events get_values_poll [raspberrypi_hwmon]
Feb 27 09:29:09 raspberrypi kernel: [ 2332.874874] [<c09c7f7c>] (__schedule) from [<c09c85ec>] (schedule+0x50/0xa8)
Feb 27 09:29:09 raspberrypi kernel: [ 2332.874894] [<c09c85ec>] (schedule) from [<c09c8a48>] (schedule_preempt_disabled+0x18/0x1c)
Feb 27 09:29:09 raspberrypi kernel: [ 2332.874912] [<c09c8a48>] (schedule_preempt_disabled) from [<c09c9af8>] (__mutex_lock.constprop.5+0x1a8/0x590)
Feb 27 09:29:09 raspberrypi kernel: [ 2332.874928] [<c09c9af8>] (__mutex_lock.constprop.5) from [<c09c9ffc>] (__mutex_lock_slowpath+0x1c/0x20)
Feb 27 09:29:09 raspberrypi kernel: [ 2332.874943] [<c09c9ffc>] (__mutex_lock_slowpath) from [<c09ca05c>] (mutex_lock+0x5c/0x60)
Feb 27 09:29:09 raspberrypi kernel: [ 2332.874959] [<c09ca05c>] (mutex_lock) from [<c083ff84>] (rpi_firmware_transaction+0x54/0xd0)
Feb 27 09:29:09 raspberrypi kernel: [ 2332.874977] [<c083ff84>] (rpi_firmware_transaction) from [<c0840140>] (rpi_firmware_property_list+0x140/0x2b0)
Feb 27 09:29:09 raspberrypi kernel: [ 2332.874993] [<c0840140>] (rpi_firmware_property_list) from [<c084032c>] (rpi_firmware_property+0x7c/0xfc)
Feb 27 09:29:09 raspberrypi kernel: [ 2332.875013] [<c084032c>] (rpi_firmware_property) from [<bf3320c0>] (get_values_poll+0x4c/0x15c [raspberrypi_hwmon])
Feb 27 09:29:09 raspberrypi kernel: [ 2332.875050] [<bf3320c0>] (get_values_poll [raspberrypi_hwmon]) from [<c023e028>] (process_one_work+0x170/0x458)
Feb 27 09:29:09 raspberrypi kernel: [ 2332.875067] [<c023e028>] (process_one_work) from [<c023e36c>] (worker_thread+0x5c/0x5a4)
Feb 27 09:29:09 raspberrypi kernel: [ 2332.875083] [<c023e36c>] (worker_thread) from [<c02446a0>] (kthread+0x138/0x168)
Feb 27 09:29:09 raspberrypi kernel: [ 2332.875097] [<c02446a0>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
Feb 27 09:29:09 raspberrypi kernel: [ 2332.875108] Exception stack(0xd80bffb0 to 0xd80bfff8)
Feb 27 09:29:09 raspberrypi kernel: [ 2332.875120] ffa0:                                     00000000 00000000 00000000 00000000
Feb 27 09:29:09 raspberrypi kernel: [ 2332.875132] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Feb 27 09:29:09 raspberrypi kernel: [ 2332.875144] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000  
```