raspberrypi/firmware #1301: Please add uart_2ndstage to documentation

mostanes, 2019-12-11:
Currently there is no mention of uart_2ndstage in the official documentation, only BOOT_UART. I only came upon it recently [in an issue here](https://github.com/raspberrypi/firmware/issues/1292#issuecomment-559085245). There are very scarce [google results](https://www.google.com/search?q=uart_2ndstage) on it, and none point to official documentation. This option is *very* useful for debugging early boot.

E.g.: I have to run an aarch64 kernel on the pi4 (virtualization), and, had I known this option a few days ago, it would have helped me tremendously find out what I was doing wrong.

JamesH65, 2020-02-24:
@pelwell @mostanes Happy to add this to the docs, what does it actually do?

mostanes, 2020-02-24:
Ugh, it's been a while since I've used it. If I still remember correctly, on the Pi4B, BOOT_UART only enables the UART output from the EEPROM bootcode, while uart_2ndstage enables the output from start.elf. This might apply to previous versions with bootcode.bin vs start.elf. I think @pelwell knows more.
From the user PoV: given that start.elf loads the kernel and the device tree, it is useful figuring out whether the kernel panics before being able to write anything to the UART or it just wasn't loaded by the firmware (particularly on aarch64, the kernel compilation process tends to build 2 versions, one EFI and one plain old ELF, while one of the 2 won't be loaded by the firmware - can't remember which one).

pelwell, 2020-02-26:
"uart_2ndstage=1" causes the second-stage loader (bootcode.bin), Pi 4 EEPROM and the main firmware (start*.elf) to output diagnostic information to UART0. Output is likely to interfere with Bluetooth operation unless disabled ("dtoverlay=disable-bt") or switched to the other UART ("dtoverlay=miniuart-bt"), and if the UART is accessed simultaneously to output from Linux then data loss could occur, so this is not a feature to activate except when trying to diagnose a loading problem.

JamesH65, 2020-02-26:
Worthwhile documenting, with appropriate warnings then I think. Will get round to it at some point!

timg236, 2020-02-26:
Yeah, I plan to do a fairly major Pi4 bootloader version when all the network boot stuff is released as stable. Probably in a few weeks time

JamesH65, 2020-02-26:
See #1471

cleverca22, 2020-02-29:
also, counter to what it does elsewhere, `uart_2ndstage=1` is required to make `recovery.bin` print debug, despite that being a 1st-stage file

pelwell, 2020-02-29:
Sorry, but the boot ROM is the first stage - recovery.bin, bootcode.bin and Pi 4 EEPROM all count as the second stage.

cleverca22, 2020-03-01:
but the EEPROM ignores `uart_2ndstage`, only `start4.elf` obeys it, in that boot path

when booting via SPI, it appears to be like this:
0th stage: mask rom
1st stage: SPI blob (debug with `BOOT_UART=1` in SPI)
2nd stage: `start4.elf` (debug with `uart_2ndstage=1`)

but when booting from SD, it appears like this:
0th stage: mask rom
1st stage: `recovery.bin` (debug with `uart_2ndstage=1`)

or do you just consider the recovery process as skipping the 1st stage entirely, and running the "2nd stage" with dram offline?
or is it that both the SPI blob and `start4.elf` are both considered 2nd stage and there is no difference between them?

pelwell, 2020-03-01:
Are you really explaining our naming scheme to us?

cleverca22, 2020-03-05:
more asking for clarification on why different stages seem to be called 2nd stage, some parts of the naming arent clear

pelwell, 2020-03-05:
It's clear to us.

cleverca22, 2020-03-05:
are the names of the stages documented publicly anywhere? could they be?

JamesH65, 2020-03-05:
I could add something to this page

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/bootflow.md

figozhang, 2020-05-17:
hi @JamesH65 @pelwell 

i have add "uart_2ndstage=1" in the config.txt for PI4.

Here is the log:
_Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
Loading recover4.elf hnd: 0x00000000
Failed to read recover4.elf error: 6
Loading recovery.elf hnd: 0x00000000
Failed to read recovery.elf error: 6
Loading start4.elf hnd: 0x000002d8
Loading fixup4.dat hnd: 0x00000594
MEM GPU: 76 ARM: 948 TOTAL: 1024
FIXUP src: 128 256 dst: 948 1024
Starting start4.elf @ 0xfec00200_

but, my question is how can i get more debug log of start4.elf?
when i using PI3, it can see such log:
MESS:00:00:02.593428:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:02.608463:0: Failed to load Device Tree file 'bcm2710-rpi-3-b.dtb'
MESS:00:00:02.614263:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:02.624294:0: brfs: File read: /mfs/sd/benos3.bin
MESS:00:00:02.628167:0: Loading 'benos3.bin' to 0x80000 size 0x5d0
MESS:00:00:02.634108:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:02.642963:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:02.649253:0: uart: Baud rate change done...
MESS:00:00:02.652682:0: uart: Baud 

my question, how i can see such log on PI 4?

lurch, 2021-06-30:
I guess this can be closed, given that https://github.com/raspberrypi/documentation/pull/1417 has been merged?

closed 2022-11-28 by timg236