raspberrypi/firmware #1302: omxplayer always stuck. Serge told me it looks like firmware issue and report it here

daimh, 2019-12-13:
Here is an email I sent to Serge.

I am omxplayer's loyal user as we have been using it for a few years to  play videos on TV as a signage. Our project has a demo at https://www.youtube.com/watch?v=7xs4GRC0qHc

    Recently I am testing Pi 4. It works great. The thing is omxplayer processes are always stuck after a few hours. I cannot even 'kill -9' the processes, and have to reboot it every time. We never had this issue in Pi 3 or older models. 

**To reproduce**
use omxplayer to keep playing multiple videos simultaneously, like the one shown in the above youtube link. 

[raspinfo.log](https://github.com/raspberrypi/firmware/files/3960943/raspinfo.log)
[daisplay-dmesg.log](https://github.com/raspberrypi/firmware/files/3960944/daisplay-dmesg.log)

**System**
raspinfo.log is attached

**Logs**
dmesg.log is attached

XECDesign, 2019-12-13:
I've seen this exact behaviour reported before, except in that case it would hang rpi-eeprom-update in the same way.

timg236, 2019-12-13:
If a VideoCore command is frozen then it will not be possible to kill the command. It's unlikely to be the same issue with rpi-eeprom-update and this, it's just different symptoms.
It's probably worth updating the firmware using apt upgrade or even rpi-update, there was a fix for frozen video since the September release

daimh, 2019-12-13:
Thanks!

"apt upgrade" was already up to date.

I am running rpi-update now and will report back in a few hours or a few days. :)

popcornmix, 2019-12-13:
Instructions to reproduce the hang would be useful.
I assume you could use a single video file with a script that launches 3 (?) copies of omxplayer with ``--win`` in loops.
If you could point to publicly available sample file (or upload a short video sample) and provide a script that provokes the hang we can take a look.

daimh, 2019-12-13:
The sample files are https://github.com/daimh/daisplay
Or you can wait a little bit. I already upgraded it to the latest firmware. It usually takes a day or two to reproduce this issue. I will report back unless a bus ran over me. :)

daimh, 2019-12-16:
It has been running smoothly for three days. I will keep it running and will report back.

popcornmix, 2019-12-17:
It sounds like your problem was fixed with this update; https://github.com/Hexxeh/rpi-firmware/commit/77444b020fe5b70f76d1921ae9c4cefd045d74bc

firmware: sysman: Fix unsafe check for h264 being enabled
See: https://github.com/popcornmix/omxplayer/issues/749

daimh, 2019-12-17:
@popcornmix  I think so too.
It has been running for four days without any issues. I will come back with a final report around New Year.

Happy Holidays!

daimh, 2020-01-02:
I think this issue is solved after firmware upgrade. omxplayer still plays all the videos fine after 20 days.

pi@raspberrypi:~ $ uptime
 08:22:20 up 19 days, 23:41,  2 users,  load average: 0.80, 0.54, 0.50

closed 2020-01-02 by daimh