raspberrypi/firmware #1304: Failed booting RPi 3 from USB (continued)

treble2019, 2019-12-15:
**Describe the bug**
RPi 3b does not boot with some USB HDD - especially large ones.

**To reproduce**
Enable the MSD using 
`echo program_usb_boot_mode=1 | sudo tee -a /boot/config.txt
`
Reference:
https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/msd.md

Works OK with a 1 TB HDD. Now use a 4 TB HDD from Intenso USB 3.0 (Standard-A USB 3.0 receptacle have a blue insert) from 4 years ago. This spinning drive was formatted as MBR (not GPT).
`scsi 0:0:0:0: Direct-Access     Intenso  External USB 3.0 1405
`
Does not boot RPi.

Now: I read couple of github issues like 
https://github.com/raspberrypi/firmware/issues/647
https://github.com/raspberrypi/firmware/issues/654

In the #647 it says just add files `bootcode.bin `and `timeout` to microSD card and then start it.
This does not solve the problem. RPi still does not boot. (This is not documented in the main documentation [1] but available at [2])

1. https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/msd.md
2.  https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/README.md#special-bootcodebin-only-boot-mode

**Work around**

I therefore copied all the files from `/boot` to a microSD card.

```
bcm2710-rpi-3-b-plus.dtb  fixup.dat
bcm2708-rpi-b-plus.dtb    bcm2710-rpi-3-b.dtb       kernel.img
bcm2708-rpi-b.dtb         bcm2710-rpi-cm3.dtb       kernel.img.md5
bcm2708-rpi-cm.dtb        bcm2711-rpi-4-b.dtb       overlays
bcm2708-rpi-zero-w.dtb    bootcode.bin              start.elf
bcm2708-rpi-zero.dtb      cmdline.txt               timeout
bcm2709-rpi-2-b.dtb       config.txt
bcm2710-rpi-2-b.dtb       distroconfig.txt
```

Edit `cmdline.txt `and change the UUIDs to match the `/boot and root`  of **spinning HDD**.
`boot=UUID=32C8-E1C1 disk=UUID=0c7erf32-39c5-33b5-8ca1-a9b063360d64 quiet
`
I can report it is **SUCCESSFUL**. Noise from starting spinning HDD indicates that it uses the HDD to boot - confirmed with ` df `command

**Expected behaviour**
Boot RPi with just only `bootcode.bin` and `timeout` in the microSD card.

**System**

* Raspberry Pi 3 Model B
* Buster
* b60349918206bf0ce2310de058cf7c5dd0958699 (clean) (release) (start_x)
* 4.19.83 #1

lurch, 2019-12-23:
Technically, your "workaround" is just a regular SD-card boot, and not a USB-boot (because `start.elf` and `kernel.img` are being loaded from the SD card and not the USB drive).
For more info, refer to:
 * https://www.raspberrypi.org/documentation/configuration/boot_folder.md
 * https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained
 * https://www.raspberrypi.org/blog/pi-3-booting-part-i-usb-mass-storage-boot/

And a possible problem with your workaround is that it loads `kernel.img` (the Linux kernel) from the SD card, but _if_ Linux thinks that `/boot` is on the USB drive, when you try to do a kernel-upgrade (e.g. through the regular `apt dist-upgrade` process), then it might update the kernel on the USB drive and not the one on the SD card. In extreme circumstances this can lead to an unbootable system (due to a kernel and module version mismatch).
But your workaround works fine as long as you've configured things so that Linux definitely uses the `/boot` on the SD card and not the `/boot` on the USB drive.  (I hope that makes sense and I've not confused you too much!!)

Are you able to do the "bootcode.bin UART enable" that [this page](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/) mentions, in order to grab the debug logs? I suspect it'll be hard to make any progress without that :confused:

treble2019, 2019-12-23:
> Technically, your "workaround" is just a regular SD-card boot
Agree. But all I want is a long life for my SD-card. so it is fine!

> Are you able to do the "bootcode.bin UART enable" that this page mentions, in order to grab the 

Not at the moment. I need to buy these cables.

>  I suspect it'll be hard to make any progress without that

Has your team tried any USB boot with bootcode.bin with a USB 3 external powered device (I mean 3.5 HDD)?

JamesH65, 2019-12-23:
> Not at the moment. I need to buy these cables.
>
>
> Has your team tried any USB boot with bootcode.bin with a USB 3 external powered device (I mean 3.5 HDD)?

Probably, but (almost) everyone is on Xmas break now, and the one person who I am pretty sure will know about this is out of contact until the New Year.

treble2019, 2020-01-02:
I have copied the latest `bootcode.bin `and a text file (without txt extension) called `timeout`. The `/boot ` folder is the repo-master.

Connected things as here https://learn.adafruit.com/adafruits-raspberry-pi-lesson-5-using-a-console-cable/test-and-configure

Connect cable
`sudo screen /dev/ttyUSB0 115200
`
The output is:

```
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 1
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Enabling PORT POWER on port 4
Enabling PORT POWER on port 5
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Found device on port 3
Found highspeed device
Device found: type = Mass storage, addr = 6
Trying booting from MSD device addr 6
Retrieving the config descriptor
Found ep_in = 00000001, ep_out = 00000002
GetMaxLun returned 0, max_lun = 0
Sending and retrying TEST_UNIT_READY
Sending SCSI_COMMAND_INQUIRY
Sending READ_CAPACITY
init_msd returned 0
USB device configuration set
Initialised mass storage, block size = 512, block count = 0
Trying booting from Ethernet device addr 5
Trying booting from Ethernet device addr 5
Trying booting from Ethernet device addr 5
Trying booting from Ethernet device addr 5

```

As I read many issues here, it is said that the delay in USB response causes problems. Therefore, after the above situation happens - (i.e) "forever printing Trying booting from Ethernet device addr 5" 

I did the following - quickly unplugged RPi power and reconnect after 2 seconds. Now the following happens. Yes it repeated the top lines. My idea was by now the USB has initialised and spinning - so more chances of it booting...

```
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 1
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Enabling PORT POWER on port 4
Enabling PORT POWER on port 5
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Found device on port 3
Found highspeed device
Device found: type = Mass storage, addr = 6
Trying booting from MSD device addr 6
Retrieving the config descriptor
Found ep_in = 00000001, ep_out = 00000002
GetMaxLun returned 0, max_lun = 0
Sending and retrying TEST_UNIT_READY
Sending SCSI_COMMAND_INQUIRY
Sending READ_CAPACITY
init_msd returned 0
USB device configuration set
Initialised mass storage, block size = 512, block count = 0
MBR read and decoded successfully
FAT32 partition decoded successfully
start.elf found in first partition on USB MSD, booting
Read File: config.txt, 1907 (bytes)

Raspberry Pi Bootcode
Read File: config.txt, 1907
Read File: start.elf, 2883108 (bytes)
Read File: fixup.dat, 6745 (bytes)
MESS:00:00:05.404098:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:05.607245:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:05.614679:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:05.620700:0: *** Restart logging
MESS:00:00:05.667630:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:05.692806:0: Failed to load Device Tree file '?'
MESS:00:00:05.709306:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:05.829466:0: No compatible kernel found
MESS:00:00:05.832567:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:10.642528:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined

```

Is there any hope? I

1. Is it possible to increase the timeout or wait for USB to larger value. 
2. Yes, I do agree vendors implement usb standard poorly but any workaround?

Thanks and wish you a very happy new year!

lurch, 2020-01-03:
Your first message said "Now use a 4 TB HDD" but your last message says "15627986944 512-byte logical blocks: (8.00 TB/7.28 TiB)" ? :confused:

treble2019, 2020-01-03:
> Your first message said "Now use a 4 TB HDD" but your last message says "15627986944 512-byte logical blocks: (8.00 TB/7.28 TiB)" ? confused

Unfortunately, I have it kept it as a off-site backup at my parents place in the new year and upgraded my device to 8TB. Unfortunately, the result is the same (does not boot)? :-(

lurch, 2020-01-03:
Ahh, fair enough.
In your second log output (where the boot process gets a bit further), it's interesting that it's able to read `config.txt`, `start.elf` and `fixup.dat` from your USB HDD, but then says `Failed to open command line file 'cmdline.txt'`. Could you paste the `ls -l` output for the FAT32 partition on your HDD when it's mounted on your Ubuntu PC?

pelwell, 2020-01-03:
The first three of those files are loaded by bootcode.bin and the fourth by start.elf, so I can see how different behaviour could arise.

treble2019, 2020-02-04:
sorry for the delayed reply.

```
ls /boot
bcm2708-rpi-b.dtb	bcm2710-rpi-2-b.dtb	  cmdline.txt	 fixup4x.dat   kernel7.img	 start4cd.elf  start.elf
bcm2708-rpi-b-plus.dtb	bcm2710-rpi-3-b.dtb	  config.txt	 fixup_cd.dat  kernel7l.img	 start4db.elf  start_x.elf
bcm2708-rpi-cm.dtb	bcm2710-rpi-3-b-plus.dtb  COPYING.linux  fixup.dat     kernel8.img	 start4.elf
bcm2708-rpi-zero.dtb	bcm2710-rpi-cm3.dtb	  fixup4cd.dat	 fixup_db.dat  kernel.img	 start4x.elf
bcm2708-rpi-zero-w.dtb	bcm2711-rpi-4-b.dtb	  fixup4.dat	 fixup_x.dat   LICENCE.broadcom  start_cd.elf
bcm2709-rpi-2-b.dtb	bootcode.bin		  fixup4db.dat	 issue.txt     overlays		 start_db.elf
```

Not sure what else I can do but I have gotten used to having the boot in a microSD.

closed 2020-04-13 by treble2019