raspberrypi/firmware #1306: geeekpi enc28j60 hat v1.02 error

brahmanggi, 2019-12-16:
hello, i cant activate enc28j60 on raspberry pi 0 w using raspbian buster lite,

 >pi@raspberrypi:~ $ dmesg | grep -e spi0.0 -e eth
[   20.570373] enc28j60 spi0.0: enc28j60 Ethernet driver 1.02 loaded
[   20.607136] enc28j60 spi0.0: enc28j60 chip not found
[   20.607258] enc28j60: probe of spi0.0 failed with error -5

while the driver loaded,

> pi@raspberrypi:~ $ lsmod
Module                  Size  Used by
bnep                   20480  2
hci_uart               40960  1
btbcm                  16384  1 hci_uart
serdev                 20480  1 hci_uart
bluetooth             405504  24 hci_uart,bnep,btbcm
ecdh_generic           28672  1 bluetooth
8021q                  32768  0
garp                   16384  1 8021q
stp                    16384  1 garp
llc                    16384  2 garp,stp
enc28j60               32768  0

in the /boot/overlays/ exist .dtbo, enc28j60-spi2.dtbo & enc28j60.dtbo in the /boot/config.txt already put 
>dtoverlay=enc28j60

is there any resolution to this?

thank you

pelwell, 2019-12-16:
Your wiring is almost certainly either incorrect (wires going to the wrong pins) or faulty (broken wires or poor connections). Post photos showing both ends of the wiring in clear detail and an overview of the whole setup.

brahmanggi, 2019-12-16:
> Your wiring is almost certainly either incorrect (wires going to the wrong pins) or faulty (broken wires or poor connections). Post photos showing both ends of the wiring in clear detail and an overview of the whole setup.

[image]

[image]

pelwell, 2019-12-16:
The HAT does appear to be connected correctly ((-8) but the blurry images do seem to show some slightly messy soldering around the 40-pin header. Also, unless the Zero W was bought with a fitted header (Zero WH) then that is another potential source of dry joints.

pelwell, 2019-12-16:
Apart from the hardware, the software configuration appears to be correct if the driver is loading. Are you using an adequate power supply?

brahmanggi, 2019-12-16:
> Apart from the hardware, the software configuration appears to be correct if the driver is loading. Are you using an adequate power supply?

I am using phone brick charger 5v 2a

sezeryalcin, 2021-04-12:
This geekpi zero ethernet header is not working with ZERO WIFI

[   22.827620] enc28j60 spi0.0: Ethernet driver 1.02 loaded
[   22.833576] enc28j60 spi0.0: chip not found
[   22.833693] enc28j60: probe of spi0.0 failed with error -5

pelwell, 2021-04-12:
Did it used to work? If so, using which software version?

sezeryalcin, 2021-04-12:
This is a module that I just acquired. I am trying to reach seller for support.
I believe this device works fine with NON-WIFI Zero modules. 
it says GEEKPI RPI Ethernet Hat

alemazzo95, 2021-06-09:
Hello, did the seller answer you? Did you resolve the issue? I bought the same hat on Amazon (https://www.amazon.it/gp/product/B07SPM137R/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) and I'm having the same issue. I'm running the last version of Raspberry Pi OS Lite (Release date: May 7th 2021, Kernel version: 5.10) on a Raspberry PI Zero W.

pi@raspberrypi:~ $ dmesg | grep -e spi0.0 -e eth
[   23.943016] enc28j60 spi0.0: Ethernet driver 1.02 loaded
[   23.992187] enc28j60 spi0.0: chip not found
[   23.992340] enc28j60: probe of spi0.0 failed with error -5

pi@raspberrypi:~ $ lsmod | grep enc28j60
enc28j60               24576  0

sezeryalcin, 2021-06-10:
It didn't work. Seller has no clue.
I think this is designed for non-wifi Zero and I have Zero W. Kernel module should be updated to make it compatible.

pelwell, 2021-06-10:
I've got one of these on order. It will take about a week to arrive, and it's not a priority, but I may be able to dig into the problem.

alemazzo95, 2021-06-10:
Hi all, yesterday I figured out what was wrong. The HAT arrived with two already-plugged jumpers. One jumper connects the two pins labeled "PUBLIC" with each others and the other jumper connects the two pins labeled "EN-CE0".
I have just re-plugged the jumpers connecting the first "PUBLIC" pin with first "EN-CE0" pin and second "PUBLIC" pin with second "EN-CE0" pin as in the picture below. After that the HAT started working correctly and eth0 compared in the list of available interfaces.

[image]

sezeryalcin, 2021-06-21:
So jumpers were placed wrong place? I tested this and eth0 is showing up but I will test ethernet performance first to make sure it's working.

pelwell, 2021-06-21:
The ordered unit arrived at the end of last week. It arrived with the jumpers in the apparently correct place - both bridging PUBLIC to EN_CE0 - and testing it now it appears to work. With them rotated 90 degrees (which to me appears a more logical configuration) causes the exactly the errors reported. I have yet to find any explanation of how the jumpers are supposed to work.

BETLOG, 2023-10-24:
For anyone finding it tedious that it randomises it's MAC every boot, and is using a pizero W (that has no existing eth0, but does have a wlan0 MAC): 

```
#!/bin/bash
# betlog - 2023-10-24--12-31-38
#
# geeekpi ENC28J60 - Raspberry Pi ZERO ENC28J60 Network Adapter Module PI0 PIW Compatible
# Randomised MAC is annoying; automatically set MAC to wlan0 MAC +1
# based on: https://wiki.dfrobot.com/Compute_Module_4_IoT_Router_Board_Mini_SKU_DFR0767
#
# /home/pi/set-mac.sh
#
# sudo nano /etc/rc.local
#   sh -c '/home/pi/set-mac.sh >/dev/null 2>&1 &' root
#
# sudo nano /boot/config.txt
#     dtoverlay=enc28j60
#
macHex=$(cat /sys/class/net/wlan0/address|tr -d ':')
[[ -z $macHex ]] && exit
macDec=$( printf "%lld\n" 0x$macHex )
macHex2=$(printf "%x" $(($macDec+1)) | sed 's/\(..\)/\1:/g;s/:$//' )
echo "wlan0: $(echo $macHex | sed 's/\(..\)/\1:/g;s/:$//' )"
echo "eth0 : $macHex2"
ip link set eth0 down
ip link set eth0 address $macHex2
ip link set eth0 up

```

samveen, 2024-02-27:
@BETLOG If you check the kernel command line you will see a MACADDR parameter passed to the kernel, which is passed for the wired mac address, but doesn't get used in the Zero family as the harware is missing. That can be utilized instead of reusing the brcm MAC. an example of the usage is available at https://github.com/samveen/PiZBMC/tree/master/ENC28J60

Jones14021, 2025-04-01:
I can confirm on my side that the orientation of the two jumpers was indeed the culprit for me. They came off by 90 degrees from the factory.