raspberrypi/firmware #1307: wrong memory on raspi 2 (only 116)

timsoer, 2019-12-17:
Hardware	: BCM2835
Revision	: a01041
Serial		: 00000000324d1604
Model		: Raspberry Pi 2 Model B Rev 1.1
#######################################
with start_x.elf
              total        used        free      shared  buff/cache   available
Mem:         116680       43256       34816         844       38608       29100
withoutstart_x.elf
              total        used        free      shared  buff/cache   available
Mem:         960488       61756      748272        6412      150460      840524

pelwell, 2019-12-17:
Memory inconsistencies like this are usually caused by an absent or non-matching fixup*.dat file. In your case the file in question is called `fixup_x.dat` and must be from the same firmware release as `start_x.elf`.

If `fixup_x.dat` is present and you believe it came from the same release, please post the output from the following commands:
```
$ strings /boot/start_x.elf | grep VC_BUILD
$ md5sum /boot/start_x.elf /boot/fixup_x.dat
```

timsoer, 2019-12-17:
strings /boot/start_x.elf | grep VC_BUILD
VC_BUILD_ID_USER: dom
VC_BUILD_ID_TIME: 15:15:38
VC_BUILD_ID_BRANCH: master
VC_BUILD_ID_VARIANT: start_x
VC_BUILD_ID_TIME: Dec 11 2019
VC_BUILD_ID_HOSTNAME: buildbot
VC_BUILD_ID_PLATFORM: raspberrypi_linux
VC_BUILD_ID_VERSION: 44e6357b14365c3029d973dc32956fc5730dc334 (clean)

md5sum /boot/start_x.elf /boot/fixup_x.dat
8ab191ce7c3b2ed13d7cc2b321af4178  /boot/start_x.elf
879f2f846f37c29e1819e6fd3d6f8151  /boot/fixup_x.dat

pelwell, 2019-12-17:
Those md5sums look correct, so something else is wrong (unless you are a NOOBs user and looking in the wrong partition). I get the expected 1GB of RAM with both a rev 1.0 2 and a rev 1.1 with a 2837 - the precise hardware variant should make no difference.

popcornmix, 2019-12-17:
@timsoer can you post contents of config.txt? Also output from:
```
vcgencmd get_mem arm
vcgencmd get_mem gpu
```

timsoer, 2019-12-17:
vcgencmd get_mem arm
arm=128M
vcgencmd get_mem gpu
gpu=128M

popcornmix, 2019-12-17:
config.txt? And are you using raspbian or NOOBS?

timsoer, 2019-12-17:
I use Raspbian Buster

timsoer, 2019-12-17:
Sorry I forgot the content of config.txt:

gpu_mem=64                                                                                               
start_file=start_x.elf
fixup_file=fixup_x.elf 
The rest is uncommented

As I mentioned:
Without start_x.elf and fixup_x.elf
I have no Probs but I can't watch webm mfiles

pelwell, 2019-12-17:
Try `fixup_file=fixup_x.dat`...

Or delete both `start_file=` and `fixup_file=` and use `start_x=1`.

timsoer, 2019-12-17:
Thank you very much 
Now I have my 1GB back and vids are working too.
But only with gpu_64
There is something weird.
I have a second raspberry pi model A with only 256 ram and gpu_mem=32
with which I have no problems with 720 vp8 videos

pelwell, 2019-12-17:
The extra features in the _x variant require more RAM. Which features do you require that aren't present in the normal start.elf?

timsoer, 2019-12-17:
With normal start.elf I can't play vp8 videos.
Since I make a update on my model 2 I need 64 gpu ram. Before 32 was fine.
My model A with only 256 ram still needs only 32 gpu ram. After update too.

timsoer, 2019-12-17:
Sorry my mistake:
With normal start.elf I CAN now play vp8 videos
But I need 64 gpu vs only 32 before the update

popcornmix, 2019-12-17:
What are you playing vp8 videos with? Kodi? omxplayer? Something else?

timsoer, 2019-12-17:
Buster without x and omxplayer

popcornmix, 2019-12-17:
You can't play VP8 in omxplayer without ``start_x=1`` (use that rather than ``start_file=``).

> But I need 64 gpu vs only 32 before the update

what was the update? If you can identify the specific rpi-update version that required more gpu memory we can look into it. Also can you identify the exact amount of extra gpu_mem required? (e.g. do you actually need 64M now, or would 33M have been enough).
There was a bug where the console framebuffer was unintentionally 16-bit. When that was fixed you would have the effect of 4M less gpu mem.

JamesH65, 2020-02-24:
@timsoer Were you able to figure out what was going on here? Can the issue be closed?

closed 2025-02-14 by popcornmix