raspberrypi/firmware #1309: Please add rtc-pcf85363 module

claplace, 2019-12-22:
Hi, can you add the rtc-pcf85363 module to the raspberrypi-firmware?

Overlay changes should look like this:
```diff --git a/arch/arm/boot/dts/overlays/i2c-rtc-overlay.dts b/arch/arm/boot/dts/overlays/i2c-rtc-overlay.dts
index af5ca042de75..35bf0a29a158 100644
--- a/arch/arm/boot/dts/overlays/i2c-rtc-overlay.dts
+++ b/arch/arm/boot/dts/overlays/i2c-rtc-overlay.dts
@@ -188,6 +188,21 @@
                };
        };

+       fragment@12 {
+               target = <&i2c_arm>;
+               __dormant__ {
+                       #address-cells = <1>;
+                       #size-cells = <0>;
+                       status = "okay";
+
+                       pcf85363@51 {
+                               compatible = "nxp,pcf85363";
+                               reg = <0x51>;
+                               status = "okay";
+                       };
+               };
+       };
+
        __overrides__ {
                abx80x = <0>,"+0";
                ds1307 = <0>,"+1";
@@ -201,6 +216,7 @@
                m41t62 = <0>,"+9";
                rv3028 = <0>,"+10";
                pcf2129 = <0>,"+11";
+               pcf85363 = <0>,"+12";

                addr = <&abx80x>, "reg:0",
                       <&ds1307>, "reg:0",```

Thanks!

pelwell, 2019-12-22:
Sure - see https://github.com/raspberrypi/linux/commit/882b0d9bee2f897feabc5fff65a5df1140af2ec2 and https://github.com/raspberrypi/linux/commit/b85f76a63d5f1b13220c61244469d55487db84f1.

pelwell, 2019-12-22:
It will take a while for those commits to make it into a Raspbian package - the updated overlay will be available via `rpi-update` first.

claplace, 2019-12-22:
Thank you very much :)

pelwell, 2020-01-10:
Please close if you are satisfied with the outcome.

claplace, 2020-01-10:
Sure, thanks

closed 2020-01-10 by claplace