raspberrypi/firmware #1311: Rpi4 eth_led0 set to 4, amber light stays on

LotusOttawa, 2019-12-30:
Latest Rasbian on Pi4, eth_led0 setting to 4 does not function as documented. Amber LED stays on.

JamesH65, 2019-12-30:
Can you point to the docs that are wrong?

LotusOttawa, 2019-12-31:
https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README line 126

JamesH65, 2020-01-01:
Just to confirm your changes are actually taking affect, do the other values do what you expect?

Thoulah, 2020-01-01:
Using this is `config.txt`:

```
dtparam=act_led_trigger=heartbeat
dtparam=pwr_led_trigger=mmc0
dtparam=eth_led0=4
dtparam=eth_led1=4
```

The eth leds are both active (green blinking, amber constant), the others 2 are behaving as expected from the config.

LotusOttawa, 2020-01-02:
Is there a logfile to know if the changes take effect?

act_led_trigger=cpu works as it should but pwr_led_trigger=cpu keeps the LED on constantly

pelwell, 2020-01-02:
This issue is about the eth_led parameters - please create a new issue for any other LED parameter problems (they are likely to have a different cause).

`sudo vcdbg log msg` displays the logging output from the firmware, which might give provide a clue in the event of an error.

Thoulah, 2020-01-02:
```
002892.686: dtparam: eth_led0=4
002905.456: Unknown dtparam 'eth_led0' - ignored
002905.548: dtparam: eth_led1=4
002918.307: Unknown dtparam 'eth_led1' - ignored
```

pelwell, 2020-01-02:
These parameters were only added in kernel 4.19.83 - if your kernel is older then you will have to update (or wait for the update to become available - I'm not sure which is the latest Raspbian kernel).

Thoulah, 2020-01-08:
Confirmed for me. Current kernel is version 4.19.75, upgrade to a version after .83 switched the lights off.

LotusOttawa, 2020-01-28:
I've now been able to upgrade to 4.19.93-v7l+ 

The eth_led0, eth_led1 Pi4 settings don't work as documented.

pelwell, 2020-01-28:
Show us your config.txt file.

LotusOttawa, 2020-01-28:
dtparam=eth_led0=4
dtparam=eth_led1=9

The comments are also hard to decipher (e.g. what's the difference between 0, 2, and 7?):

Pi4

                                0=Speed/Activity (default)       1=Speed
                                2=Speed/Flash activity   3=FDX
                                4=Off                    5=On
                                6=Alt                    7=Speed/Flash
                                8=Link                   9=Activity

pelwell, 2020-01-28:
With your settings the test Pi4 on my desk has the green LED off and the orange LED flashing with activity. What do you see?

> The comments are also hard to decipher (e.g. what's the difference between 0, 2, and 7?):

My understanding is:
* `Speed` means on for a fast link (probably 1Gb/s), otherwise off.
* `Activity` inverts the current state for a short period for every packet, so more traffic means more flicker.
* `Flash activity` is like `Activity` but without the proportionality - the pulses (which could be on or off depending on the other part of the mode) are a fixed width and the gaps are of (the same) minimum width, such that you see a clear flash pattern if there is activity and a steady state if not.

It looks like 2 is mislabelled and should just say `Flash activity`.

JamesH65, 2020-01-28:
The data we had on this was extremely sketchy - seems very likely I have got some terminology wrong in the docs.

pelwell, 2020-01-28:
The datasheet isn't at all clear.

LotusOttawa, 2020-01-28:
Ok, so according to pelwell's test, 'led_0' is green and 'led_1' is amber. Opposite to the 'usual' statement in the README file.

pelwell, 2020-01-28:
You haven't yet said what you observed, just that it didn't work "as documented". Are we talking about total failure or a few inconsistencies?

LotusOttawa, 2020-01-28:
Inconsistencies. I'll check in a few hours.

closed 2020-01-29 by LotusOttawa

pelwell, 2020-01-29:
The overlay README has been updated - see https://github.com/raspberrypi/linux/commit/f6e7d78b1560451a119d483f5e1b49d65ef4522e.