raspberrypi/firmware #1312: What is the purpose of libbrcmEGL/librcmGLESv2 libraries?

jpalus, 2020-01-01:
Sorry if I'm asking the question in wrong place but not really sure what would be the best place to do it. Please advise if there's a better one.

Recently I was debugging broken mpv which appeared to be caused by linking with both libEGL and libbrcmEGL. My question is what is the purpose of the latter? Is it somehow different than libEGL provided by platform? Is libbrcmEGL gonna replace libEGL completely? Why cause so much trouble to projects if they want to support both Mesa's libEGL and rpi's libbrcmEGL in one binary? It basically forces projects to introduce modular design and do quite a lot of gimmicks with linking to have common EGL/GLES code without being tied to single library.

cmitu, 2020-01-02:
The `libbrcm[EGL|GLES]` are the Broadcom GLES/EGL libraries, the ones without `brcm` are the MESA provided libraries. You can't mix and match the two because they're incompatible, you either use one or the other.
To use the MESA drivers, you'll have to enable the `(f)fkms` overlay in `config.txt` or use `raspi-config` to enable the MESA GL driver. Note that the Pi4 model has only support for the legacy GL drivers, you'll need to use the MESA drivers for it.

jpalus, 2020-01-02:
You mean to say that /opt/vc/lib/libEGL.so is MESA provided library? Because that's the one I was asking about. My point really is why not use /opt/vc/lib/libEGL.so which is preferred or why not use libglvnd to switch between the two?

cmitu, 2020-01-02:
@jpalus I have no `/opt/vc/lib/libEGL.so` on my system (Raspbian Buster), but if it would be (I think `rpi-update` used to add them ?), it would still be the part of the Broacdom legacy drivers. 

Your question(s) are better suited for the RPF forum, you should try asking there - https://www.raspberrypi.org/forums/viewforum.php?f=68.

XECDesign, 2020-01-06:
You can't use glvnd because the broadcom library is not compatible. It's not a drop-in alternative.

6by9, 2020-01-06:
> Note that the Pi4 model has only support for the legacy GL drivers, you'll need to use the MESA drivers for it.

Wrong way round. Pi4 has NO support for the legacy GL drivers, It only supports the MESA drivers.

libbrcmEGL uses the GPU firmware based GLES library on Pi0-3. It is not supported on Pi4 - use Mesa.

jpalus, 2020-01-06:
To somewhat rephrase my question -- I'd like to be able to ship a single mpv binary that can make use of proprietary rpi interface (if available and instructed to do so) as well as make use of Mesa stack otherwise. Now the most important issue is that rpi firmware apparently forces non standard libraries libbrcmEGL/libbrcmGLESv2 which make above goal impossible to achieve. One cannot link against both Mesa and libbrcm* stuff as symbols get completely messed up. That's why I asked about libglvnd as an alternative approach that in my opinion would work better.

On the other hand rpi firmware ships libEGL.so without soname number (there's no .1 at the end), so above goal is impossible anyway. This really seems like a simple thing that ie NVIDIA was doing right for years -- until libglvnd was invented it shipped libEGL.so.1 that was configured with higher priority in ld.so.conf, with libglvnd vendor's implementation is used on demand. Not sure why rpi firmware cannot choose one of these two solutions to make our lifes easier.

cmitu, 2020-01-06:
> Wrong way round. Pi4 has NO support for the legacy GL drivers, It only supports the MESA drivers.

Yes, I mistakenly phrased it the other way around.