raspberrypi/firmware #1313: Feature Request: Add config.txt option to locate device trees and overlays

bit-bash, 2020-01-03:
Not sure if this is the best place to post this, but here goes...

When playing with different kernel builds, it would be nice if I could specify (in config.txt) a subfolder in /boot where the *.dtb and overlays/*.dtbo file are located.

Since device trees and overlays are heavily tied to a kernel build, it would very handy if I could create a "kernel package" and write it to the /boot folder without necessarily overwriting the default/current kernel and device trees/overlays.

For example, in config.txt I could add a line like:
   include mykernel.txt

Now, in mykernel.txt, I could specify something like:
   kernel=test42-4.19.88/mykernel7.img
   **devicetrees**=test42-4.19.88

The idea would be that the *.dtb files would be in **/boot/test42-4.19.88** and the *.dtbo files would be in **/boot/test42-4.19.88/overlays**.  A new option, like "devicetrees" could tell the bootcode where to locate these files.

Of course, the kernel image could be anywhere as long as it has a unique file name.

I wouldn't need to worry about the bootcode.bin or *.elf, fixup*.dat files because they're not generally kernel build dependent.

If I #comment out the include mykernel.txt line in config.txt, then the system would boot the default settings and I wouldn't have to worry about re-copying the correct device tree and overlay files.

JamesH65, 2020-01-03:
There is the device_tree config option, which might help. https://www.raspberrypi.org/documentation/configuration/device-tree.md

JamesH65, 2020-01-03:
Actually, "overlay_prefix" might do exactly what you want.

pelwell, 2020-01-03:
You might prefer the (very) new `os_prefix` setting, which allows parallel OS installations (kernel, initramfs, dtbs and overlays) but a common firmware. This would remove the need for explicit `kernel=` and `devicetree=` settings.

One thing to watch with the prefixes - the firmware does not append a slash, so the chosen directory name should have a slash appended.

bit-bash, 2020-01-03:
Yes, **os_prefix** does seem to be the ticket as it handles ALL the files that would be bolted to the kernel - not just the overlays.

So for clarification, if I set:
os_prefix=mystuff/

Then the bootloader will use these files:
/boot/mystuff/kernel7.img
/boot/mystuff/cmdline.txt
/boot/mystuff/\*.dtb
/boot/mystuff/overlays/\*.dtbo

I did discover and read https://github.com/raspberrypi/linux/issues/3237 and noticed that if I specify "kernel=somekern.img" then the bootload will ignore os_prefix, is this true?  I'm wondering if "kernel=" can co-exist with os_prefix so that would give me the ability to either force the name of the kernel in the os_prefix folder or simply hardcode the kernel image filename but still honor the os_prefix setting.

pelwell, 2020-01-03:
The proposal at the start of #3237 was modified slightly by the end:

> After some internal discussion I changed the file location logic to make it simpler (i.e. more consistent) and more flexible. In the new scheme, os_prefix is applied to all OS files (kernels, cmdline.txt, dtbs, initramfs). It is also applied to overlay_prefix, provided ${os_prefix}${overlay_prefix}README exists.
>
> If upstream_kernel=1, the firmware sets os_prefix=upstream/ provided os_prefix hasn't already been set and it can locate the kernel or the dtb there (remember that we can't check for the existence of a directory).
>
> To allow the OS prefix to be skipped, all user-specified filenames may include a leading / character, making the paths absolute w.r.t. the boot directory/FAT partition.

I think that answers your questions and gives you the facilities you need.

bit-bash, 2020-01-04:
Thanks everyone for the info!
Turns out the new "os_prefix" option works exactly the way I want it to.
Installed the most recent "firmware" files (bootcode.bin, *.elf, *.dat) in /boot, set os_prefix to my desired folder name, installed all my kernel-specific files in the new folder, and it all worked awesome.

pelwell, 2020-01-10:
Please close if you are satisfied with the outcome.

closed 2020-01-10 by bit-bash