raspberrypi/firmware #1315: Latest LibreElec testbuild crashes when playing video

peno64, 2020-01-05:
I report this bug here on recommendation of @popcornmix

**Describe the bug**
When playing a video with the latest LibreElec 19 testbuilds, suddenly a crash appears and kodi restarts. This does not happen always and at random moments. However it happens often enough to reproduce.

**To reproduce**
Create a brand new fresh install of libreelec kodi 18.5 and then update with the latest libreelec 19 testbuild. I used #0104, 04-Jan-2020 on a rpi3
Play a movie and the crash occurs after some time. Sometimes a second try must be done.

**Expected behaviour**
No crash should happen

**Actual behaviour**
It crashes
The crahlog file contains:

Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x76f224d0 in memcpy () from /usr/lib/libarmmem-v7l.so

**System**
the raspinfo command does not exist on my system
* Which model of Raspberry Pi? Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)? cat /etc/rpi-issue gives no such file or directory... 
* Which firmware version (`vcgencmd version`)?
Dec 11 2019 15:15:38
Copyright (c) 2012 Broadcom
version 44e6357b14365c3029d973dc32956fc5730dc334 (clean) (release) (start_x)

* Which kernel version (`uname -a`)?
Linux LibreELEC 5.4.7 #3 SMP Sat Jan 4 21:03:00 GMT 2020 armv7l GNU/Linux

**Logs**
I include the crashlog file here: https://paste.kodi.tv/xajacecuda.kodi
This crashlog also includes the kodi.log at that moment.

**Additional context**
I have a complete image available from this setup. So you can restore the image on a micro sd card (min 8 gb) and boot with it and then goto video and play the video there.
Please provide me an email address to whom I can send a link of this image.
If you prefer the movie only I can also provide it. It is a movie I recorded myself. I prefer not to post these links here publically but if really needed I will do.

pelwell, 2020-01-05:
You can send me the link - phil@raspberrypi.org - and I'll pass it on.

peno64, 2020-01-05:
@pelwell 
Link to image send to email address.
Thank you

popcornmix, 2020-01-06:
Ping @6by9 This crash seems to be caused by https://github.com/raspberrypi/userland/pull/586

popcornmix, 2020-01-06:
LE image played video (72m) without crashing.
I'll try leaving it looping overnight.
I'll also try with a raspbian kodi build with debug enabled which will hopefully fail in the same way on this video file (and provide more information when it crashes).

popcornmix, 2020-01-14:
We reverted the suspected commit a few days ago (10 Jan). It should be in last few Milhouse nightly builds. Can you confirm if the issue is resolved.

peno64, 2020-01-14:
@popcornmix 
ok, trying with #0113, 13-Jan-2020 on a pi3. Will play movies all night long to see if it is ok.
Thank you

peno64, 2020-01-15:
Playing already 22 hours of movies without interruption and all goes well...

peno64, 2020-01-16:
Played 24 hours extra and no problem at all. So I think this was the issue

blganesh101, 2020-10-01:
Seen same issue with chromium playing video on raspberry pi4 and latest raspberry pi os available in the imager 

Logs as below 

Program terminated with signal SIGSEGV, Segmentation fault.
#0 0x76f224d0 in memcpy () from /usr/lib/libarmmem-v7l.so

Played video http://camendesign.com/code/video_for_everybody/test.html

Any easy fix for this?

popcornmix, 2020-10-06:
I don't see how this is related to issue reported here. Create a new issue if you have a different problem.