raspberrypi/firmware #1316: Pi3B+ with 2019-09-26-raspbian-buster-lite installed on SD card, and USB Stick inserted, won't boot up on reboot

dastagg, 2020-01-07:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.
Using a Pi3B+ with 2019-09-26-raspbian-buster-lite.img installed on a SD card.
If there is a USB Stick inserted and you reboot the RPI, it won't boot up.
HOWEVER:
Using a Pi3B+ with 2019-09-26-raspbian-buster.img installed on a SD card.
It works.

**To reproduce**
List the steps required to reproduce the issue.
Install 2019-09-26-raspbian-buster-lite on SD card.
Insert card
Do update/upgrade, raspi-config to expand.
reboot - This works
Shutdown and unplug the power.
Insert empty USB Stick.
Power up - it works and the USB is recognized.
reboot
RPi does not boot.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.
I expect the RPi to boot up 

**Actual behaviour**
Add a clear and concise description of what actually happened.
It doesn't boot up

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)?
2019-09-26-raspbian-buster-lite.img
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

pelwell, 2020-01-08:
After installing a clean Lite Buster image and booting (at which point the file system automatically expands), I plugged in a memory stick and rebooted without difficulty. An apt update and an upgrade didn't change this - it still boots and reboots with and without a memory stick inserted.

Have you tried a different memory stick? Is there anything else you might have changed? If not, please provide a clear, step-by-step set of instructions. While you are compiling this list, try and find the earliest point the image stops rebooting with the memory stick.

Also, when you get to the point that a reboot fails:
1) What pattern of flashes do you see on the green ACT LED?
2) Does it also fail to boot from cold, or is it just rebooting that doesn't work?
3) With the memory stick plugged into an already booted system, what does `lsblk` report?

dastagg, 2020-01-08:
Here is the complete walkthrough.

Just so you know, I have nine 3B+ and I ran into this problem on three of them.

I also thought I might have "a bad one" but the tests I ran today were on the third one. The other two were the cause of a "wasted weekend".

I also included the copy/paste info requested by the template.

From my perspective:

--- Boots up fine by itself

-- If the USB sticks are completely empty:
--- Will not either reboot or cold boot if there is a fat32 USB stick inserted
--- Will not reboot but will cold boot if there is a ext4 USB stick inserted

-- If the USB stick has data on it
--- It worked with a third USB stick with data on it
--- It worked with the ext4 USB stick with data on it once and then it didn't
--- It didn't work with the fat32 stick with data on it.
--- It did work fine with a 500GB mass storage drive

Sandisk 16GB SD card
2 32GB USB sticks:
    - One Partion formatted fat32
    - Two Partions formatted ext4 and swap

Install OS: 2019-09-26-raspbian-buster-lite.img:

- sudo dd bs=4M if=2019-09-26-raspbian-buster-lite.img of=/dev/mmcblk0 conv=fsync status=progress

- sync

- cd to boot partition, add an empty file named "ssh"

Turn on the Pi
Find out the IP address (nmap -sP 192.168.0.0/24)

ssh into it

Do the raspi-config:
    change the root password
    change the hostname
    update the localization
    pick a timezone
    expand the filesystem

reboot

ssh back in

sudo apt update
sudo apt upgrade

reboot

ssh back in

Here is the additional information requested by the template

pi@load2:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

pi@load2:~ $ vcgencmd version
Sep 24 2019 17:37:19
Copyright (c) 2012 Broadcom
version 6820edeee4ef3891b95fc01cf02a7abd7ca52f17 (clean) (release) (start)

pi@load2:~ $ uname -a
[... 107 lines cut ...]
And just for more kicks, I tried with a third USB stick that has data on it.

pi@load2:~ $ lsblk -p
NAME             MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
/dev/sda           8:0    1 29.2G  0 disk 
└─/dev/sda1        8:1    1 29.2G  0 part 
/dev/mmcblk0     179:0    0 14.9G  0 disk 
├─/dev/mmcblk0p1 179:1    0  256M  0 part /boot
└─/dev/mmcblk0p2 179:2    0 14.6G  0 part /

$ ssh pi@192.168.0.26
pi@192.168.0.26's password: 
Linux load2 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Jan  8 16:22:23 2020 from 192.168.0.11

As another experiment, I will put some data on the ext4 stick and see if that makes a difference

As another experiment, I will attach a 500GB mass storage device and see if that makes a difference.

dastagg, 2020-01-08:
I am coming to the conclusion that either the 2 USB sticks are bad or GParted didn't do the job correctly because every other USB stick or mass storage device I have attached, the 3B+ worked as it is supposed to.

closed 2020-01-08 by dastagg