raspberrypi/firmware #1317: Rpi 2b+ random freeze raspbian buster and nweset firmware

studiowena, 2020-01-10:
Hi

I have random freezes on raspbian buster and newest rpi firmware
i have Linux raspberrypi 4.19.93-v7+ #1286 SMP Mon Jan 6 13:18:44 GMT 2020 armv7l GNU/Linux

It happend random, sometimes with high cpu load sometimes without any high load. It seems to start after upgrading to raspbian buster my rpi 2b+ start freezes

After freeze i can see my console on screen but cannot log into with ssh, i dont use x server. Cannot do anything even with keyboard connected to rpi  - works only enter tap and then nothing happend.

Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW

Rpi 2b+
grep Revision /proc/cpuinfo
Revision        : a01041

* Which OS and version (`cat /etc/rpi-issue`)?

Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2

* Which firmware version (`vcgencmd version`)?

Jan  6 2020 13:05:25
Copyright (c) 2012 Broadcom
version f3e39fed4763ce4efbffabb2d4787aaacddeda0f (clean) (release) (start)

* Which kernel version (`uname -a`)?

Linux raspberrypi 4.19.93-v7+ #1286 SMP Mon Jan 6 13:18:44 GMT 2020 armv7l GNU/Linux

pelwell, 2020-01-10:
Have you tried reverting to an older firmware?

1. Pick a commit from the [Hexxeh/rpi-firmware repo](https://github.com/Hexxeh/rpi-firmware/commits/master).
2. Make a note of the commit hash (the 7-8 character hexadecimal number associated with each commit).
3. Run `sudo rpi-update <hash>`, substituting your chosen commit.

In general, random freezes are often power-related. What power supply are you using? Do you ever see under-voltage warnings? You could also try adding `over_voltage=2` to config.txt to see if that makes a difference.

studiowena, 2020-01-10:
didnt see any errors on dmesg or syslog, ill try Your suggestion and put on config over_voltage=2 and see what happends. Maybe it helps. If not ill try to revert to older firmware or buy another power supply to test with.

pelwell, 2020-01-10:
If it's a 2.5A known-good supply then unless you're also using it to power (for example) lots of USB peripherals then it should be fine. It's more likely that the voltage tuning algorithm (which gets updated from time to time) is just a bit marginal on your board.

studiowena, 2020-01-10:
im using external usb drive but with own power supply and only usb stick is connected to rpi so i dont know. For now im using over_voltage=2 and see what happend. Then i can buy another power supply to test with or i can revert to older firmware

pelwell, 2020-01-10:
I wouldn't recommend replacing the power supply - it's unlikely to fix the problem.

studiowena, 2020-01-10:
ok so ill try to revert to older firmware but dont know what version cuz some freezes starts from upgrading to raspbian buster and then firmware, so maybe if problem occur again i have to check firmware before buster release date and rever to this hash