raspberrypi/firmware #1318: Support two initramfs directives

alkisg, 2020-01-12:
Hi, most boot loaders like grub, pxelinux, ipxe etc support loading two initramfs files, concatenating the result.
In our ltsp.org netbooting software we use that to load the ltsp code in addition to the normal initramfs.

So our request is if you could please support this syntax in config.txt:

```shell
initramfs ltsp.img
initramfs initrd.img
```

Of course we do have a workaround for now, i.e. `cat ltsp.img initrd-xxx.img > initrd.img` to concatenate both files into a single one, but it's less flexible, it requires to run that command every time ltsp.img is updated.

pelwell, 2020-01-14:
Looking at the "loader" part of the firmware, this wouldn't be too difficult to do. The most painful part is likely to be the handling of the filenames. `initramfs` is a directive that takes two parameters - the filename and the load address (or "followkernel"). There is also the alternative method using two separate settings - `ramfsfile` and `ramfsaddr`, so we would need a mechanism that works for both scenarios.

The simplest change would be to support multiple filenames separated by commas. Users would have to be careful not to hit the 80+ character line length limit, but there's space to write something like:
```
initramfs quite_an_extensive_name.img,another_fairly_long_name.img followkernel
```
The other way of writing this is:
```
ramfsfile=quite_an_extensive_name.img,another_fairly_long_name.img
# ramfsaddr defaults to "followkernel" a.k.a. -1
```
If that sounds reasonable then I'll discuss the possibility of making such a change.

alkisg, 2020-01-14:
Yes of course, it would solve the problem perfectly.

Thank you very much.

pelwell, 2020-01-14:
This is now supported by the latest firmware, available via rpi-update (or copied directly from this repo).

alkisg, 2020-01-14:
You're awesome. :)

pelwell, 2020-01-15:
Please close the issue once you've had time to test the new firmware (assuming it works as you expect).

alkisg, 2020-01-15:
@pelwell, I ran the following:

```shell
rpi-update
 *** If no errors appeared, your firmware was successfully updated to f0fdb006abe402210fba5873c4328f9839c8480f
reboot
update-initramfs -k $(uname -r)
```

In my config, I used:

```shell
initramfs ltsp.img,initrd.img-4.19.95-v7+ followkernel
```

Yet in my dnsmasq logs I see:

```shell
file /srv/tftp/3e891756/ltsp.img,initrd.img-4.19.95-v7+ not found
```

Btw I was thinking that this change would be in bootcode.bin, but then I saw in https://github.com/raspberrypi/firmware/tree/master/boot that I was wrong (no new date for bootcode.bin), so I used an rpi3 without an SD card as a test.

Is my test above correct?

pelwell, 2020-01-15:
Boot the system (manually concatenate the ramfs file if you have to) then run `vcgencmd version` - I suspect you aren't running the updated version.

pelwell, 2020-01-15:
Doh - you're not running a Pi 4 are you? We're currently maintaining a separate source tree for Pi 4 (although we're hoping to merge soon...), and the patch only went into the Pi 4 tree. I'll get it back ported ASAP.

alkisg, 2020-01-15:
Thank you. I got:

```shell
root@raspberrypi:~# vcgencmd version
Jan 14 2020 15:03:12 
Copyright (c) 2012 Broadcom
version f3e39fed4763ce4efbffabb2d4787aaacddeda0f (clean) (release) (start)
```

Does rpi-update accept any parameters?
I didn't see any errors on the upgrade, but I'm running without an sd card, directly on NFS mounted rw root, hopefully that doesn't break anything.
I'm currently testing on an rpi3 without an SD card. I can test on other rpi devices if needed.

I'll see if manually downloading at least the /boot contents makes things work...

alkisg, 2020-01-15:
I used this command to download the firmware manually:

```shell
svn export --force https://github.com/raspberrypi/firmware/branches/master/boot .
```

This was exactly the same as my existing /boot dir, so I'm not sure what to try next.

pelwell, 2020-01-15:
See my previous comment:

> Doh - you're not running a Pi 4 are you? We're currently maintaining a separate source tree for Pi 4 (although we're hoping to merge soon...), and the patch only went into the Pi 4 tree. I'll get it back ported ASAP.

alkisg, 2020-01-15:
> and the patch only went into the Pi 4 tree

Sorry! I misread that part. My rpi4 is at another place; I could fetch it tomorrow if the backport isn't yet available.

popcornmix, 2020-01-16:
This should now be supported on Pi3 and earlier with rpi-update firmware.

alkisg, 2020-01-16:
Thank you, I will test tomorrow and report back.

alkisg, 2020-01-16:
Success! Specifically:

rpi-update:
> *** If no errors appeared, your firmware was successfully updated to e8da15f1ba0befe9d6dd1edb5a2d696900539d07

config.txt:
> initramfs ltsp/ltsp.img,initrd.img-4.19.95-v7+ followkernel

dnsmasq log:
> Jan 16 20:44:47 alkis dnsmasq-tftp[10936]: sent /srv/tftp/ltsp/ltsp.img to 192.168.67.142
> Jan 16 20:44:50 alkis dnsmasq-tftp[10936]: sent /srv/tftp/initrd.img-4.19.95-v7+ to 192.168.67.142

Thank you very much! Closing issue.

paul-1, 2020-02-08:
This was working nicely on all of my test rigs, until I tried to boot a pi3 from USB.  (No sdcard)  I could not see the whole kernel panic, but the last line was that it was unable to mount the root fs.

I quickly merged the 2 initrd's into one, and the device boots fine.

This was the line with 2 files on it....well under the 80 characters.
initramfs initrd_pcp_10.3.gz,initrd-4.19.102v7.gz followkernel

pelwell, 2020-02-10:
What are the sizes of those two files? Have you tried appending some padding to round them up?

paul-1, 2020-02-10:
initrd_pcp_10.3.gz - 3530454
initrd-4.19.102v7.gz - 1116675

How much should I add?  Is this a block boundry issue between the two files?

pelwell, 2020-02-10:
Start by adding two bytes to the end of initrd_pcp_10.3.gz and 1 to the end of initrd-4.19.102v7.gz so that they are both multiples of 4 in length.

paul-1, 2020-02-11:
Yes, that worked.
File sizes are 3530456 and 1116676, and it booted just fine.

pelwell, 2020-02-11:
Thanks - that's easy to fix in the firmware.

paul-1, 2020-02-13:
Let me know when you have something to test out.   I should have asked if you wanted a new issues, or if your good tracking in this one.

pelwell, 2020-02-13:
A patch is sitting in an internal Merge Request waiting for some attention, but you can download a trial version (just start.elf and fixup,dat) here: https://drive.google.com/file/d/10HQiONY995dmXqEdcwHE13IsDb0nxaOW/view?usp=sharing

I consider this an unfinished issue, so here is fine.

paul-1, 2020-02-14:
I replaced those two files and tried to boot with the original files.  And I still get the panic.  Then I remembered that since piCorePlayer is headless or just a framebuffer, we use gpu_mem=16, so we use the start_cd.elf.   I increased the gpu_mem setting such that start.elf is used, and then the system booted fine.

pelwell, 2020-02-14:
Thanks for the feedback. I'll get the patch merged as soon as possible.

paul-1, 2020-02-23:
Latest firmware seems to work fine on my systems now...both SD card and USB stick.  With files not rounded to 4 bytes.

pelwell, 2020-02-23:
Thanks - closing.

closed 2020-02-23 by pelwell

kirgene, 2020-05-26:
These changes break initramfs with LZ4 compression

pelwell, 2020-06-01:
FYI There's a test build of an updated firmware that removes the padding while still (I hope) supporting booting from USB. You can download it here: https://drive.google.com/file/d/1GQFhQPt0uNGi2iYvnc20VpjjtG9Ugo65/view?usp=sharing

paul-1, 2020-06-06:
Tested on a pi3b+ boot from microsd, and a 3B booting from USB

```
[    0.040052] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-06-01 17:02, variant start_cd
[    0.050041] raspberrypi-firmware soc:firmware: Firmware hash is 9ec86e54f459a2ed7cc32a028f10ca62581e7a16

```