raspberrypi/firmware #1319: i'm Using two usb Camera but return error

buYoung, 2020-01-13:
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
PI4 4G
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage4

* Which firmware version (`vcgencmd version`)?
39b2ae90606289b2b5d8ffba51f238892711eb4e
* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.50-v7l+ #901 SMP Fri Jun 21 18:33:08 BST 2019 armv7l GNU/Linux

issue: 
using program lang python3 / golnag 1.13.1 / using usb camera two interface 2.0 product sonnix
test firmware 4.19.50-v7l+ // 4.19.75-v7l+ // 4.19.50-v7l+ // 4.19.93 is same..

but firmware 4.19.50 (rasberry pi official site downloaded first version is safety, but.. firware downgrade is camera getting error)

python3 using library opencv 3.4.9, 4.1.0, 4.1.2, 4.2.0
golang using library v4l / opencv 4.2.0

golang is getting all camera but so slowly buffer (When importing an image from the camera, import it 3-5 times, stop and repeat)

python3 is getting one camera is safety,
However, if you open the second camera and get an image, the first camera stops.

and now getting error :
Python 3.7.3 (default, Apr 3 2019, 05:39:12)
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.

import cv2
Traceback (most recent call last):
File "", line 1, in
File "/usr/local/lib/python3.7/dist-packages/cv2/init.py", line 3, in
from .cv2 import *
ImportError: /usr/lib/arm-linux-gnueabihf/neon/vfp/libavcodec.so.58: undefined symbol: bcm_host_is_fkms_active

what is that? Searching there is no workaround.

pelwell, 2020-01-13:
> (rasberry pi official site downloaded first version is safety, but.. firware downgrade is camera getting error

I can't understand this comment - are you saying you downgraded because of a camera error?

`bcm_host_is_fkms_active` was added in the firmware update of Sep 20, which corresponds to a kernel version of 4.19.73. `sudo apt update && sudo apt upgrade` will install the latest kernel and firmware which should resolve the build error.

buYoung, 2020-01-15:
i'm sorry  The question was too difficult.
I just "tested on different firmware versions." I wrote.
I will try again.

I have the latest update with rpi-update.
I connected two usb cameras to the rpi board and installed opencv 4.1.2 to test.
But here's the problem.
It works fine without any errors when only one camera is turned on. However, when one more camera is turned on, one camera has a green screen and one camera is stopped.
It could be a camera problem, so even if you plug in another camera, the results are the same.

However, I used the following command to get green screen.
sudo rmmod uvcvideo
sudo modprobe uvcvideo nodrop = 1 timeout = 5000 quirks = 0x80
Without these two commands, the camera will "just stop".

In addition, "select time out" appears intermittently on the console screen.

Also, if the camera is "just stopped", the results of using the v4l2-ctl --list-device command are not listed.

JiangXL, 2020-02-28:
Hi, I have same problem with `bcm_host_is_fkms_active` in mpv. After upgrade to least, no this error again.

``` bash
$ mpv tv.m3u8 
mpv: symbol lookup error: /usr/lib/arm-linux-gnueabihf/neon/vfp/libavcodec.so.58: undefined symbol: bcm_host_is_fkms_active
```