raspberrypi/firmware #1321: Allow ${srv} in the kernel cmdline

alkisg, 2020-01-20:
Hi, please allow the `${srv}` keyword in the kernel cmdline, replacing it with the boot server IP.
For example, `nfsroot=${srv}:/srv/ltsp/raspbian` would be passed to the kernel as `nfsroot=192.168.67.1:/srv/ltsp/raspbian`.
This will allow netbooting Raspberries from a server with a dynamic IP, which is the default in [LTSP](https://ltsp.org).

Rationale.
A typical netbooting procedure on Raspberries is as follows:
 - The Pi does a DHCP request; a real and/or a proxy DHCP server might reply
 - Bootcode.bin, cmdline.txt, the kernel etc are fetched from TFTP
 - The kernel is loaded and the cmdline is passed to it

At that point, the DHCP offer packet contents are lost, so there's no way for the kernel to discover the proxyDHCP server and connect to it via nfsroot=. In iPXE (ipxe.org), this is done with the ${srv} variable. In PXELinux, IPAPPEND passes the server in the ip= variable. Raspberries should offer one of those solutions, and the one offered by iPXE is the most flexible.

Note that Raspberries by default do not use the initramfs, so we cannot develop and add programs that would discover the PXE server; this information needs to be passed to the kernel before any userland code has a chance to run.

Of course it's understandable that this solution wouldn't reach rpi3 ROMs and an SD card would be needed there. Thanks!

JamesH65, 2020-02-24:
@timg236 I can see this would be possible on the Pi4 (If we chose to implement it), but not sure about earlier models.

timg236, 2020-02-24:
Kernel command line setup is handled by start.elf which could use the TFTP server IP if someone wants to add it

JamesH65, 2020-02-24:
I'll take a look, just need to know where to get the IP address from.

JamesH65, 2020-02-29:
Bit more complicated than originally thought, might be a while as have other stuff to do.

alkisg, 2020-02-29:
No hurry; thank you for your efforts.