raspberrypi/firmware #1322: Unable to USB boot using a USB gadget emulation on RPi3 B+

ggardet, 2020-01-24:
I have a linux system which emulates a USB stick (g_mass_storage) and I want to boot my RPi3 B+ from this. But RPi3B+ refuses to boot it, whereas it can boot from a regular USB stick.

**Logs**
`bootcode.bin` (only) log:
```
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 4 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Enabling PORT POWER on port 4
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Hub device found at addr 5, enumerating HUB
Initialise hub
Found 3 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Waiting for devices to respond to reset
Found device on port 2
Found highspeed device
Device found: type = Ethernet adapter, addr = 6
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 7
Trying booting from Ethernet device addr 7
Initialise ethernet with MAC b8:27:eb:44:05:18
Wait for Link up
Link up
Sending DHCP request
Waiting for dhcp_reply
```

It seems that my USB emulation is seen as an Ethernet adapter instead of a USB mass storage.
Here are some useful info (`idVendor`/`idProduct`, etc.)  when USB mass storage is connected to a Linux host:
```
usb 3-1.3.2: New USB device found, idVendor=0525, idProduct=a4a5, bcdDevice= 5.04
usb 3-1.3.2: New USB device strings: Mfr=3, Product=4, SerialNumber=0
usb 3-1.3.2: Product: Mass Storage Gadget
usb 3-1.3.2: Manufacturer: Linux 5.4.10-1-default with 2184000.usb
```

I think `idVendor`/`idProduct` may be wrong in firmware.

ggardet, 2020-01-24:
I also tried to use fake `idVendor`/`idProduct` from a USB stick which is working with RPi3B+ to boot, but firmware still detect it as Ethernet.

pelwell, 2020-01-24:
The firmware is more likely to care about the device class (bDeviceClass) than the vendor and product info. What does `lsusb -v` report for the gadget? (Use `lsusb -t`) to find the bus and device number, then `lsusb -b -s <b>:<d>`, where `<b>` is replaced with the bus number and `<d>` is replaced with the device number.

ggardet, 2020-01-24:
`bDeviceClass` is `0`, but `bInterfaceClass` is `8 Mass Storage`.
Here is the full log of `lsusb -v -s <b>:<d>`:
```
Bus 003 Device 020: ID 0525:a4a5 Netchip Technology, Inc. Pocketbook Pro 903 / Mobius 2 Action Cam
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0525 Netchip Technology, Inc.
  idProduct          0xa4a5 Pocketbook Pro 903 / Mobius 2 Action Cam
  bcdDevice            5.04
  iManufacturer           3 
  iProduct                4 
  iSerial                 0 
  bNumConfigurations      1
OTG Descriptor:
  bLength                 3
  bDescriptorType         9
  bmAttributes         0x03
    SRP (Session Request Protocol)
    HNP (Host Negotiation Protocol)
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0023
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
```

pelwell, 2020-01-24:
For comparison, here's the output for an SD card reader:
```
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x0109 microSDXC Card Reader [Hama 00091047]
  bcdDevice            9.59
  iManufacturer           1
  iProduct                2
  iSerial                 3
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
```
Notice that your gadget has a different hierarchy:
```
Device Descriptor
OTG Descriptor
    Configuration Descriptor
        Interface Descriptor
            bInterfaceClass 8 Mass Storage
```
vs.
```
Device Descriptor
    Configuration Descriptor
        Interface Descriptor
            bInterfaceClass 8 Mass Storage
```
The Configuration Descriptor (and, in turn, the Interface Decriptor) is a child of an OTG Descriptor rather than Device Descriptor. I think this could be the cause of the problem.

ggardet, 2020-01-24:
> The Configuration Descriptor (and, in turn, the Interface Decriptor) is a child of an OTG Descriptor rather than Device Descriptor. I think this could be the cause of the problem.

Good catch. Could we get an update of Firmware to handle this case, please?

pelwell, 2020-01-24:
> Could we get an update of Firmware to handle this case, please?

There's a good chance you could, if that turns out to be the real cause.

ggardet, 2020-01-27:
Let me know if you have some binaries to test, I would be happy to test them.

ggardet, 2020-02-25:
Any update on this issue?

pelwell, 2020-02-25:
It's still on the list of things to look at, but it's been preempted by a number of more urgent tasks.

ghollingworth, 2020-05-01:
The only way to update this is to add an SD card to the device containing a bootcode.bin file (and nothing else).  Otherwise this code is only stored in the bootROM and as such is not fixable.  It might instead be worth seeing if you can move the OTG descriptor in the gadget emulation instead.

ggardet, 2020-05-04:
> The only way to update this is to add an SD card to the device containing a bootcode.bin file (and nothing else). Otherwise this code is only stored in the bootROM and as such is not fixable. It might instead be worth seeing if you can move the OTG descriptor in the gadget emulation instead.

I know we must have a bootcode.bin only file on uSD card.
Except if this is a bug, I do not think that OTG descriptor will be moved in the Linux kernel driver.

Vogtinator, 2020-05-12:
> Except if this is a bug, I do not think that OTG descriptor will be moved in the Linux kernel driver.

The kernel sends an OTG descriptor if it has a Mini-AB port: https://github.com/torvalds/linux/blob/4c6ef3b156c67e8867e04668cb2af902d44e4086/include/linux/usb/gadget.h#L348

It might be possible to override that using module or device tree parameters.