raspberrypi/firmware #1323: After upgrade to 000137ad USB-stability problems

robsy, 2020-01-24:
**Describe the bug**
after the upgrade to 000137ad (latest stable) I have the issue, that the USB-hub "disappears" after several hours.

[67672.041354] xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.
[67672.057405] xhci_hcd 0000:01:00.0: Host halt failed, -110
[67672.057414] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[67672.057463] xhci_hcd 0000:01:00.0: HC died; cleaning up
[67672.057558] usb 1-1: USB disconnect, device number 2
[67672.057577] usb 1-1.4: USB disconnect, device number 3
[67672.057895] ftdi_sio ttyUSB0: ftdi_set_termios FAILED to set databits/stopbits/parity
[67672.057916] ftdi_sio ttyUSB0: ftdi_set_termios urb failed to set baudrate
[67672.057928] ftdi_sio ttyUSB0: failed to set flow control: -19
[67672.060856] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[67672.060963] ftdi_sio 1-1.4:1.0: device disconnected

and lsusb:
root@xxxx:~ # lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

after a reboot (without any disconnecting of cables, only "reboot"):
root@xxxx:~ # lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**To reproduce**
wait.

**Expected behaviour**
USB-Hub stays on system

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi4 4GB

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

* Which firmware version (`vcgencmd version`)?
Jan 10 2020 16:52:25 
Copyright (c) 2012 Broadcom
version fdb5c37e330e7cb3027ac4fcc5b1cd5f244b351f (clean) (release) (start_cd)

* Which kernel version (`uname -a`)?
4.19.93-v8+

**Logs**
[67672.041354] xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.
[67672.057405] xhci_hcd 0000:01:00.0: Host halt failed, -110
[67672.057414] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[67672.057463] xhci_hcd 0000:01:00.0: HC died; cleaning up
[67672.057558] usb 1-1: USB disconnect, device number 2
[67672.057577] usb 1-1.4: USB disconnect, device number 3
[67672.057895] ftdi_sio ttyUSB0: ftdi_set_termios FAILED to set databits/stopbits/parity
[67672.057916] ftdi_sio ttyUSB0: ftdi_set_termios urb failed to set baudrate
[67672.057928] ftdi_sio ttyUSB0: failed to set flow control: -19
[67672.060856] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[67672.060963] ftdi_sio 1-1.4:1.0: device disconnected

**Additional context**
This now happend 4 times (between 16k and 67k after reboot), now I switch back to 000137ab, this version run for many week without any problems.

Any Hints or questions?

popcornmix, 2020-01-25:
Can you try adding ``over_voltage=1`` to config.txt?

robsy, 2020-01-26:
> Can you try adding over_voltage=1 to config.txt?

This is the solution.

The firmware was the false trace. 000137ab and 000137ad have similar behaviour.
But the kernel-update some days before makes the difference:
4.19.75-v8+ from raspberrypi-kernel_1.20190925-2_armhf.deb without over_voltage=1 and 000137ab/137ad: no problem
4.19.93-v8+ from raspberrypi-kernel_1.20200114-1_armhf.deb without over_voltage=1 and 000137ab/137ad: usb hub disappears after several hours
4.19.93-v8+ from raspberrypi-kernel_1.20200114-1_armhf.deb with over_voltage=1 and 000137ab/137ad: no problem

So for the "new" kernel the over_voltage=1 option seems to be a good idea...

Thanks and regards

P.S If I understood correctly, over_voltage=1 should give 1,2V to CPU/GPU. Gives the former kernel give a different voltage to the CPU by default?

P.P.S. If you give "xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command. raspberry" to google it shows that this issue is more or less common

popcornmix, 2020-01-27:
I suspect the issue is actually firmware rather than kernel (if it is fixed by over_voltage).

> P.S If I understood correctly, over_voltage=1 should give 1,2V to CPU/GPU. Gives the former kernel give a different voltage to the CPU by default?

over_voltage=1 adds 25mV to the default core voltage, rather than setting it to a fixed value.

closed 2020-01-29 by robsy