raspberrypi/firmware #1324: Raspberry pi 4 usb mouse and keyboard stop working after upgrade

mptimonen, 2020-01-26:
Raspberry pi 4 usb mouse and keyboard stop working after upgrade.

sudo apt update
sudo apt upgrade
=> mouse and keyboard stop working

can be reproduced

sudo rpi-update 993f475
4.19.97
=> mouse and keyboard stop working

and fixed

sudo rpi-update 77444b0
4.19.81
=> mouse and keyboard works ok

pelwell, 2020-01-26:
There are quite a few commits between the two you list - try and narrow it down to find the first failing commit (there's a list of commits with hashes at https://github.com/Hexxeh/rpi-firmware/commits/master). Once you have a pair of adjacent commits where one is bad and one good, use `sudo SKIP_KERNEL=1 rpi-update` to work out if the change was caused by the firmware or the kernel.

appelboom, 2020-01-27:
Similar issue for me. After upgrading to 000137ad (from 000137ab if I recall correctly ) the RPi4 looses WLAN and USB keyboard / mouse. Although it is booting /dev/sda2 which is SSD

Tried overvoltage fix from https://github.com/raspberrypi/firmware/issues/1323

mptimonen, 2020-01-27:
I think 4.19.84 is first failing version.

sudo rpi-update 16ce6121464d53a9eb9cf68afbdba7ada8ec7bcd
 4.19.84
=> mouse and keyboard stop working

I can't test SKIP_KERNEL=1 now. There is API call limitations...

sudo SKIP_KERNEL=1 rpi-update 16ce6121464d53a9eb9cf68afbdba7ada8ec7bcd
 *** Github API is currently rate limited - please try again after Mon 27 Jan 2020 06:45:56 PM EET

pelwell, 2020-01-27:
That's the commit that modified how voltages are controlled. Try adding `over_voltage=2` to config.txt to give it a bit more oomph.

mptimonen, 2020-01-27:
sudo SKIP_KERNEL=1 rpi-update 16ce6121464d53a9eb9cf68afbdba7ada8ec7bcd
=> works ok (after installing 4.19.83)

mptimonen, 2020-01-27:
sudo rpi-update 16ce6121464d53a9eb9cf68afbdba7ada8ec7bcd
4.19.84

and

config.txt 
over_voltage=2

=> it is working ok

pelwell, 2020-01-27:
I think you just got lucky that time you didn't update the firmware - you must have a marginal voltage (from the onboard SMPS - I'm not blaming your power supply), and there will be small variations between boots. You might even get away with dropping over_voltage to 1.

JamesH65, 2020-01-27:
Extensively covered on the forums - one thread is here. https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=262649

closed 2025-02-14 by popcornmix