raspberrypi/firmware #1325: Reproducibility, Auditability, Verifiability, Source Code

orangecms, 2020-01-27:
**Describe the bug**
I cannot reproduce the binaries, and neither verify nor audit them regarding correctness and security properties. And besides that, I cannot write my own firmware.

**To reproduce**
Look for source code in this repository.

**Expected behaviour**
Source code is available.

**Actual behaviour**
There are only binaries.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

- raspinfo does not apply

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
 I bought an RPi 4.

* Which OS and version (`cat /etc/rpi-issue`)?
not applicable

* Which firmware version (`vcgencmd version`)?
any version found in this repo

* Which kernel version (`uname -a`)?
does not apply

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

not applicable

**Additional context**
I bought the mentioned device believing that I could use it as I desire. That requires access to low levels such as hardware and firmware. It would be good to have full source code available or at least the necessary datasheets to write the code independent from vendors, without NDA or other blockers.

timg236, 2020-01-27:
The firmware is closed source. The binaries are in this repository.

pelwell, 2020-01-27:
We are attempting to reduce the scope of the firmware, moving more services to the ARM cores as they become more powerful, but closed source BLOBs are going to be with us for the foreseeable future. You have to decide whether you value full disclosure over value, performance and support.

JamesH65, 2020-02-07:
Closing this issue as questions answered/issue resolved.

closed 2020-02-07 by JamesH65