raspberrypi/firmware #1326: Kernel 4.19.93 - Built-in WIFI is gone on Raspbian

pandalatte, 2020-01-29:
Hi everyone!

I have a Raspbian Buster distro on Pi3, it's been working since November without any problems - both wifi and ethernet.
I use ethernet most of the time so only just 6-7 days ago I've noticed that WIFI is gone and fails to initialize.

**dmesg** shows a bunch of errors:

[    4.976639] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    4.981684] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    4.982646] usbcore: registered new interface driver brcmfmac
[    5.163658] random: crng init done
[    5.163670] random: 7 urandom warning(s) missed due to ratelimiting
[    5.262487] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[    5.262526] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[    5.266345] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[    5.266365] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[    5.271961] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[    5.271983] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[    5.377786] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    5.377892] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    5.378539] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[    6.655403] uart-pl011 3f201000.serial: no DMA platform data
[    6.905640] 8021q: 802.1Q VLAN Support v1.8
[    7.441457] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    7.485134] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    7.622810] brcmfmac: brcmf_cfg80211_reg_notifier: not an ISO3166 code (0x55 0x53)
[    8.455450] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    9.112115] Under-voltage detected! (0x00050005)
[    9.962630] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x45E1
[   12.040056] Bluetooth: Core ver 2.22
[   12.040114] NET: Registered protocol family 31
[   12.040118] Bluetooth: HCI device and connection manager initialized
[   12.040280] Bluetooth: HCI socket layer initialized
[   12.040288] Bluetooth: L2CAP socket layer initialized
[   12.040311] Bluetooth: SCO socket layer initialized
[   12.059484] Bluetooth: HCI UART driver ver 2.3
[   12.059493] Bluetooth: HCI UART protocol H4 registered
[   12.059571] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   12.064682] Bluetooth: HCI UART protocol Broadcom registered
[   12.292939] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.292946] Bluetooth: BNEP filters: protocol multicast
[   12.292959] Bluetooth: BNEP socket layer initialized
[   13.281967] Voltage normalised (0x00000000)
[   17.996427] fuse init (API version 7.27)
[   19.790268] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
[   20.746366] Bluetooth: RFCOMM TTY layer initialized
[   20.746393] Bluetooth: RFCOMM socket layer initialized
[   20.746417] Bluetooth: RFCOMM ver 1.11
[   22.311902] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   22.315307] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[   22.633427] EXT4-fs (mmcblk0p7): mounted filesystem with ordered data mode. Opts: (null)
[   24.871885] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout

[   24.875173] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle

[   24.875183] brcmfmac: brcmf_pno_clean: failed code -110

[   32.392066] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   32.395626] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[   32.395642] brcmfmac: brcmf_run_escan: error (-110)
[   32.395655] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[   35.911968] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   35.915484] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[   35.915500] brcmfmac: brcmf_run_escan: error (-110)
[   35.915513] brcmfmac: brcmf_cfg80211_scan: scan error (-110)

1. The hardware is fine, I can boot any other livecd (Kali, Mate) and it would work.
2. The power source is the same (1.5A per USB port I think), it would throw a yellow lightning bolt under a VERY VERY high load (4 cores with 100%), but very rarely.
3. Everything has been fine until recently.
4. I dont have auto-updates turned ON and I dont remember installing anything kernel-related in the past 2 weeks.
5. I dont think I should upgrade to the latest 4.19.97 right away if _this_ problem hasnt been specifically targeted, maybe the real cause is different?

So what could be the problem? Thanks!

pandalatte, 2020-01-29:
PS: Also tried with 2 different power sources (Apple, Samsung 2A fast-charging). The issue remains.

pelwell, 2020-01-29:
The hardware is not fine:
```
[ 9.112115] Under-voltage detected! (0x00050005)
```
systemd is highly parallel, and the first 10-20 seconds after power on can be the most demanding time. You need a 2.5A power supply for a Pi 3. Anything less and you run the risk of hardware malfunctions like this.

pandalatte, 2020-01-29:
I have connected a 2.5A charger. The under-voltage warning is gone.
But the problem stands. It has to be something else. I would probably reinstall Raspbian completely.

[image]

DrSchottky, 2020-02-14:
7.45.41.46 isn't the latest official fw for bcm43430a1 for some time now. 
Did you touch fw related stuff in any way?

JamesH65, 2020-02-24:
Can you try with a fresh install of the latest Raspbian to see if the issue is still present?

mjudeikis, 2020-08-01:
I see the same in 5.4.0-1008-raspi, BCM43430/1 7.45.98.38. Looks like not OS directly related.