raspberrypi/firmware #1328: Display trouble while installing Raspbian with NOOBS 3.3.0

neuralassembly, 2020-02-07:
When I try to install Raspbian to Pi 4 with NOOBS 3.3.0, 
I noticed that my display does not show anything.

When using with NOOBS 3.2.1, it has no troubles.

Therefore, I finished the installation using another display,
and then I re-connected Pi 4 to the display with trouble.

It still does not show the rainbow square, boot messages, etc.
However, after finishing the boot, it suddenly shows the desktop of Raspbian.

I think that there might be some trouble of the firmware of NOOBS.
Are there any solutions?

pelwell, 2020-02-07:
Please answer @procount's questions:
> What make & model of display, default resolution, maybe even the EDID file from the troublesome display?
> Did you try pressing key '2' in NOOBS to change to hdmi_safe mode?

And I'll add my own - have you tried the other HDMI port?

pelwell, 2020-02-07:
Does adding `hdmi_safe=1` to config.txt make a difference?

neuralassembly, 2020-02-07:
My display with trouble is as follows.

UPERFECT 7inch 1024x600@60hz
https://www.amazon.com/dp/B07CNNP9LN/

I summarize the situation.

*Pi 4 B with NOOBS 3.2.1

 No problems with 1024x600 resolution.

*Pi 4 with NOOBS 3.3.0

 NOOBS installation page is not shown.
 The rainbow square and boot sequence are not shown.
 Only the Raspbian desktop with 1024x600 resolution is shown.
 "hdmi_safe=1" makes the situation worse. Even the Raspian desktop is not shown.
 Two port (HDMI0 and HDMI1) give the same result.

*Pi 3 B+ with NOOB 3.2.1 and NOOBS 3.3.0

 No problems with 1366x768.
 I don't know why the resolution is different from that of Pi 4.