raspberrypi/firmware #1329: 4.19.97 Booting RPi 4 with rootfs on SSD on USB 3 hub fails

ralphrmartin, 2020-02-08:
A system which has been working perfectly for a long time, booting from an SD card with rootfs on a 1TB Sandisk Extreme SSD connected via a powered Startech USB 3 hub will no longer boot after using apt upgrade to move to kernel 4.19.97 today. Various USB related errors are shown during booting.

Swapping out the USB 3 hub with a powered USB 2 hub allows the machine to boot just fine.
Also, the USB-3 hub appears to still work fine when tested on a PC. 

I conclude that something has changed USB-3 -wise in the 4.19.97 kernel to cause this breakage.

I notice at least one other person has reported a similar issue on the forum.

popcornmix, 2020-02-08:
Can you try adding over_voltage=1 to config.txt. If that doesn’t work try 2.

ralphrmartin, 2020-02-08:
I have just tried a different brand - UGREEN - of powered USB 3 hub, and it works OK without using over_voltage=1 .

With the original hub, using over_voltage=1  also fixed the issue.

Hopefully this can be fixed in the next release.

peetereczek, 2020-02-08:
> I have just tried a different brand - UGREEN - of powered USB 3 hub, and it works OK without using over_voltage=1 .
> 
> With the original hub, using over_voltage=1 also fixed the issue.
> 
> Hopefully this can be fixed in the next release.

This means that your 1TB Sandisk Extreme SSD consumes too much energy to start. It's hardware issue of underpower and can't be fixed with just software. On Pi4's USB you can have maximum 1.2A current (total shared between 4 ports). I would say that your disk consumes more for startup

ralphrmartin, 2020-02-08:
First, note that this setup has been rebooted many (hundreds? of) times successfully, with earlier versions of the kernel etc.  (It also worked fine when plugging in a 2nd SSD for backup).

Second, I dont believe that  my 1TB Sandisk Extreme SSD "consumes too much energy to start". Even when the self-powered hub is already powered up (with a beefy 4A external power supply), so the SSD is already powered, the Pi wont boot.  This is not an HD with a motor spinning up, and it is not drawing power from the Pi anyway.

As it worked very reliably before, I believe a recent changed in the kernel has caused this. 
(Of course, the SSD or the hub may be out of spec - but, it did work reliably before...)

Also note that even now, it boots OK with a different hub (so maybe its the hub which is out of spec - but Startech generally are a well thought of brand).

bcutter, 2020-02-09:
I see strange behavior with my USB devices attached directly or using USB hubs to the Pi 4 since the last kernel update. In fact mounted flashdrives are unmounted from time to time and even the whole system is being rebooted about 3 times a day. Before that kernel update there were NO usb relevant incidents and the system was running for 80 days without any issues. There must have been some changes in the kernel.

JamesH65, 2020-02-10:
> I see strange behavior with my USB devices attached directly or using USB hubs to the Pi 4 since the last kernel update. In fact mounted flashdrives are unmounted from time to time and even the [...]

Sounds like a USB power problem. Try reducing any overclocking, but retain overvoltage of 1 or 2.

bcutter, 2020-02-10:
> Sounds like a USB power problem. Try reducing any overclocking, but retain overvoltage of 1 or 2.

Thanks, I´m working on that (finding new safe overclocking values), see https://github.com/raspberrypi/firmware/issues/1331#issuecomment-584051726.

Sorry for misusing this issue.

plee82, 2020-02-10:
Same issue here. RPi 4 with rootfs on USB 3 HDD is not booting with the latest kernel.

ralphrmartin, 2020-02-27:
Just updated to the mid-feb relaease, ands it has stopped working again, with the Startech hub - even with over_voltage=2.

I can confirm that with a UGreen hub, everything is OK.

Something has definitely changed, as I was using the Startech hub successfully for ages.

tyjtyj, 2020-03-19:
Does unplugging the hub and re plug during power up fixed the issue? 

It could be powered hub is back powering usb 3.0 port. Mention here 
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245218

ralphrmartin, 2020-03-19:
No idea. I need the pi to restart by itself after a power cut. It used to do that.

crberger, 2020-04-18:
Trying to boot my RPi3B+ from an mSATA SSD off of a JMicron Technology adapter.  Using the bootcode.bin UART enable feature to monitor USB boot progress it looks like my RPi3b+ is able to find the MBR on my mSATA SSD and then finds and loads the u-boot.bin file just as on my working system with the Orico SATA enclosure / 120GB Crucial SATA SSD. Here are the last several lines of the boot sequence I captured:

> MESS:00:00:05.479269:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:05.563330:0: brfs: File read: /mfs/sd/bcm2710-rpi-3-b-plus.dtb
MESS:00:00:05.568428:0: Loading 'bcm2710-rpi-3-b-plus.dtb' to 0x100 size 0x69ae
MESS:00:00:05.796287:0: brfs: File read: 27054 bytes
MESS:00:00:05.855613:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.860106:0: dtparam: audio=on
MESS:00:00:05.894513:0: brfs: File read: 1755 bytes
MESS:00:00:05.941117:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:05.945082:0: Read command line from file 'cmdline.txt':
MESS:00:00:05.950972:0: 'dwc_otg.lpm_enable=0 console=tty1'
MESS:00:00:07.900530:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:08.071499:0: brfs: File read: 34 bytes
MESS:00:00:08.322569:0: brfs: File read: /mfs/sd/u-boot.bin
MESS:00:00:08.326448:0: Loading 'u-boot.bin' to 0x8000 size 0x5fab4
MESS:00:00:08.332445:0: Device tree loaded to 0x2eff9200 (size 0x6d91)
MESS:00:00:08.340073:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:08.346394:0: uart: Baud rate change done...
MESS:00:00:08.349826:0: uart: Baud rate

> 
Can anyone please help me figure out how to dig further and find out what the problem is?  I don't think it is power.  Unless I'm wrong, the device reports that MaxPower as only 224mA using lsusb -v.  Of course the KingSpec SSD will consume power as well, but I've had another mSATA adapter working before with the same exact KingSpec SSD and that mSATA adapter consumed even more power than the one I am currently trying to get working.  If there are any other tools I can use to debug, I'd be grateful for someone pointing me in that direction.

timg236, 2022-11-28:
Closing bug due to old kernel version 4.19.97

closed 2022-11-28 by timg236