raspberrypi/firmware #1330: Am I blocked or something?

persianpros, 2020-02-09:
Hello,

Would you please check your blocklist for my account? We at Open Vision are working on enigma2 for RPi and I can't see any of your repos which I don't know why.

https://github.com/OpenVisionE2/meta-raspberrypi

pelwell, 2020-02-09:
You can see this repo. What happens when you visit https://github.com/raspberrypi/linux/ ?

persianpros, 2020-02-09:
I can see it but when I browse https://github.com/raspberrypi I see only 6 pinned repos and below is empty, is it ok?

pelwell, 2020-02-09:
It's sounds like you are on the mobile site. Click the Repositories tab at the top of the page, or switch to the desktop version.

I'm pretty sure you aren't blocked, but reopen this issue if you find something specific that you can't do

closed 2020-02-09 by pelwell

persianpros, 2020-02-09:
I'm using Firefox on Ubuntu 18.04 and I'm on the repositories tab but still I can see only pinned repos that's all!

pelwell, 2020-02-09:
Try with the Firefox version of Incognito mode.

persianpros, 2020-02-09:
It works but the moment I login everything will be gone. That's I'm trying to tell.

pelwell, 2020-02-09:
If you can create an issue then you're not blocked by our organisation.

https://github.com/raspberrypi/usbboot is not one of the 6 pinned repos - can you view it?

pelwell, 2020-02-10:
I've asked an admin and I can confirm that you aren't blocked - there must be something wrong with the settings on your browser or GitHub account.