raspberrypi/firmware #1332: Option gpu_mem not effective when not placed in config.txt but included extraconfig.txt

gvolt, 2020-02-10:
**Describe the bug**

As stated in the documentation (https://www.raspberrypi.org/documentation/configuration/config-txt/misc.md) the <include> option "Causes the content of the specified file to be inserted into the current file.".

This currently does not work for the gpu_mem* options: Setting gpu_mem not in /boot/config.txt but in a file that is "linked" by the <include> option in /boot/config.txt has no effect.

**To reproduce**
1. Remove any gpu_mem* settings from /boot/config.txt.
2. Add <include extraconfig.txt> to /boot/config.txt.
3. Create /boot/extraconfig.txt.
4. Add <gpu_mem=32> to /boot/extraconfig.txt.

**Expected behaviour**
GPU memory should be set to the value given by the gpu_mem option in the file referenced by the <include> option in /boot/config.txt.

In the example GPU memory should be: 32MB

**Actual behaviour**
The gpu_mem setting is not obeyed if it is placed in a file that is referenced by the <include> option in /boot/config.txt.

In the example on a RPi3B <vcgencmg get_mem gpu> shows a GPU memory of 76MB instead of 32MB.

**System**
Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

    Which model of Raspberry Pi?
      Pi3B+ v1.3

    Which OS and version (cat /etc/rpi-issue)?
      Raspbian GNU/Linux 9 \n \l

      => 2019-04-08-raspbian-stretch-lite

    Which firmware version (vcgencmd version)?
      Aug 15 2019 12:06:42
      Copyright (c) 2012 Broadcom
      version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start)

    Which kernel version (uname -a)?
      Linux xxxx 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

**Logs**
sudo vcdbg log msg
001138.687: brfs: File read: /mfs/sd/config.txt
001138.840: brfs: File read: 1625 bytes
001139.534: brfs: File read: /mfs/sd/extraconfig.txt
001140.422: brfs: File read: 11 bytes
001150.945: HDMI:EDID error reading EDID block 0 attempt 0
001152.251: HDMI:EDID error reading EDID block 0 attempt 1
001153.549: HDMI:EDID error reading EDID block 0 attempt 2
001154.847: HDMI:EDID error reading EDID block 0 attempt 3
001156.145: HDMI:EDID error reading EDID block 0 attempt 4
001157.443: HDMI:EDID error reading EDID block 0 attempt 5
001158.741: HDMI:EDID error reading EDID block 0 attempt 6
001160.039: HDMI:EDID error reading EDID block 0 attempt 7
001161.337: HDMI:EDID error reading EDID block 0 attempt 8
001162.635: HDMI:EDID error reading EDID block 0 attempt 9
001163.697: HDMI:EDID giving up on reading EDID block 0
001164.058: HDMI:EDID error reading EDID block 0 attempt 0
001165.353: HDMI:EDID error reading EDID block 0 attempt 1
001166.646: HDMI:EDID error reading EDID block 0 attempt 2
001167.941: HDMI:EDID error reading EDID block 0 attempt 3
001169.234: HDMI:EDID error reading EDID block 0 attempt 4
001170.529: HDMI:EDID error reading EDID block 0 attempt 5
001171.823: HDMI:EDID error reading EDID block 0 attempt 6
001173.117: HDMI:EDID error reading EDID block 0 attempt 7
001174.410: HDMI:EDID error reading EDID block 0 attempt 8
001175.705: HDMI:EDID error reading EDID block 0 attempt 9
001176.762: HDMI:EDID giving up on reading EDID block 0
001194.943: brfs: File read: /mfs/sd/config.txt
[... 355 lines cut ...]
[    4.603526] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.650221] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    5.484005] tun: Universal TUN/TAP device driver, 1.6
[    5.743103] uart-pl011 3f201000.serial: no DMA platform data
[    6.154113] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    6.165255] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    6.166076] brcmfmac: power management disabled
[    6.415864] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   12.183256] Bluetooth: Core ver 2.22
[   12.183351] NET: Registered protocol family 31
[   12.183358] Bluetooth: HCI device and connection manager initialized
[   12.183379] Bluetooth: HCI socket layer initialized
[   12.183393] Bluetooth: L2CAP socket layer initialized
[   12.183434] Bluetooth: SCO socket layer initialized
[   12.202773] Bluetooth: HCI UART driver ver 2.3
[   12.202787] Bluetooth: HCI UART protocol H4 registered
[   12.202918] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   12.203152] Bluetooth: HCI UART protocol Broadcom registered
[   12.408452] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.408460] Bluetooth: BNEP filters: protocol multicast
[   12.408474] Bluetooth: BNEP socket layer initialized
[   12.559220] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

**Additional context**
The issue has been mentioned some time ago for Alpine Linux (look for keyword "Update 3"): https://www.raspberrypi.org/forums/viewtopic.php?f=56&t=60569&p=765924&hilit=gpu_mem+config.txt+include#p764210

timg236, 2020-02-10:
Include directives are not supported in the second stage bootloader on either Pi3 or Pi4

pelwell, 2020-02-10:
To expand a bit on @timg236's comment, the `gpu_mem` setting is used by bootcode.bin (on Pi 3 and earlier) and by the SPI EEPROM (Pi 4) to determine where to load the main `start*.elf` firmware to. The config parsing done at that time is a restricted version of the full parser and is missing the `include` feature.

gvolt, 2020-02-10:
Thank you very much for your very fast replies.

Is there any other possibility to set gpu_mem from outside config.txt?

pelwell, 2020-02-10:
The answer is likely to be "No", but a bit of background on your use case would help.

gvolt, 2020-02-10:
The question originally arose on a Volumio (audio player) system whose system updates overwrite the config.txt. The config.txt provided by Volumio already contains an include option pointing to a file called userconfig.txt. Its purpose is to give the user the ability to save customized boot configuration parameters to a file that does not get overwritten on system updates. After first discovering the issue regarding gpu_mem on that kind of system I tested on the Stretch based Rasbian system to verify.

Out of interest: When (on what bootloader) does the include option get processed?

pelwell, 2020-02-10:
`include` is only expanded by the main firmware, i.e. `start*.elf`.

gvolt, 2020-02-10:
Good to know.

Is there a list which of the options from the [documentation](https://www.raspberrypi.org/documentation/configuration/config-txt/) do / don't work when placed in an included file?

pelwell, 2020-02-10:
The following are supported by bootcode.bin (and probably the EEPROM as well):

1. These settings:
```
gpu_mem
gpu_mem_256
gpu_mem_512
gpu_mem_1024
total_mem
require_total_mem
startx
start_x
start_debug
boot_partition
legacy_mapping
bootcode_delay
force_pvt
uart_2ndstage
start_file
fixup_file
```
2. Filtering my model (`[pi3]`)
3. Filtering by serial number
4. Filtering by gpio levels (`[gpio4=0]`)

Of those, only the filters are supported by the main firmware, so they are the only ones which will work in an included file (plus all of the other config settings).

timg236, 2020-02-10:
Force_pvt and legacy_mapping aren't supported in the EEPROM but the others are. The Pi4 bootloader should probably have a dedicated page for config.txt, Something to do next time I update https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

gvolt, 2020-02-10:
Ok, that's very helpful to know. Thanks a lot!

Regarding the options from the [list above](https://github.com/raspberrypi/firmware/issues/1332#issuecomment-584166999) (and especially gpu_mem) would there be another way to achieve the same result like setting them in config.txt (well, I know that's what config.txt is for...)?

pelwell, 2020-02-10:
> would there be another way to achieve the same result like setting them in config.txt (well, I know that's what config.txt is for...)?

No, sorry. Would putting the user configuration options in the included file work (after some user retraining), or is there another case where the user setting comes from the restricted list above? It feels like there shouldn't be - they are all very low-level settings.

gvolt, 2020-02-10:
In my case the users would not have to edit system files manually. They can install a plugin to the Volumio system which allows them to adjust e.g. the setting for gpu_mem through a GUI.

The plugin can easily work on the config.txt, but settings put in there would be overwritten each time the system gets updated whereas the included file does not get touched. So the plan was to use the include <file> option for gpu_mem (amongst others but not from the listed ones). If there would be another way to set gpu_mem in the boot process this would / should also be handled not directly by the user who only accesses the GUI.

pelwell, 2020-02-10:
The various "official" update methods - apt, rpi-update - never overwrite config.txt. You only get a new one by installing a new image.

Can the Volumio devs not be encouraged to put "system" settings in an included file that they are free to overwrite as they wish? Or split the file with a recognised separator so that user settings are preserved by an update?

gvolt, 2020-02-10:
Yes, system updates come as new images ;)

It might be worth a tought to flip the concept where "vanillla" settings brought by system updates are being placed. I will put that up for discussion.