raspberrypi/firmware #1334: RPi4 omxplayer vertical fullhd video corruption

s-aversa, 2020-02-13:
**Describe the bug**
When playing a vertical video (h264, 1080x1920) using mmal decoder, 
video display is corrupted.: only the left part of the frame is correct and the other part is full of noise/random colors.

**To reproduce**
- prepare a microsd with Raspbian buster with desktop 2020-02-05;
- boot the Rpi4
- copy the attached vidio to the Rpi
- play the video with: omxplayer vert.mp4
- the video display is corrupted

**Expected behaviour**
Video should play without corruption.

**Actual behaviour**
When playing a vertical video (h264, 1080x1920) using mmal decoder, 
video display is corrupted: only the left part of the frame is correct and the other part is full of noise/random colors I attached a picture of the video corruption.

After upgrading the firmware, the problem still exists.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

- hardware: RaspberryPI 4B 4Gb
- OS version:
Raspberry Pi reference 2020-02-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e577677b623b577f2a0ec7cfaffc3c27da005da3, stage4
- firmware version: Feb 12 2020 12:36:21
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)
- kernel: Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

**Additional info**
If I put the same microsd on a rpi3B+, the video plays correctly.

**Attached file**
Zip containing:
- fullhd vertical video sample
- picture of video corruption

[vert.zip](https://github.com/raspberrypi/firmware/files/4198641/vert.zip)

popcornmix, 2020-02-13:
I'm afraid this isn't supported. H264 video decode hardware is designed for BluRay playback (High Profile 4.1 up to 1920x1080).
Videos exceeding 1080 high cannot be decoded by the hardware.

s-aversa, 2020-02-13:
@popcornmix thanks for the explanation,
but, since it was working with a previous firmware, isn't possible to keep compatibility with vertical fullhd videos?

I'm sure that upto kernel 4.19.93, vertical video playing worked as expected on Rpi4.

popcornmix, 2020-02-13:
What version of firmware are you currently running (vcgencmd version)?
@6by9 this may be a regression from the restructuring of image allocation flags.

6by9, 2020-02-13:
I suspect it's the moving of the allocation of those extra 16 lines of context. I suspect I forgot to skip it if the height is > 1920-16, therefore the column stride exceeds the hardware limit.

6by9, 2020-02-13:
1080x1920 is supported as it still fits under the 8192 macroblock limit for Level 4/4.1.

We have hardware limits at 1920 lines high, and 2016(ish) pixels wide (it should be 2048, but something doesn't quite allow it).

s-aversa, 2020-02-13:
Now I'm using:
firmware version: Feb 12 2020 12:36:21
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)

6by9, 2020-02-13:
Fix pushed for review internally. Thanks for the report.

s-aversa, 2020-02-14:
Thank you all for fast response :)

DrSchottky, 2020-02-27:
Hi all,
I tested your latest VC firmware (00f355997ccdacfaac70e6e07803eccc0a6f1d6e) since I was facing the same issue.
The update apparently solved only partially the issue: playing author's jellyfish video in omxplayer now works, but if I rotate it by 90 degrees (`omxplayer --orientation 90`) frames get corrupted
[Here's](https://github.com/raspberrypi/firmware/files/4260917/issue.zip) a couple of screenshots showing the problem
@6by9 @popcornmix is it something fixable?

6by9, 2020-02-27:
Most things are fixable if they worked previously. (I'm assuming it previously worked)

Transpose is a funny case as things have to be transposed to memory first, and then that frame rendered. Something is obviously misinterpreting the image info, although in this case in a very oddly striped manner.

DrSchottky, 2020-02-27:
Yes, it stopped working from VC version 8f792e011c8a0ea3c5d47e7cc10172cc10b93c09 (commit 63bdbe0ea609a69f5c0a2212d608ef3d9915a7ad)

eduardm, 2020-02-29:
I can confirm that is not working after VC version 8f792e011c8a0ea3c5d47e7cc10172cc10b93c09
`omxplayer --orientation 90` is corrupting any video, no matter of the resolution.

I have more than 100 PIs, all models running most of the Raspbian versions and the exact same set of movies are working fine with or without rotation on all, but latest PI4

DrSchottky, 2020-03-07:
Hi, is this issue being under investigation or is it a won't fix?
Just asking to see if I should downgrade my Pi fleet or wait for an update

Thanks

JamesH65, 2020-03-08:
Not sure if anyone is looking in to it right now, but I would think someone will at some point.

6by9, 2020-03-09:
Issue found (and it is in video_render transposing). A patch should be pushed internally today, and in the next rpi-update release.

And please try to be accurate with your reports - the first clip I tried was 1080p and worked fine, so it is not "no matter of the resolution". In future please provide the details of what you have done rather than generalise, otherwise we may end up dismissing an issue due to different test conditions.

popcornmix, 2020-03-10:
We've pushed a potential fix to rpi-update firmware. Please test and confirm is issue is resolved.

rafhtl, 2020-04-11:
4.19.114 solves this problem
thank you

popcornmix, 2020-04-12:
@s-aversa is this resolved for you?

rafhtl, 2020-04-13:
Yes! I had the exact problem OMXPLAYER rotated 90 degrees and snow on the video
and now with rpi-update to 4.19.114  has solved
thank you

s-aversa, 2020-04-14:
Sorry about the delay in response...yes, it fixed the playing of vertical videos;
thanks

pelwell, 2020-04-14:
Please close the issue if the matter is resolved.

jjdisplay, 2020-05-27:
> Issue found (and it is in video_render transposing). A patch should be pushed internally today, and in the next rpi-update release.
>
> And please try to be accurate with your reports - the first clip [...]

@6by9 
I am facing the same problem after doing "sudo apt upgrade" and "omxplayer --orientation 90 video1.mp4" flickering. I am using raspberry4B-4G

I tried to do "sudo rpi-update" but the boot partition of 44MB insufficient space. Seems like rpi-update needed 256MB. 

I followed this tutorial of using gparted, but raspberry cannot boot up.
(https://learn.adafruit.com/resizing-raspberry-pi-boot-partition/edit-partitions)

May I know how to expand the boot partition to do the "sudo rpi-update".

Many thanks in advance

James

6by9, 2020-05-27:
> I tried to do "sudo rpi-update" but the boot partition of 44MB insufficient space. Seems like rpi-update for raspberry4B needed 256MB.

Raspbian Buster should have that anyway. What OS are you using? This isn't a general help forum for how to repartition SD cards.

Your other alternative is to download the firmware from https://github.com/Hexxeh/rpi-firmware (I'd suggest specifying a max depth if using git clone), and installing the files manually.

jjdisplay, 2020-05-27:
@6by9 
I am using Raspbian Buster.

for "installing the files manually", do I just copy all the files into boot partition? Just to confirm. 

Many thanks

jjdisplay, 2020-05-27:
> Raspbian Buster should have that anyway. What OS are you using? 
I am using Raspbian Buster

> Your other alternative is to download the firmware from https://github.com/Hexxeh/rpi-firmware (I'd suggest specifying a max depth if using git clone), and installing the files manually.
Noted. Thanks. Will try

Aiexis, 2020-10-28:
Hi,
I still have this issue even after rpi-update, on a rpi3B+
 uname -r
5.4.72-v7+
[image]

Thanks for help !

6by9, 2020-10-28:
@Aiexis Please confirm the output of ```vcgencmd version```, the geometry of the video file, and the full omxplayer command line you're using.

Aiexis, 2020-10-28:
Hi 6by9,
vcgencmd version -> 
Oct 22 2020 14:06:02
Copyright (c) 2012 Broadcom
version 74e754ff8947c58d2773253f77f6f68a303188f8 (clean) (release) (start)

Video geometry
omxplayer -i video.mp4
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'video.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: mp42mp41
    creation_time   : 2019-09-06T08:59:01.000000Z
  Duration: 00:00:08.92, start: 0.000000, bitrate: 8988 kb/s
    Stream #0:0(eng): Video: h264 (Main) (avc1 / 0x31637661), yuv420p(tv, bt709), 1080x1920, 8867 kb/s, 25 fps, 25 tbr, 25k tbn, 50 tbc (default)
    Metadata:
      creation_time   : 2019-09-06T08:59:01.000000Z
      handler_name    : ?Mainconcept Video Media Handler
      encoder         : AVC Coding

Command line:
omxplayer --orientation 270 video.mp4

Thank you!

Aiexis, 2020-11-02:
Hello
any idea ? :-)

6by9, 2020-11-02:
Yours is a slightly different use case from the original.
The original was taking 1920x1080 and rotating it 90/270degrees. You're taking 1080x1920 and rotating it 90/270 degrees. It's a slightly different permutation but needs a bit of time to find a suitable test clip and actually be able to test.

Aiexis, 2020-11-02:
You can download my test video here:
http://87.98.165.172/video.mp4

6by9, 2020-11-02:
Hmm, something odd.
My Pi did have firmware from August 2020, and that decoded your video fine. Updated to the latest and I get the same as you.
....
Cause found, and hopefully fixed.

Aiexis, 2020-11-02:
Indeed, Issue occurs just after apt-get /update upgrade & reboot from the last raspbian (or raspbian lite).

funoftracer, 2020-12-02:
Hi@all,
i have exacty the same problem as Aiexis:
vcgencmd version
Oct 22 2020 14:06:02 
Copyright (c) 2012 Broadcom
version 74e754ff8947c58d2773253f77f6f68a303188f8 (clean) (release) (start)

omxplayer -i test.mp4
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'test.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: isommp42
    creation_time   : 2020-12-02T15:29:35.000000Z
    com.android.version: 10
    com.android.capture.fps: 30.000000
  Duration: 00:00:07.09, start: 0.000000, bitrate: 14343 kb/s
    Stream #0:0(eng): Video: h264 (High) (avc1 / 0x31637661), yuv420p(tv, bt709/bt709/smpte170m), 1920x1080, 14081 kb/s, SAR 1:1 DAR 16:9, 29.92 fps, 30 tbr, 90k tbn, 180k tbc (default)
    Metadata:
      rotate          : 90
      creation_time   : 2020-12-02T15:29:35.000000Z
      handler_name    : VideoHandle
    Side data:
      displaymatrix: rotation of -90.00 degrees
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 256 kb/s (default)
    Metadata:
      creation_time   : 2020-12-02T15:29:35.000000Z
      handler_name    : SoundHandle
have a nice day ;)

Test Video and picture: 

[TestData](https://www.dropbox.com/sh/usxylhnhivm9kpz/AAC4Fsv97CCgOMbz4cHbKR3qa?dl=0) 

I hopefully fixed,
thx

popcornmix, 2020-12-03:
@funoftracer your firmware isn't new enough to have the fix referred to (it was mentioned on 2nd November on this issue).
You'll need to run rpi-update to get a testing firmware that includes this fix, or wait for apt to update (probably in a week or so).