raspberrypi/firmware #1335: vc4-fkms-v3d blank screen during boot with Plymouth

presslab-us, 2020-02-14:
**Describe the bug**
When using `dtoverlay=vc4-fkms-v3d` in `/boot/config.txt`, the Plymouth boot screen does not remain shown until boot is finished.  This is using HDMI0 on a rpi4.  This only occurs when the Plymouth theme is a graphical type, the text themes work properly. 

The graphical boot screen can be seen for a split second, and then the display backlight flickers like the HDMI is being reinitialized.  After that the screen backlight is on, but remains blank.  Once fully booted the graphics work as expected.

**To reproduce**
Configure `/boot/config.txt` with `dtoverlay=vc4-fkms-v3d`.
Configure a graphical Plymouth theme.
Reboot.

**Expected behaviour**
The Plymouth graphics are visible during boot.

**Actual behaviour**
The Plymouth graphics are visible for a split second, but then go blank during boot.

**System**
https://pastebin.com/dyEsFAXk

**Logs**
See pastebin above.

**Additional context**
It looks like 6by9 has seen a problem like this too in the past according to this post:
https://www.raspberrypi.org/forums/viewtopic.php?t=243564#p1485227

For testing I have blacklisted the vc4 kernel module and Plymouth graphics remain on the screen during boot.

6by9, 2020-02-14:
Not a firmware bug.

KMS replaces the simple framebuffer device as it loads.
If Plymouth has written to the simple framebuffer, then that is cleared.

You either need to compile vc4 as part of the main kernel image rather than a module, or otherwise tell Plymouth to delay/restart when the frame buffer is replaced.

presslab-us, 2020-02-14:
Thanks for the info!  I understand.

Do you know how is this issue handled in most distributions (like x86_64)?  It seems like it's pretty common to have the KMS drivers built as a module.  Is this a problem with Plymouth?  Or maybe Raspbian, as in Plymouth should be loaded after vc4?

presslab-us, 2020-02-14:
Here is my Plymouth debug log:
https://pastebin.com/cRCAaqPX

It looks like it initially uses the framebuffer, and uses that.  But later in the log, it looks like it finds the DRM devices, but cannot connect to any of them.

presslab-us, 2020-02-16:
It looks like this is the function that fails (returns zero):
https://gitlab.freedesktop.org/plymouth/plymouth/blob/42bbee8c3883507e51c722f75a80f19757ec5406/src/plugins/renderers/drm/plugin.c#L1093

presslab-us, 2020-03-12:
I'm going to close this.  Plymouth does try to restart (as you recommended) but for some reason it cannot connect using DRM.  I guess it could still be a Plymouth bug, but I also would not go so far to say it is not a firmware bug.

closed 2020-03-12 by presslab-us