raspberrypi/firmware #1340: /memreserve/ doesn't account for PSCI

kevans91, 2020-02-26:
Hi,

The /memreserve/ regions in the RPi3/4 DTB currently include only the first page of memory. For spin-table, the default, this is fine, but the PSCI stubs (used by FreeBSD) available over in raspberrypi/tools surpass this by one more page, so early bootloaders or OS boot are free to clobber the latter part of the PSCI stubs and hilarity ensures.

Any chance of increasing these to cover the first two pages instead?

pelwell, 2020-02-26:
I'm happy to take a Pull Request against our Linux repo.

kevans91, 2020-02-26:
> I'm happy to take a Pull Request against our Linux repo.

Ah, it wasn't clear to me that the .dts for the .dtb's committed in this repo were available over there -- will do, thanks.

pelwell, 2020-02-26:
By the way, where are the PSCI stubs in raspberrypi/tools?

kevans91, 2020-02-26:
Whoops, sorry, I was referring to these guys: https://github.com/raspberrypi/tools/tree/master/armstubs

pelwell, 2020-02-26:
Those stubs aren't PSCI stubs, and they're small, hence my confusion.

kevans91, 2020-02-26:
Ah, shoot- I managed to confuse myself, since we derived parts of what we use from there. Local problem; sorry...

closed 2020-02-26 by kevans91