raspberrypi/firmware #1341: RPi3: out of memory error when loading Linux with UEFI+GRUB since start.elf 4.19.93

pbatard, 2020-02-27:
## Description

Ever since the release of `start.elf` from commit [dc5622560a2589a417007156b75db71f52b60d80 &mdash; _kernel: Bump to 4.19.93_], attempts to boot __vanilla__ Debian 10.3 Linux through UEFI + GRUB result in the following:
```
Loading Linux 4.19.0-8-arm64 ...`
Loading initial ramdisk ...
error: out of memory.

Press any key to continue...
```

With `start.elf` from the previous commit [f3a973d104baa04bf6cff22d369336b54790dcaa &mdash; _kernel: Bump to 4.19.89_], which was released on the same day, the issue does not occur.

All subsequent releases of `start.elf` since 2020.01.06 produce the same issue.

This is very problematic for people trying to use the official EDK2 UEFI firmware (releases of which can be found [here](https://github.com/pbatard/RPi3)) as we currently always embed the latest version of `start.elf` which means our [most recent release](https://github.com/pbatard/RPi3/releases/tag/v1.15) is affected...

## To reproduce

To reproduce the issue you need to install vanilla Debian in UEFI mode, which can be carried out by following [this guide](https://pete.akeo.ie/2019/07/installing-debian-arm64-on-raspberry-pi.html) or [this similar guide from the Raspberry Pi forums](https://www.raspberrypi.org/forums/viewtopic.php?f=50&t=249449&sid=a51cb1453774825a3e7172059001b49c).

You will probably want to use the files from [release 1.14](https://github.com/pbatard/RPi3/releases/tag/v1.14) to carry out the Debian/GRUB installation, since it is unaffected, before replacing `start.elf` on the SD card.

## Expected behaviour

`start.elf` should not result in GRUB running out of memory and behave in the same manner as it did before the 2020.01.06 `start.elf` update.

## System

As mentioned above, this issue manifests itself when using a fully updated __vanilla__ Debian 10.3 on a Raspberry Pi 3 B+:

```
root@pi3:~# cat /etc/debian_version
10.3
root@pi3:~# uname -a
Linux pi3 4.19.0-8-arm64 #1 SMP Debian 4.19.98-1 (2020-01-26) aarch64 GNU/Linux
root@pi3:~# cat /proc/meminfo
MemTotal:         928592 kB
MemFree:          583924 kB
MemAvailable:     693336 kB
Buffers:           21736 kB
Cached:           146796 kB
SwapCached:            0 kB
Active:           151600 kB
Inactive:         108136 kB
Active(anon):      91684 kB
Inactive(anon):     9408 kB
Active(file):      59916 kB
Inactive(file):    98728 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:        976892 kB
SwapFree:         976892 kB
Dirty:                12 kB
Writeback:             0 kB
AnonPages:         89216 kB
Mapped:            89292 kB
Shmem:              9892 kB
Slab:              58440 kB
SReclaimable:      25220 kB
SUnreclaim:        33220 kB
KernelStack:        2480 kB
PageTables:         2300 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     1441188 kB
Committed_AS:     593520 kB
VmallocTotal:   135290290112 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
Percpu:              928 kB
HardwareCorrupted:     0 kB
AnonHugePages:     34816 kB
ShmemHugePages:        0 kB
ShmemPmdMapped:        0 kB
CmaTotal:          65536 kB
CmaFree:           65000 kB
HugePages_Total:       0
HugePages_Free:        0
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
Hugetlb:               0 kB
root@pi3:~# cat /etc/default/grub
# If you change this file, run 'update-grub' afterwards to update
# /boot/grub/grub.cfg.
# For full documentation of the options in this file, see:
#   info -f grub -n 'Simple configuration'

GRUB_DEFAULT=0
GRUB_TIMEOUT=5
GRUB_DISTRIBUTOR=`lsb_release -i -s 2> /dev/null || echo Debian`
GRUB_CMDLINE_LINUX_DEFAULT="quiet"
GRUB_CMDLINE_LINUX="console=ttyS0,115200"

# Uncomment to enable BadRAM filtering, modify to suit your needs
# This works with Linux (no patch required) and with any kernel that obtains
# the memory map information from GRUB (GNU Mach, kernel of FreeBSD ...)
#GRUB_BADRAM="0x01234567,0xfefefefe,0x89abcdef,0xefefefef"

# Uncomment to disable graphical terminal (grub-pc only)
#GRUB_TERMINAL=console

# The resolution used on graphical terminal
# note that you can use only modes which your graphic card supports via VBE
# you can see them in real GRUB with the command `vbeinfo'
#GRUB_GFXMODE=640x480

# Uncomment if you don't want GRUB to pass "root=UUID=xxx" parameter to Linux
#GRUB_DISABLE_LINUX_UUID=true

# Uncomment to disable generation of recovery mode menu entries
#GRUB_DISABLE_RECOVERY="true"

# Uncomment to get a beep at grub start
#GRUB_INIT_TUNE="480 440 1"

GRUB_TERMINAL=serial
GRUB_SERIAL_COMMAND="serial --unit=0 --speed=115200 --stop=1"
```

pelwell, 2020-02-27:
There isn't an obvious change between those points that might have caused a problem for Grub, We aren't Grub users, let alone Grub experts, so you need to take this up with the Grub devs. If they can explain what has changed (from the viewpoint of Grub) then we have a chance of fixing it.

pbatard, 2020-02-27:
Thanks for replying.

We'll be trying to investigate this from another angle (for now we've just reverted to using last know "good" versions of `start.elf` and `fixup.dat` in the archives we release, since it also seems that `fixup.dat` is also responsible for part of the issue), but we were hoping you guys might provide some insight as to what might have changed, such as an increased buffer allocation from the blobs or something, that could provide some hints.

At this stage, we're kind of expecting any GRUB investigation to turn out with _"Yes, there's just less left over memory with recent  `start.elf` and `fixup.dat` than there used to be before"_, at which stage we'll have no choice but to come back to you again.

I'll try to instrument GRUB to get a more detailed report for the cause of the error, but I would still invite you to take a second look at what might have changed between December and January in `start.elf` and `fixup.dat` that may adversely affect memory usage.

pelwell, 2020-02-27:
If fixup.dat doesn't exactly match start.elf then you will be reduced to the lowest common denominator, i.e. 128MB of RAM - they should always be treated as matched pairs.

pelwell, 2020-02-27:
How is GRUB working out how much RAM is available? Can't you get it to display its view of the start and end of RAM?

pbatard, 2020-02-27:
Okay. Up to this issue, we were always downloading the most recent version from this repo, so they should always have matched, but that's good to know!

pbatard, 2020-02-27:
Actually, your note on `fixup.dat` and `startup.elf` needing to match made me re-test the files from our problematic archive, just to make sure that I did properly extract both and wasn't testing with unmatched files, and I can't seem to replicate the issue...

So it would look like I might have been testing with an unmatched pair and ran into the 128MB RAM issue.

I'm going to run some more tests to make sure, and close the issue if I can confirm that it was just due to bad testing...

pelwell, 2020-02-27:
I suggest that the "Out of memory" error reporting (or general diagnostics) be extended to include the RAM limits, if possible.

pbatard, 2020-02-27:
Okay, I have confirmed that my issue boiled down to using unmatched `fixup.dat` and `start.elf`. Apologies for the noise.

But at least, I'm hoping having this issue logged will help people running into memory issues check that they aren't using unmatched blobs, as this is definitely something to be aware of.

I've contributed to GRUB a few times in the past, so I have just added improving the `out of memory` GRUB error message to report RAM layout on my TODO list, as this is good advice. Not sure when I'll get around to look into that though.

Before I close this issue, I can only thank you for the great help you've provided, even if it all boiled down to a user mistake.

closed 2020-02-27 by pbatard