raspberrypi/firmware #1342: CEC device control over HDMI1 Raspberry pi 4

DSBMartellock, 2020-02-29:
CEC-client does not detect any device plugged into the HDMI1 port on the raspberry pi 4 

No idea what this could be an issue with, or if it is an issue with the CEC protocol in the first place, as having multiple sinks controlled by a single source seems like a scenario uncovered in documentation. 

Ideally, I would be able to turn on or off both displays connected to the device.

6by9, 2020-02-29:
Currently CEC only supports a single connection. You should be able to choose for that to be HDMI1   via ```hdmi_preferred_port=1``` in config.txt.

uzlonewolf, 2020-03-02:
Any way to switch between them without editing config.txt and rebooting?

6by9, 2020-03-02:
No, and it's not a simple thing to fix via the firmware.
The VCHI service that provides CEC can't easily be adapted to run two instances, therefore it's not something that is worth investigating.

We are working to bring up a full KMS/DRM driver (instead of the current hybrid), and that should support independent CEC from the two connectors. How libcec handles that is anyone's guess.

audas, 2020-07-15:
The only way to turn off a second device plugged into the second HDMI port that I can see is to use a peripheral infrared device which is a complex work around.

Is there any timeline or road map of any description which points towards the second HDMI port having CEC functionality ?

Raspberry pi foundation sees it as a low use case because everyone who is using dual monitors has already switched to competing platforms on this issue.

popcornmix, 2020-07-16:
The kms driver has working CEC. You can update to it [here](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769&sid=4a9c909457a7ecc5f900794057ab06b6)
libcec is [hardcoded to use the first port](https://github.com/Pulse-Eight/libcec/blob/76551ea1dd9a55f0ce1533e440dc12dbc594f7ba/include/cectypes.h#L287) but it should be possible to change that to /dev/cec1 and rebuild.

audas, 2020-07-16:
It can not utilize both ports though ? Is that correct ?

popcornmix, 2020-07-16:
I believe the kernel supports both ports. How you drive them is up to the userland apps/lib like libcec.
I haven't tried but I'd imagine building libcec twice, one hard-coded to /dev/cec0 and one hardcoded to /dev/cec1
should allow both to work concurrently. Obviously a command line argument to cec-client would be cleaner.

audas, 2020-07-16:
Thanks. I understand this is bugs only - not discussion, so sorry for this question.

That all looks scary and above my pay grade messing directly with firmware on a new kernel.

 I have built the HDMI-CEC lib and use it from the command line inside Python to turn TV on/off on port HDMI 0

Updating the kernel (and firmware) will allow me to call both HDMI 0 and HDMI 1 from command line - or will I have to rebuild these libraries hard coded to the ports and the use each for their port.

I have several Raspi 4 4GB spare with cards all duplicated so I can experiment with one - I am rolling out commercial / industrial application.

Again, sorry and thanks.

popcornmix, 2020-07-16:
Firmware is largely irrelevant with kms (arm controls hdmi). You will need to use rpi-update to get 5.4 kernel.
You will need to build libcec with kernel interface enabled:
```
cmake -DHAVE_LINUX_API=1 -DHAVE_RPI_API=0 .. && \
make -j4
```
Hopefully that will work on hdmi0. Then try changing /dev/cec0 to /dev/cec1 and rebuild libcec. Check if that works on cec1.

audas, 2020-07-16:
Thank you for this.