raspberrypi/firmware #1343: RPI4 - No HDMI signal on Panasonic Viera

blacksocket, 2020-03-03:
Hi, I am having issues with displaying any HDMI output from my RPI4 (4GB) on TV Panasonic Viera TX-l55DT50E. I am using os.: 2020-02-05-raspbian-buster-lite.img up-to-date including rpi-firmware.

I tried almost every possible important configurations located in /boot/config.txt without success.
After rpi reboot the Panasonic TV is showing nothing (no video available). Other devices works fine on this TV HDMI input. 
But when connect the same rpi via the same hdmi cable on my PC Monitor Philips 276E7QDSW/00 it works fine.

Currently I have this configuration in /boot/config.txt:

----------------------------
hdmi_force_mode=1
hdmi_group=1
hdmi_mode=31 
hdmi_drive=2
config_hdmi_boost=9
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
---------------------------

pi@raspberrypi:~ $ /opt/vc/bin/tvservice -m CEA
Group CEA has 1 modes:
  (prefer) mode 31: 1920x1080 @ 50Hz 16:9, clock:148MHz progressive

**System**
-----------------------------
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.106-v7l+ #1297 SMP Tue Feb 25 13:19:54 GMT 2020 armv7l GNU/Linux
pi@raspberrypi:~ $ sudo vi /boot/config.txt
------------------------------------------
pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2020-02-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e577677b623b577f2a0ec7cfaffc3c27da005da3, stage2
pi@raspberrypi:~ $ vcgencmd version
Feb 28 2020 18:44:10
Copyright (c) 2012 Broadcom
version 9520663ed4fa4ca8c2a6923a522bff85fcdedef5 (clean) (release) (start)
-------------------------------------------------
pi@raspberrypi:~ $ /opt/vc/bin/edidparser edid.dat
Enabling fuzzy format match...
Parsing edid.dat...
HDMI:EDID version 1.3, 1 extensions, unknown aspect ratio
HDMI:EDID features - videodef 0x80 !standby !suspend !active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is Panasonic-TV
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 23-61 Hz, horizontal is 15-68 kHz, max pixel clock is 150 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID found preferred CEA detail timing format: 1920x1080p @ 50 Hz (31)
HDMI:EDID found CEA detail timing format: 1920x1080p @ 60 Hz (16)
HDMI:EDID established timing I/II bytes are 00 00 00
HDMI:EDID standard timings block x 8: 0x0101 0101 0101 0101 0101 0101 0101 0101
HDMI:EDID parsing v3 CEA extension 0
HDMI:EDID monitor support - underscan IT formats:no, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:2
HDMI:EDID found CEA detail timing format: 1920x1080i @ 50 Hz (20)
HDMI:EDID found CEA detail timing format: 1920x1080i @ 60 Hz (5)
HDMI:EDID found CEA detail timing format: 1280x720p @ 50 Hz (19)
HDMI:EDID found CEA detail timing format: 1280x720p @ 60 Hz (4)
HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz (native)
HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz (native)
HDMI:EDID found CEA format: code 20, 1920x1080i @ 50Hz
HDMI:EDID found CEA format: code 5, 1920x1080i @ 60Hz
HDMI:EDID found CEA format: code 32, 1920x1080p @ 24Hz
HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz
HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz
HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz
HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz
HDMI:EDID found CEA format: code 17, 720x576p @ 50Hz
HDMI:EDID found CEA format: code 2, 720x480p @ 60Hz
HDMI:EDID found CEA format: code 22, 1440x576i @ 50Hz
HDMI:EDID found CEA format: code 7, 1440x480i @ 60Hz
HDMI:EDID found CEA format: code 21, 1440x576i @ 50Hz
HDMI:EDID found CEA format: code 6, 1440x480i @ 60Hz
HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz
HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48 kHz, sample size: 16 bits
HDMI:EDID found HDMI VSDB length 24
HDMI:EDID HDMI VSDB has physical address 2.0.0.0
HDMI:EDID HDMI VSDB supports AI:yes, dual link DVI:no
HDMI:EDID HDMI VSDB deep colour support - 48-bit:no 36-bit:yes 30-bit:yes DC_yuv444:yes
HDMI:EDID HDMI VSDB max TMDS clock 190 MHz
HDMI:EDID HDMI VSDB content type support: game|cinema|photo|text
HDMI:EDID HDMI VSDB supports 3D formats
HDMI:EDID HDMI VSDB 3D_Mask: 0xfffc
HDMI:EDID HDMI VSDB 3D_Structure_All: 0x8149
HDMI:EDID HDMI VSDB explicit 3D mode: 2D VIC order 0 (CEA 31), 3D struct 0x06, 3D detail 0x00
HDMI:EDID HDMI VSDB explicit 3D mode: 2D VIC order 1 (CEA 16), 3D struct 0x06, 3D detail 0x00
HDMI:EDID HDMI VSDB explicit 3D mode: 2D VIC order 0 (CEA 31), 3D struct 0x08, 3D detail 0x00
HDMI:EDID HDMI VSDB explicit 3D mode: 2D VIC order 1 (CEA 16), 3D struct 0x08, 3D detail 0x00
HDMI:EDID adding mandatory support for DMT (4) 640x480p @ 60Hz
HDMI:EDID VSDB 3D legend:
FP=frame packing, F-Alt=Field Alternative, L-Alt=Line Alternative
SbS-Full=Side by Side Full, Ldep=L Depth, Ldep+Gfx=L Depth + Graphics Depth
TopBot=Top and Bottom, SbS-HH=Side by Side half horizontal
SbS-OLOR=Side by Side odd left odd right, SbS-OLER=Side by Side odd left even right
SbS-ELOR=Side by Side even left odd right, SbS-ELER=Side by Side even left even right
HDMI:EDID CEA (32) 1920x1080p 24Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (4) 1280x720p 60Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (5) 1920x1080i 60Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (19) 1280x720p 50Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (20) 1920x1080i 50Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (31) 1920x1080p 50Hz 3D supports: TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (16) 1920x1080p 60Hz 3D supports: TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (18) 720x576p 50Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (3) 720x480p 60Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (17) 720x576p 50Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (2) 720x480p 60Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (22) 1440x576i 50Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (7) 1440x480i 60Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (21) 1440x576i 50Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (6) 1440x480i 60Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID CEA (1) 640x480p 60Hz 3D supports: FP|SbS-Full|TopBot|SbS-HH|SbS-OLOR|SbS-OLER|SbS-ELOR|SbS-ELER
HDMI:EDID filtering formats with pixel clock unlimited MHz or h. blanking unlimited
HDMI:EDID best score mode initialised to CEA (1) 640x480p @ 60 Hz with pixel clock 25 MHz (score 0)
HDMI:EDID best score mode is now CEA (1) 640x480p @ 60 Hz with pixel clock 25 MHz (score 61864)
HDMI:EDID best score mode is now CEA (2) 720x480p @ 60 Hz with pixel clock 27 MHz (score 66472)
HDMI:EDID CEA mode (3) 720x480p @ 60 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID best score mode is now CEA (4) 1280x720p @ 60 Hz with pixel clock 74 MHz (score 2635592)
HDMI:EDID DMT mode (4) 640x480p @ 60 Hz with pixel clock 25 MHz has a score of 18432
HDMI:EDID best score mode is now CEA (5) 1920x1080i @ 60 Hz with pixel clock 74 MHz (score 3773832)
HDMI:EDID CEA mode (6) 1440x480i @ 60 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID CEA mode (7) 1440x480i @ 60 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID best score mode is now CEA (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz (score 4898248)
HDMI:EDID CEA mode (17) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID CEA mode (18) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID CEA mode (19) 1280x720p @ 50 Hz with pixel clock 74 MHz has a score of 3117160
HDMI:EDID CEA mode (20) 1920x1080i @ 50 Hz with pixel clock 74 MHz has a score of 4232360
HDMI:EDID CEA mode (21) 1440x576i @ 50 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID CEA mode (22) 1440x576i @ 50 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID best score mode is now CEA (31) 1920x1080p @ 50 Hz with pixel clock 148 MHz (score 5336040)
HDMI:EDID CEA mode (32) 1920x1080p @ 24 Hz with pixel clock 74 MHz has a score of 124532
HDMI0:EDID preferred mode remained as CEA (31) 1920x1080p @ 50 Hz with pixel clock 148 MHz
HDMI:EDID has HDMI support and audio support
/opt/vc/bin/edidparser exited with code 0

Can anyone please help with this issue ?

Calzor-Suzay, 2020-04-07:
I have the same issue, RPi4 on latest image of Buster.
I can't get any display, not even rainbow boot screen on a Panasonic TH-37PX70B.
Works perfectly OK on newer Samsung 4K TV.

pi@raspberrypi4:~ $ tvservice -s
state 0xa [HDMI CUSTOM RGB lim 16:9], 1280x720 @ 50.00Hz, progressive
pi@raspberrypi4:~ $ tvservice -m CEA
Group CEA has 15 modes:
           mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive
           mode 2: 720x480 @ 60Hz 4:3, clock:27MHz progressive
           mode 3: 720x480 @ 60Hz 16:9, clock:27MHz progressive
  (native) mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive
           mode 5: 1920x1080 @ 60Hz 16:9, clock:74MHz interlaced
           mode 6: 720x480 @ 60Hz 4:3, clock:27MHz x2 interlaced
           mode 7: 720x480 @ 60Hz 16:9, clock:27MHz x2 interlaced
           mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive
           mode 17: 720x576 @ 50Hz 4:3, clock:27MHz progressive
           mode 18: 720x576 @ 50Hz 16:9, clock:27MHz progressive
  (prefer) mode 19: 1280x720 @ 50Hz 16:9, clock:74MHz progressive
           mode 20: 1920x1080 @ 50Hz 16:9, clock:74MHz interlaced
           mode 21: 720x576 @ 50Hz 4:3, clock:27MHz x2 interlaced
           mode 22: 720x576 @ 50Hz 16:9, clock:27MHz x2 interlaced
           mode 31: 1920x1080 @ 50Hz 16:9, clock:148MHz progressive
pi@raspberrypi4:~ $ tvservice -m DMT
Group DMT has 1 modes:
           mode 4: 640x480 @ 60Hz 4:3, clock:25MHz progressive
pi@raspberrypi4:~ $ tvservice -d edid
Written 256 bytes to edid
pi@raspberrypi4:~ $ edidparser edid
Enabling fuzzy format match...
Parsing edid...
HDMI:EDID version 1.3, 1 extensions, unknown aspect ratio
HDMI:EDID features - videodef 0x80 !standby !suspend !active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is PANASONIC-TV
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 49-61 Hz, horizontal is 15-68 kHz, max pixel clock is 150 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID found preferred CEA detail timing format: 1280x720p @ 50 Hz (19)
HDMI:EDID found CEA detail timing format: 1280x720p @ 60 Hz (4)
HDMI:EDID established timing I/II bytes are 00 00 00
HDMI:EDID standard timings block x 8: 0x0101 0101 0101 0101 0101 0101 0101 0101
HDMI:EDID parsing v3 CEA extension 0
HDMI:EDID monitor support - underscan IT formats:no, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:2
HDMI:EDID found CEA detail timing format: 1920x1080i @ 50 Hz (20)
HDMI:EDID found CEA detail timing format: 720x576p @ 50 Hz (17)
HDMI:EDID found CEA detail timing format: 1920x1080i @ 60 Hz (5)
HDMI:EDID found CEA detail timing format: 720x480p @ 60 Hz (2)
HDMI:EDID found CEA detail timing format: 1920x1080p @ 50 Hz (31)
HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz (native)
HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz (native)
HDMI:EDID found CEA format: code 20, 1920x1080i @ 50Hz
HDMI:EDID found CEA format: code 5, 1920x1080i @ 60Hz
HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz
HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz
HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz
HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz
HDMI:EDID found CEA format: code 17, 720x576p @ 50Hz
HDMI:EDID found CEA format: code 2, 720x480p @ 60Hz
HDMI:EDID found CEA format: code 22, 1440x576i @ 50Hz
HDMI:EDID found CEA format: code 7, 1440x480i @ 60Hz
HDMI:EDID found CEA format: code 21, 1440x576i @ 50Hz
HDMI:EDID found CEA format: code 6, 1440x480i @ 60Hz
HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz
HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48 kHz, sample size: 16 bits
HDMI:EDID found HDMI VSDB length 6
HDMI:EDID HDMI VSDB has physical address 1.0.0.0
HDMI:EDID HDMI VSDB supports AI:yes, dual link DVI:no
HDMI:EDID HDMI VSDB deep colour support - 48-bit:no 36-bit:no 30-bit:no DC_yuv444:no
HDMI:EDID HDMI VSDB has no latency information
HDMI:EDID adding mandatory support for DMT (4) 640x480p @ 60Hz
HDMI:EDID filtering formats with pixel clock unlimited MHz or h. blanking unlimited
HDMI:EDID best score mode initialised to CEA (1) 640x480p @ 60 Hz with pixel clock 25 MHz (score 0)
HDMI:EDID best score mode is now CEA (1) 640x480p @ 60 Hz with pixel clock 25 MHz (score 61864)
HDMI:EDID best score mode is now CEA (2) 720x480p @ 60 Hz with pixel clock 27 MHz (score 2566472)
HDMI:EDID CEA mode (3) 720x480p @ 60 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID best score mode is now CEA (4) 1280x720p @ 60 Hz with pixel clock 74 MHz (score 4690888)
HDMI:EDID DMT mode (4) 640x480p @ 60 Hz with pixel clock 25 MHz has a score of 18432
HDMI:EDID CEA mode (5) 1920x1080i @ 60 Hz with pixel clock 74 MHz has a score of 3273832
HDMI:EDID CEA mode (6) 1440x480i @ 60 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID CEA mode (7) 1440x480i @ 60 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID CEA mode (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz has a score of 273832
HDMI:EDID CEA mode (17) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 3566472
HDMI:EDID CEA mode (18) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID best score mode is now CEA (19) 1280x720p @ 50 Hz with pixel clock 74 MHz (score 5163240)
HDMI:EDID CEA mode (20) 1920x1080i @ 50 Hz with pixel clock 74 MHz has a score of 4232360
HDMI:EDID CEA mode (21) 1440x576i @ 50 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID CEA mode (22) 1440x576i @ 50 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID CEA mode (31) 1920x1080p @ 50 Hz with pixel clock 148 MHz has a score of 2232360
HDMI0:EDID preferred mode remained as CEA (19) 1280x720p @ 50 Hz with pixel clock 74 MHz
HDMI:EDID has HDMI support and audio support
edidparser exited with code 0
pi@raspberrypi4:~ $

JamesH65, 2020-04-07:
It's odd that neither are choosing 1080p60, but rather slightly less FPS. Not sure why. I'd normally expect it to use 1080pp60 if it's present, and it is on both those devices.

Calzor-Suzay, 2020-04-07:
I'm not picking anything it's what it's picking itself but I've tried loads of options and basically never get anything, no signal to the TV.

I'm not trying to match a decent resolution to the screen I'm trying to get anything.

JamesH65, 2020-04-07:
> I'm not picking anything it's what it's picking itself but I've tried loads of options and basically never get anything, no signal to the TV.
> 
> I'm not trying to match a decent resolution to the screen I'm trying to get anything.

Sorry, not saying it's anything you have done. The EDID parser code is choosing an unexpected mode - i.e. one that isn't 1080p60 which when given the option is usually the one that is chosen. Its instead choosing a lesser mode. Which is odd. Or as least, it seems to to me.

6by9, 2020-04-07:
Whilst edidparser's output isn't perfect (and we can't analyse a parsed output), it appears the EDID includes a preferred detail timing for 1280x720@50, hence that is chosen in preference to anything else
```HDMI:EDID found preferred CEA detail timing format: 1280x720p @ 50 Hz (19)```

JamesH65, 2020-04-07:
Didn't appear to be the case in the second dump though, both 50 and 60hz were native, so I would have expected 60, unless I misread it.

6by9, 2020-04-07:
Detailed modes take precedence over listed CEA VIC modes. 

Without the raw EDID then there isn't much analysis that can be done.

Calzor-Suzay, 2020-04-07:
How do I get the raw EDID information?

Calzor-Suzay, 2020-04-07:
Had to zip it is that what you're after following the command
tvservice -d edid.bin
[edid.zip](https://github.com/raspberrypi/firmware/files/4444991/edid.zip)

Calzor-Suzay, 2020-04-07:
BTW it's natively a 720p screen with a resolution of 1024x720 like most old Plasmas

JamesH65, 2020-04-07:
An easy way of posting the EDID is to use base64 to convert to text then cut and paste here.

Calzor-Suzay, 2020-04-07:
Bin file as attached above but used https://base64.guru/converter/encode/file as plain text and got...

> AP///////wA0qWSgOwQAAAsRAQOAAAB4Ctr/o1hKoikXSUsAAAABAQEBAQEBAQEBAQEBAQEBAR0AvFLQHiC4KFVAmiZTAAAeAR0AclHQHiBuKFUAmiZTAAAeAAAA/ABQQU5BU09OSUMtVFYKAAAA/QAxPQ9EDwAKICAgICAgASQCAx9yT5OEFAUfEBIDEQIWBxUGASMJBwFmAwwAEACAAR2A0HIcFiAQLCWAmiZTAACejArQkCBAMSAMQFUAmiZTAAAYAR2AGHEcFiBYLCUAmiZTAACejArQiiDgLRAQPpYAmiZTAAAYAjqA0HI4LUAQLEWAmiZTAAAYAAAAAAAADQ==

Is that what you're after?

JamesH65, 2020-04-07:
Yes, that's the ticket!

6by9, 2020-04-07:
Either base64 or zipping works.

It lists 720p50 first, and then 720p60 as detailed timings, and both VIC 19 (720p50) and 4 (720p60) as native in the CEA modes list. The first of those will therefore be taken as the preferred mode.

```
edid-decode (hex):

00 ff ff ff ff ff ff 00 34 a9 64 a0 3b 04 00 00
0b 11 01 03 80 00 00 78 0a da ff a3 58 4a a2 29
17 49 4b 00 00 00 01 01 01 01 01 01 01 01 01 01
01 01 01 01 01 01 01 1d 00 bc 52 d0 1e 20 b8 28
55 40 9a 26 53 00 00 1e 01 1d 00 72 51 d0 1e 20
6e 28 55 00 9a 26 53 00 00 1e 00 00 00 fc 00 50
41 4e 41 53 4f 4e 49 43 2d 54 56 0a 00 00 00 fd
00 31 3d 0f 44 0f 00 0a 20 20 20 20 20 20 01 24

02 03 1f 72 4f 93 84 14 05 1f 10 12 03 11 02 16
07 15 06 01 23 09 07 01 66 03 0c 00 10 00 80 01
1d 80 d0 72 1c 16 20 10 2c 25 80 9a 26 53 00 00
9e 8c 0a d0 90 20 40 31 20 0c 40 55 00 9a 26 53
00 00 18 01 1d 80 18 71 1c 16 20 58 2c 25 00 9a
26 53 00 00 9e 8c 0a d0 8a 20 e0 2d 10 10 3e 96
00 9a 26 53 00 00 18 02 3a 80 d0 72 38 2d 40 10
2c 45 80 9a 26 53 00 00 18 00 00 00 00 00 00 0d

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.3
  Vendor & Product Identification:
    Manufacturer: MEI
    Model: 41060
    Serial Number: 1083
    Made in: week 11 of 2007
  Basic Display Parameters & Features:
    Digital display
    Image size is variable
    Gamma: 2.20
    RGB color display
    First detailed timing is preferred timing
  Color Characteristics:
    Red  : 0.6396, 0.3447
    Green: 0.2910, 0.6347
    Blue : 0.1630, 0.0927
    White: 0.2880, 0.2958
  Established Timings I & II: none
  Standard Timings: none
  Detailed Timing Descriptors:
    DTD 1:  1280x720    50.000 Hz  16:9    37.500 kHz  74.250 MHz (1434 mm x 806 mm)
                 Hfront  440 Hsync  40 Hback 220 Hpol P
                 Vfront    5 Vsync   5 Vback  20 Vpol P
    DTD 2:  1280x720    60.000 Hz  16:9    45.000 kHz  74.250 MHz (1434 mm x 806 mm)
                 Hfront  110 Hsync  40 Hback 220 Hpol P
                 Vfront    5 Vsync   5 Vback  20 Vpol P
    Display Product Name: 'PANASONIC-TV'
  Display Range Limits:
    Monitor ranges (GTF): 49-61 Hz V, 15-68 kHz H, max dotclock 150 MHz
  Extension blocks: 1
Checksum: 0x24

----------------

Block 1, CTA-861 Extension Block:
  Revision: 3
  Basic audio support
  Supports YCbCr 4:4:4
  Supports YCbCr 4:2:2
  Native detailed modes: 2
  Video Data Block:
    VIC  19:  1280x720    50.000 Hz  16:9    37.500 kHz  74.250 MHz (native)
    VIC   4:  1280x720    60.000 Hz  16:9    45.000 kHz  74.250 MHz (native)
    VIC  20:  1920x1080i  50.000 Hz  16:9    28.125 kHz  74.250 MHz
    VIC   5:  1920x1080i  60.000 Hz  16:9    33.750 kHz  74.250 MHz
    VIC  31:  1920x1080   50.000 Hz  16:9    56.250 kHz 148.500 MHz
    VIC  16:  1920x1080   60.000 Hz  16:9    67.500 kHz 148.500 MHz
    VIC  18:   720x576    50.000 Hz  16:9    31.250 kHz  27.000 MHz
    VIC   3:   720x480    59.940 Hz  16:9    31.469 kHz  27.000 MHz
    VIC  17:   720x576    50.000 Hz   4:3    31.250 kHz  27.000 MHz
    VIC   2:   720x480    59.940 Hz   4:3    31.469 kHz  27.000 MHz
    VIC  22:  1440x576i   50.000 Hz  16:9    15.625 kHz  27.000 MHz
    VIC   7:  1440x480i   59.940 Hz  16:9    15.734 kHz  27.000 MHz
    VIC  21:  1440x576i   50.000 Hz   4:3    15.625 kHz  27.000 MHz
    VIC   6:  1440x480i   59.940 Hz   4:3    15.734 kHz  27.000 MHz
    VIC   1:   640x480    59.940 Hz   4:3    31.469 kHz  25.175 MHz
  Audio Data Block:
    Linear PCM:
      Max channels: 2
      Supported sample rates (kHz): 48 44.1 32
      Supported sample sizes (bits): 16
  Vendor-Specific Data Block (HDMI), OUI 00-0C-03:
    Source physical address: 1.0.0.0
    Supports_AI
  Detailed Timing Descriptors:
    DTD 3:  1920x1080i  50.000 Hz  16:9    28.125 kHz  74.250 MHz (1434 mm x 806 mm)
                 Hfront  528 Hsync  44 Hback 148 Hpol P
                 Vfront    2 Vsync   5 Vback  15 Vpol P Vfront +0.5 Odd Field
                 Vfront    2 Vsync   5 Vback  15 Vpol P Vback  +0.5 Even Field
    DTD 4:   720x576    50.000 Hz   5:4    31.250 kHz  27.000 MHz (1434 mm x 806 mm)
                 Hfront   12 Hsync  64 Hback  68 Hpol N
                 Vfront    5 Vsync   5 Vback  39 Vpol N
    DTD 5:  1920x1080i  60.000 Hz  16:9    33.750 kHz  74.250 MHz (1434 mm x 806 mm)
                 Hfront   88 Hsync  44 Hback 148 Hpol P
                 Vfront    2 Vsync   5 Vback  15 Vpol P Vfront +0.5 Odd Field
                 Vfront    2 Vsync   5 Vback  15 Vpol P Vback  +0.5 Even Field
    DTD 6:   720x480    59.940 Hz   3:2    31.469 kHz  27.000 MHz (1434 mm x 806 mm)
                 Hfront   16 Hsync  62 Hback  60 Hpol N
                 Vfront    9 Vsync   6 Vback  30 Vpol N
    DTD 7:  1920x1080   50.000 Hz  16:9    56.250 kHz 148.500 MHz (1434 mm x 806 mm)
                 Hfront  528 Hsync  44 Hback 148 Hpol N
                 Vfront    4 Vsync   5 Vback  36 Vpol N
Checksum: 0xd
```

pelwell, 2020-04-07:
FYI base64 is easily installable with `sudo apt install coreutils`, after which you can `base64 edid.bin`.

JamesH65, 2020-04-07:
OK, so the choice is seemingly valid, but still not outputting? Not had any reports that I know of, of that particular mode not working elsewhere.  I guess the HDMI analyser is in the office? Both devices are Panasonic's but different models, one 720p, one 1080p but they are both plasmas' so fairly old.

Calzor-Suzay, 2020-04-07:
Similar experience here with an 'old' device https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=255862&p=1639075#p1639075

Calzor-Suzay, 2020-04-08:
Don't really want to do it but would a Micro HDMI to VGA connector work or have the same issue?
My TV has a PC-VGA connector, would I get a proper 720p picture?

Calzor-Suzay, 2020-04-13:
Having played with a Micro HDMI to VGA connector this has proved no good, highest resolution I can get is 1024x768 so I cannot play 720p picture.

Any idea, do we need a firmware update, anything I can do my end?

jksmurf, 2020-06-29:
Hi, recently moved to a Pi4 from a 3B, the latter which worked on my old Panasonic Viera T32LX700 - but only a black screen with the Pi4. Tried most of the recommended config.txt settings I could find. Any update since April please ?

pilbzy, 2022-06-30:
I just commented the uncommented #Enable DRM VC4 V3D under [pi4] at the bottom of busters config.txt and have hdmi picture on a panisonic Vieira with a 4b. Might be worth a try?