raspberrypi/firmware #1344: Image file will not boot

chuckb, 2020-03-07:
**Describe the bug**
When I build an image file with https://github.com/waldheinz/fat32-lib (version 0.6.5) using the SuperFloppyFormatter and ImageBuilder classes, and deploy to an SD using dd, my Pi will not boot. No lights, no nothing. 

**To reproduce**
I will include a link to an image file shortly...

**Expected behaviour**
I can deploy my image on to the SD and it will boot the Pi.

**Actual behaviour**
Pi does not boot as expected.

**System**
PiZeroW
No OS...bare metal.
Firmware version: Tried both 1.20190925 and master.
Kernel = Serial bootloader located: https://github.com/chuckb/raspbootin

**Additional context**
Setting the uart_2ndstage=1 produces no output in terminal. When I mount the card on my Mac, I see bootcode.bin, start.elf, config.txt, kernel.img, and fixup.dat files as expected. Further, if I md5 the files, they match my source files.

If I then copy all my source files on top of SD card image deployed files on my Mac, then the Pi will boot as expected (I am using a custom serial bootloader kernel...not Linux).

If I copy just bootloader.bin (from a freshly deployed image), I get
```
Raspberry Pi Bootcode
Read File: config.txt, 84
```
and 4 short green LED flashes (meaning cannot find start.elf...which appears to be there according to Mac).

If I then copy start.elf, then the Pi will boot correctly, and interestingly the log notes that it can load the kernel.img (which I did not copy over).

It appears to me that FAT file handling is different between loading bootloader.bin, within bootloader.bin, and then works further differently within start.elf (given that it can find kernel.img without an over copy).

Clearly it seems to me that my FAT library and the FAT handling contained within the firmware have some compatibility issues. But my Mac has no problem mounting, reading/finding the files after image is deployed with dd.

chuckb, 2020-03-07:
I figured out the root cause. The Java FAT library was writing out long file name (LFN) directory entries even if files names were 8.3 (in this case, because the names were lower case). Once I changed the file names to upper case and rebuilt and deployed the image, then the Pi will boot.

Interestingly, I also confirmed the behavior of overwriting the file from the Mac, that even if the file is in lower case on the Mac file system, the Mac FAT driver does not write an LFN entry and further converts the file name to uppercase within the directory record. I am also aware of bit flags introduced in Windows NT and certain Windows XP versions that deal with lowercase handling of 8.3 file names without having to write a LFN, which is probably what the Mac driver is doing (I did not look that carefully yet).

Before I close this issue, can a RPi team member confirm whether first and second stage boot loaders deal with FAT volumes containing LFNs? I'd like to know because then I will look into how the Java lib builds them, because obviously something is not quite right.

chuckb, 2020-03-11:
I forked and fixed the Java library https://github.com/chuckb/fat32-lib to write lower case short file names using the user attributes in the 0x0c offset of the fat directory entry. Pi bootloaders can now find files in volumes created by this library.

closed 2020-03-11 by chuckb

timg236, 2020-03-11:
To confirm your comments both the ROM and bootcode.bin / SPI EEPROM all require short filenames.
https://www.raspberrypi.org/forums/viewtopic.php?t=20588#p198998

Start.elf supports long filenames for device-tree blobs and maybe the Pi4 EEPROM will in future. However, since the chip ROM requires short filenames for recovery.bin I think you should assume that the boot partition on Raspberry Pi must always contain the short filenames.

chuckb, 2020-03-11:
The nuance and confusion stems from the FAT/VFAT specification, which has much "history". Mixed case file names, regardless of length, require long file names. In the early days of the FAT specification, 8.3 filenames did not support lower case, and any file name created would be stored with an upper case name within the directory structure. Then came along VFAT, which supported mixed case, UNICODE, long file names (LFN). Then came along a modification to VFAT supporting lowercase 8.3 basename and extension without requiring use of the LFN specification. So it seems like the 1st and 2nd stage bootloaders support a type of hybrid VFAT.