raspberrypi/firmware #1345: Pi4: Display goes blank / something inside the kernel or firmware crashes when running xrandr -o too often

ganzgustav22, 2020-03-10:
We are using the screen orientation functionality of xrandr for a kiosk-stand operating system image with the Raspberry Pi4. There are different kiosk-stand models with different screens mounted in different orientations, so after initial bootup we start a setup webpage that lets the user rotate the screen (via xrandr -o normal, -o left, -o inverted, -o right) until the correct orientation is set.

When doing this, we found that when rotating the screen too fast and/or too often (not sure exactly what causes it), the display turns off (monitor shows "no signal") and these messages appear in the kernel log:

> [ 845.945909] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[ 845.945923] [drm] dumb: 243016kb BOs (31)
[ 845.947141] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[ 845.947151] [drm] dumb: 243016kb BOs (31)

How to reproduce:
- Write a fresh Raspbian Lite 2020-02-13 image to sdcard
- run "sudo apt-get install xorg openbox"
- run "startx"
- start an xterm and run this script there:

> #!/bin/bash
   export DISPLAY=:0
>    i=0
    while true; do
    /usr/bin/xrandr -o normal
    /usr/bin/xrandr -o left
    /usr/bin/xrandr -o inverted
    /usr/bin/xrandr -o right
    ((i=i+1))
    echo "Iteration: $i"
    done

JamesH65, 2020-03-10:
@6by9 I had a quick look but too unfamiliar with the code to make much progress.

6by9, 2020-03-10:
(Linux kernel, not firmware)
Generally it is Mesa that is doing the rotations, not DRM/KMS. If you have 2 HDMI monitors connected with one rotated, the primary (non-rotated) frame buffer is the combined size and one monitor will be pointed at a portion of that. The second monitor will be pointed at a second framebuffer that GL will have transposed from the primary FB into that one.
As both resources need to be passable to the HVS they will both be CMA allocations.

Annoyingly there is no tracking of who/what allocated CMA regions.
The contents of /sys/kernel/debug/dma_buf may be useful to show if any are still being exported (to V3D predominantly).
You want config options CMA_DEBUG and CMA_DEBUGFS enabled for more debug info.

ganzgustav22, 2020-03-10:
There is only one monitor connected. I am using the micro-HDMI port next to the USB-C power connector.
 
cat  /sys/kernel/debug/dma_buf/bufinfo before the issue occurs:
```
Dma-buf Objects:
size            flags           mode            count           exp_name
08294400        00000000        00080005        00000009        drm
        Exclusive fence: drm_sched v3d_tfu signalled
        Attached Devices:
        fec00000.v3d
        fec00000.v3d
Total 2 devices attached

08294400        00000000        00080005        00000009        drm
        Exclusive fence: drm_sched v3d_tfu signalled
        Attached Devices:
        fec00000.v3d
        fec00000.v3d
Total 2 devices attached

08294400        00000000        00080005        00000005        drm
        Exclusive fence: drm_sched v3d_render signalled
        Attached Devices:
        fec00000.v3d
Total 1 devices attached

08294400        00000000        00080005        00000005        drm
        Exclusive fence: drm_sched v3d_tfu signalled
        Attached Devices:
        fec00000.v3d
Total 1 devices attached

Total 4 objects, 33177600 bytes

```

cat  /sys/kernel/debug/dma_buf/bufinfo after the issue occured:
```
Dma-buf Objects:
size            flags           mode            count           exp_name
08294400        00000000        00080005        00000005        drm
        Exclusive fence: drm_sched v3d_render signalled
        Attached Devices:
        fec00000.v3d
Total 1 devices attached

08294400        00000000        00080005        00000005        drm
        Exclusive fence: drm_sched v3d_tfu signalled
        Attached Devices:
        fec00000.v3d
Total 1 devices attached

08294400        00000000        00080005        00000005        drm
        Exclusive fence: drm_sched v3d_tfu signalled
        Attached Devices:
        fec00000.v3d
Total 1 devices attached

08294400        00000000        00080005        00000009        drm
        Exclusive fence: drm_sched v3d_tfu signalled
        Attached Devices:
        fec00000.v3d
        fec00000.v3d
Total 2 devices attached

08294400        00000000        00080005        00000005        drm
        Exclusive fence: drm_sched v3d_tfu signalled
        Attached Devices:
        fec00000.v3d
Total 1 devices attached
[... 124 lines cut ...]
        Attached Devices:
        fec00000.v3d
Total 1 devices attached

08294400        00000000        00080005        00000005        drm
        Exclusive fence: drm_sched v3d_tfu signalled
        Attached Devices:
        fec00000.v3d
Total 1 devices attached

08294400        00000000        00080005        00000005        drm
        Exclusive fence: drm_sched v3d_tfu signalled
        Attached Devices:
        fec00000.v3d
Total 1 devices attached

08294400        00000000        00080005        00000005        drm
        Exclusive fence: drm_sched v3d_tfu signalled
        Attached Devices:
        fec00000.v3d
Total 1 devices attached

Total 29 objects, 240537600 bytes

```

6by9, 2020-03-10:
Thanks. I was making comment more for our own reference rather than expecting you to do it, but you've confirmed my suspicion.
Mesa/v3d appears to be holding on to buffers for longer than it should. 
```
Attached Devices:
        fec00000.v3d
```
Probably quickest pinging Igalia over this one as I have no idea as to how the interaction between DRM and V3D is managed. It may be an X issue instead of Mesa itself.
@naushir You normally look after that interaction - could you pass this one on?

txenoo, 2020-03-11:
I've wrote a script to simplify the issue :

The script can be run with the following sintax:

./tests.sh [normal|left|right|inverted] [normal|left|right|inverted] 

```
#!/bin/bash
export DISPLAY=:0
sudo systemctl restart lightdm
sleep 5
uname -a
glxinfo
dpkg -l|grep mesa
dmesg > last-dmesg
sudo cat /sys/kernel/debug/dma_buf/bufinfo
XRAND_FAIL=0
DEFAULT_FLIP="$1"
OTHER_FLIP="$2"
FLIP=$DEFAULT_FLIP
i=0
while [ $XRAND_FAIL = 0 ] && [ $i -lt 20 ]
do
echo "================== BEGIN ITERATION $i - FLIP: $FLIP ================================"
/usr/bin/xrandr -o $FLIP
XRAND_FAIL=$?
sudo cat /sys/kernel/debug/dma_buf/bufinfo
dmesg > new-dmesg
diff last-dmesg new-dmesg
mv new-dmesg last-dmesg
echo "================== END ITERATION $i - FLIP: $FLIP================================"
if [ "$FLIP" = "$DEFAULT_FLIP" ]
then
	FLIP="$OTHER_FLIP"
else
	FLIP="$DEFAULT_FLIP"
fi
((i=i+1))
done
sudo systemctl restart lightdm
sleep 5
sudo cat /sys/kernel/debug/dma_buf/bufinfo
dmesg > last-dmesg

```
This has been happening since first buster raspbian release (2019-07-10) so we can reproduce with that version and and with current mesa master and last raspbian.

This is happening with 90 and 270 degrees transitions on xrand, so with a RPI4 with 2GB so it starts failing with the raspbian image with the iteration starting on 0 when:

- left or right <-> inverted: fails on iteration 6 
- inverted <-> left or right : fails on iteration 7
- left or right <-> normal: fails on iteration 10
- normal <-> left or right: fails on iteration 11
- normal <-> inverted: NO fails.
- left <-> right: NO fails

I'm attaching the output for the left-inverted transitions se we can check what is happening with CMA every iteration [out-left-inverted.txt](https://github.com/raspberrypi/firmware/files/4319697/out-left-inverted.txt)

6by9, 2020-03-11:
You've probably realised this already, but it looks to me like v3d isn't releasing the DRM buffer that has been passed to it for rendering the transposed frame.

X composes the desktop to a single framebuffer in the native orientation, and then passes it through GL again to do the transpose/rotate. The size of 8294400 bytes would be spot on for 1920x1080@ 32bpp (or transposed).
Most other systems wouldn't necessarily need to pass the buffer from DRM to GL/Mesa as a dmabuf as they are part of the same subsystem, but Pi4 is odd with V3D and VC4 being independent.

I have tried this on Pi3 with full KMS and that doesn't leak buffers.

This may be an X issue rather than Mesa or V3D specifically, but I don't know enough to say.

```v3d_prime_import_sg_table``` has imported the dmabuf from DRM and attached to it with the help of part of the DRM core framework.
It looks like ```drm_gem_shmem_free_object``` is going to be the release operation, so I guess a ```v3d_free_object``` isn't being triggered at an appropriate time.

txenoo, 2020-03-30:
Last days I've been debugging this issue.

And I think that the issue is related with the xserver at this [line 3208 of drmmode_display.c](https://gitlab.freedesktop.org/xorg/xserver/-/blob/server-1.20-branch/hw/xfree86/drivers/modesetting/drmmode_display.c#L3208)

The issue I found is that the number of calls to drmmode_bo_destroy and drmmode_create_bo don't match. 

When we transition from `xrandr-o left`to `xrandr -o normal` the old_front that is the previous `drmmode->front_bo` is not freed because the old_fb_id is 0 at this point and the referenced memory at old_front is never released.

If I force the drmmode_bo_destroy(drmmode, &old_front) if old_front.gbm is not null, we don't leak the memory. But the good solution would be understanding why we don't have a correct id of the old_fb_id. I will try to find what is happening here with the fd, by I guess that is related on that we free the bo for the rotation and the bo of the resize, and there is some point where the fd is reset to 0.

txenoo, 2020-05-15:
I've already submitted a MR to the xserver project. At the end, I think that we can safely do the drmmode_bo_destroy unconditionally.

Here it is the MR with the fix for the memory leak:
https://gitlab.freedesktop.org/xorg/xserver/-/merge_requests/444

ganzgustav22, 2020-05-26:
Thanks a lot for your efforts guys.

txenoo, 2020-06-29:
Fix has already landed in xorg-xserver repository

https://gitlab.freedesktop.org/xorg/xserver/-/commit/73480f172aeced074dd9301ae4d97f7d2f3a9a45

ganzgustav22, 2021-02-04:
Just did a test with the new Raspberry Pi OS 2021-01-11 Release: Rotated the screen about 4000 times (normal, right, inverted, left, normal, right, inverted, left, and so on), no issues anymore. 

Thanks a lot guys.

popcornmix, 2021-02-04:
@ganzgustav22 feel free to close the issue if you are happy it's resolved.

closed 2021-02-05 by ganzgustav22