raspberrypi/firmware #1346: PI3: Camera displays vertical lines when it gets video image from OpenCV library

roxanaluca, 2020-03-10:
I am using VideoCapture library to capture the images from Raspberry Pi camera V.2 8mp. 
**Describe the bug**
Have a Raspberry Pi updated to the newest version. Have a python virtualenv with OpenCV library installed. 

**To reproduce**
Have a python program that has following lines:
```
import time
import cv2

vc = cv2.VideoCapture(0)
vc.set(cv2.CAP_PROP_FRAME_HEIGHT,300)
vc.set(cv2.CAP_PROP_FRAME_WIDTH,300)
vc.set(cv2.CAP_PROP_FPS, 64)

cv2.namedWindow("eyecameras")

while vc.isOpened():
    key = cv2.waitKey(40)
    if key == 27: # exit on ESC
      break
    rval, frame = vc.read()
    time.sleep(0.1)
    cv2.imshow("eyecameras", frame)
    
cv2.destroyWindow("eyecameras")
vc.release()
```

**Expected behaviour**
The camera should show a normal image.

**Actual behaviour**
The camera show horizontal lines that disturbs the clarity of image.

**System**

* Which model of Raspberry Pi? Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)? 
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage5
* Which firmware version (`vcgencmd version`)? Feb 12 2020 12:39:27 
Copyright (c) 2012 Broadcom
version 53a54c770c493957d99bf49762dfabc4eee00e45 (clean) (release) (start_x)
* Which kernel version (`uname -a`)? 
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.
[image]

6by9, 2020-03-10:
Complain to the OpenCV developers.

V4L2 supports a value called bytesperline in the format structure, used to define how many bytes are used per line, and it can include padding. 
The Pi aligns bytesperline to a multiple of 32 bytes. Your width of 300 pixels is not a multiple of 32, therefore bytesperline will have been rounded up to 320. OpenCV should be abiding by that value, but isn't, hence the diagonal shift.
The workaround is to choose resolutions where the width is a multiple of 32.

roxanaluca, 2020-04-27:
Thank you a lot. It works now.

closed 2020-04-27 by roxanaluca