raspberrypi/firmware #1347: Disable USB device boot on Compute Module

RICCIARDI-Adrien, 2020-03-13:
I would like to disable Raspberry Pi Compute Module 3+ USB device boot, so Raspberry Pi can only boot from eMMC. The goal is to make eMMC content retrieval harder.
Is there a way to clear bootmode register bit 28 from OTP using vcmailbox or another interface?

pelwell, 2020-03-13:
The answer to your question is no - OTP stands for One-Time-Programmable, and although bits can be set to one individually at any time they can never be cleared back to zero.

RICCIARDI-Adrien, 2020-03-13:
Thank you for your quick reply.

It would be great for industrial uses to disable USB device boot, because an attacker would need to unsolder eMMC to retrieve its content. This should discourage low-skilled attackers for low-secure applications.
There might be an unused reserved bit in the boot mode register that could be set to turn off USB device boot, maybe using a config.txt key like `program_usb_boot_mode=1`?

pelwell, 2020-03-13:
That could only ever be possible if it was a feature that is supported by the ROM, and it isn't.

RICCIARDI-Adrien, 2020-03-13:
Maybe future Compute Module versions will support such a feature.
Thank you for your time.

closed 2020-03-13 by RICCIARDI-Adrien

JamesH65, 2020-03-13:
I believe this is a chip level thing, not a PCB level thing, so it would require a modification to the SoC itself. That seems unlikely, but never say never.

RICCIARDI-Adrien, 2020-03-13:
Yes, for instance a B0 revision of the BCM2711, if a Compute Module is made with this SoC.
Don't answer if it's confidential : is the ROMed bootloader a stock Broadcom one or Raspberry Pi Foundation can customize some aspects before mass production?

pelwell, 2020-03-13:
The BCM2711 ROM and OTP are configured to only try SD card and SPI EEPROM - everything else is left to the bootloader in the EEPROM.

timg236, 2020-03-13:
You might be able to use program_gpio_bootmode, and disable USB host boot by ensuring that the GPIO is zero. Of course, you'd have to make it difficult to remote the CM board from the device

Sets GPIO bootmode. 1 to select 22-26 and 2 to select 39-43.

https://www.raspberrypi.org/documentation/hardware/raspberrypi/otpbits.md

RICCIARDI-Adrien, 2020-03-13:
@timg236 Thank you for the idea, I will look for it.
@pelwell and @JamesH65 Thank you for taking the time to answer my questions.