raspberrypi/firmware #1348: [Question] What is the reason for a leading "2" in revision code?

MichaIng, 2020-03-13:
We found a case where the revision code of an RPi3 B+ appears to be `2a020d3` (in `/proc/cpuinfo`). According to [docs](https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md) the revision code should be `a020d3` without the leading `2`. I know that overvolted devices show a leading `1000` in revision code, but what is the reason for a `2`?

And are there any other cases where the revision code gets some addition, if so, could someone list them? Probably I'll add a PR to has this info on the docs page as well. If one wants to parse the code via scripts to get certain hardware info, this is useful to know.

MilhouseVH, 2020-03-13:
Official documents say those bits are [unused](https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md) but 2 bits appear to be used to represent a [void warranty](https://github.com/AndrewFromMelbourne/raspberry_pi_revision) so values 0, 1, 2 and 3 are possible.

JamesH65, 2020-03-13:
Is it covered already by this PR?

https://github.com/raspberrypi/documentation/pull/1370/files

MichaIng, 2020-03-13:
I anyway try to understand this hex numbers:
`uuuuuuuu` means 8 bits? And since 2^8 == 16^2, those 8 bits can be 2 hexadezimal chars? But why was the prior overvolted flag `1000` then, which would require 16 bits. And now I see according to your link that for old revisions had `100` at the front, not `1000`

Since `F` and `MMM` are together 4 bits, they are a single hexadezimal char? Since `F` is always one on these new revision codes, it is 8 up to f, based on memory size?

And now I begin to understand: https://github.com/AndrewFromMelbourne/raspberry_pi_revision#raspberry-pi-2--model-zero
- With old revision codes the warranty bit was 24, hence the 7th hexadezimal char, so indeed leading `100` in front of the 4 base chars.
- With new revision codes it moved to bit 25, hence it is now a leading `2` at 7th char.

MichaIng, 2020-03-13:
@JamesH65 
Ah warranty vs overvolted and some more. So `1000` is overvolted only, `100` is warranty void only, in the old revision codes probably?

And in the new codes:
```
| uuu      | Unused       | Unused                     |
| W        | Warranty bit | 0: Warranty is intact      |
|          |              | 1: Warranty has been voided by [overclocking](../../../configuration/config-txt/overclocking.md) |
| u        | Unused       | Unused                     |
```
- So the 7th hex char can be non-existent or `0` (if 8th char is present), meaning warranty is intact or it can be `2`, meaning warranty void.
```
| N        | Overvoltage  | 0: Overvoltage enabled     |
|          |              | 1: Overvoltage disabled    |
| O        | OTP Program<sup>1</sup> | 0: OTP programming enabled |
|          |              | 1: OTP programming disabled |
| Q        | OTP Read<sup>1</sup> | 0: OTP reading enabled |
|          |              | 1: OTP reading disabled    |
| uuu      | Unused       | Unused                     |
```
- So there is indeed an 8th char possible which can be 2, 4, 6, 8, a, c, e based on OTP + overvoltage state combo. But shouldn't those bits be `0` if `disabled` and `1` if enabled?

JamesH65, 2020-03-14:
Ignore the old revision codes, they have not been used for years - not sure exactly when the new ones took over but pretty sure it was before or at the time the 3b came out.

MichaIng, 2020-03-14:
@JamesH65 
> Ignore the old revision codes

But they still show up on those old devices, regardless of kernel/bootloader/firmware packages, don't they?

From RPi 2 release on, the new revision codes have been used.

JamesH65, 2020-03-14:
Yes, they do, but not many of them about compared with the toal sales. Or at least, still in use.

MichaIng, 2020-03-14:
@JamesH65 
That is true, however we need things to work reliable on all RPi revisions, but thanks to your PR it is now clear. I mark this issue hence as closed.

closed 2020-03-14 by MichaIng

pelwell, 2020-03-16:
Sorry for the delay - I wanted to get a few facts straight before replying.

The combination of the version of 2837 used and the power supply changes allowed us spec the ARMs in 3B+ to run at 1400MHz - previously 1200MHz was the top end. We stress test at a higher clock than that to give us a confidence margin. A small number of devices fail that stress test - not enough to be a problem, but enough to start to build up when you have production volumes like ours.

We sell a lot into industrial applications where CPU performance is not really an issue, so once the pile grew large enough a second stress test was introduced at a lower speed - devices that failed the first test but passed the second were sold to informed customers at a reduce price. This kind of speed binning and pricing is common practise for Intel, AMD etc.

In order to reduce the upper operating point automatically, and as a way of quickly identifying these devices should any problems be encountered, bit 1 of the user bits (bit 25 in the board revision - 0x02000000) is set in the OTP - it's known as the stress-fail bit, but it's more of a reduced-stress-pass bit). If you remove any overclocking from config.txt but then find a way to keep the ARMs at the higher operating point - `force_turbo=1`, running `while true; do true; done &`, or engaging the performance governor - you will find that on those boards `vcgencmd measure_clock arm` will report 1200 instead of the usual 1400.

I'm curious to find out how that device crossed your path - email me (phil@raspberrypi.com) if you don't feel happy discussing it in public.

JamesH65, 2020-03-16:
This seems like something that needs an update to the docs, as it will surely confuse people (I didn't know about it, so was confused)

pelwell, 2020-03-16:
Also, if you believe the Pi wasn't bought in the full knowledge that its maximum performance would be limited then email me and we will replace it for you.

MichaIng, 2020-03-16:
@pelwell 
Many thanks for the details. Indeed this makes sense and is indeed common practice when you have large production volumes where a small weak share is still a reasonable amount.

But now I am confused:
> bit 1 of the user bits (bit 25 in the board revision - 0x02000000) is set in the OTP

I though that bit 25 is the "new" warranty bit? Just to make it clear, since I have no experience in reading/converting bit to hexadecimal: bit 25 bit means the second bit of the 7th hexadecimal number, hence changes it from 0 to 2 (as long as the previous bit is `0`)? Then https://github.com/AndrewFromMelbourne/raspberry_pi_revision#raspberry-pi-2--model-zero and the PR would be wrong?

pelwell, 2020-03-16:
[ Re-reads the relevant header file. Notices that the 2 is the userbits bit number not the value. Feels stupid. ]

OK, scratch that. It is just the new overvoltage flag, as used on 2836 and above, as you say. Bit 26 (0x04000000) would be a stress test failure - it can get added to the PR as the `S` bit, and perhaps bit 24 should be marked as `w` - the original over-clocking bit.

I'll get my coat.

MichaIng, 2020-03-16:
@pelwell 
Okay that makes sense, also to add the old (pre-Pi2) overvoltage/warranty flag in bit 24 @JamesH65.
Okay so if one would face `4a020d3` or `6a020d3` (if even possible with the stress test failed device?) THEN it would be this special stress test failed device, right?

pelwell, 2020-03-16:
> So if one would face 4a020d3 or 6a020d3 (if even possible with the stress test failed device?) THEN it would be this special stress test failed device, right?

Yes it would, but regular users should NEVER encounter one of these in the wild.

MichaIng, 2020-03-16:
@pelwell 
Still would be useful in the docs, so you can know that you were swindled and contact the vendor, otherwise just for completeness 😉.