raspberrypi/firmware #1350: Firmware append video cmdline parameter with invalid syntax

lategoodbye, 2020-03-15:
**Describe the bug**
Using firmware from 2020-02-12 i noticed that the video commandline parameter passed by the firmware is using its own syntax, which is incompatible to the kernel.

`[    0.000000] Kernel command line: video=HDMI-A-1:1920x1200M@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0 dma.dmachans=0x7f35 bcm2709.boardrev=0xa02082 bcm2709.serial=0x53a68e44 bcm2709.uart_clock=48000000 smsc95xx.macaddr=B8:27:EB:A6:8E:44 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS1,115200 console=tty1 root=PARTUUID=ea7d04d6-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles`

Until Linux 5.2 i was getting the following error:
`[   10.869250] [drm] parse error at position 69 in video mode '1920x1200M@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0'`

The official syntax supported by DRM is described [here](https://www.kernel.org/doc/Documentation/fb/modedb.txt).

**To reproduce**
- connect a Raspberry Pi 3 with a HP ZR2440w
- place recent Raspbian on SD card
- replace kernel with Mainline kernel e.g. 5.2

**Expected behaviour**
- firmware generate video cmdline parameter according to the kernel documentation

**Actual behaviour**
- firmware generate video cmdline parameter according its own syntax

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? 3B
* Which OS and version (`cat /etc/rpi-issue`)? Raspbian 2020-02-13
* Which firmware version (`vcgencmd version`)? 2020-02-12
* Which kernel version (`uname -a`)? Mainline 5.2 using multi_v7_defconfig

**Logs**
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.2.0 (stefan@Latitude-E4310) (gcc version 7.2.1 20171011 (Linaro GCC 7.2-2017.11)) #9 SMP Sat Mar 14 12:15:34 CET 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 64 MiB at 0x2a800000
[    0.000000] On node 0 totalpages: 229376
[    0.000000]   DMA zone: 1536 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 32768 pages, LIFO batch:7
[    0.000000] percpu: Embedded 20 pages/cpu s52684 r8192 d21044 u81920
[    0.000000] pcpu-alloc: s52684 r8192 d21044 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 227840
[    0.000000] Kernel command line: video=HDMI-A-1:1920x1200M@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0 dma.dmachans=0x7f35 bcm2709.boardrev=0xa02082 bcm2709.serial=0x53a68e44 bcm2709.uart_clock=48000000 smsc95xx.macaddr=B8:27:EB:A6:8E:44 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS1,115200 console=tty1 root=PARTUUID=ea7d04d6-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 820956K/917504K available (12288K kernel code, 1638K rwdata, 5128K rodata, 2048K init, 401K bss, 31012K reserved, 65536K cma-reserved, 131072K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=16 to nr_cpu_ids=4.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2fc/0x48c with crng_init=0
[    0.000008] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000026] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000079] bcm2835: system timer (irq = 27)
[    0.002463] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.002481] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.002497] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.002510] Switching to timer-based delay loop, resolution 52ns
[    0.002942] Console: colour dummy device 80x30
[... 233 lines cut ...]
[   10.985993] [drm] Driver supports precise vblank timestamp query.
[   10.987459] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[   11.070065] Console: switching to colour frame buffer device 240x67
[   11.103477] vc4-drm soc:gpu: fb0: vc4drmfb frame buffer device
[   11.292463] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   11.385167] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   11.410866] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.txt failed with error -2
[   11.587410] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   11.587500] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   11.588112] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
[   12.634900] Bluetooth: hci0: command 0xfc18 tx timeout
[   19.311520] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SS
[   20.098501] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   20.954875] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   20.954885] Bluetooth: hci0: Failed to set baudrate
[   21.079107] Bluetooth: hci0: BCM: chip id 94
[   21.081266] Bluetooth: hci0: BCM: features 0x2e
[   21.104277] Bluetooth: hci0: BCM43430A1
[   21.104299] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0000
[   24.885235] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0360
```

**Additional context**
This issue has been discovered during investigation of:
http://lists.infradead.org/pipermail/linux-arm-kernel/2020-March/715809.html

6by9, 2020-03-15:
If you don't want the firmware command line mode, then you can add ```disable_fw_kms_setup=1``` to /boot/config.txt.

DRM accepts additional configuration options over and above those listed in modedb.txt. They are correctly documented in https://github.com/torvalds/linux/blob/master/Documentation/fb/modedb.rst.

Those include ```margin_[top|bottom|left|right]``` which were added with https://github.com/torvalds/linux/commit/3d46a3007cd8f73bae502bf5c171977b91d7aacc, ```rotate``` ```reflect_x``` and ```reflect_y``` added with https://github.com/torvalds/linux/commit/1bf4e09227c345e246062285eba4b8fe660e512e.

4.19 has all the relevant commits backported. I haven't checked the other branches.

6by9, 2020-03-15:
Updated docs also available at https://www.kernel.org/doc/Documentation/fb/modedb.rst - it looks like just the modesdb.txt file that is out of date. It hasn't been present in the kernel tree since
```
commit ab42b818954c040fa13639dc031d8541edcecb4b
Author: Mauro Carvalho Chehab <mchehab+samsung@kernel.org>
Date:   Wed Jun 12 14:52:45 2019 -0300

    docs: fb: convert docs to ReST and rename to *.rst
```

lategoodbye, 2020-03-15:
@6by9 Thank you for your fast feedback, after opening the bug i noticed `disable_fw_kms_setup=1`.
The problem is in Mainline these new parameter has been implemented in DRM since 5.3, which hided a different issue with the pixel clock limitation.

Another option could be an additional config.txt option which keep the video parameter backward compatible.

6by9, 2020-03-15:
It's a damned if you do, damned if you don't situation.
Having switched to using DRM/KMS on Pi4 we had complaints from people who had TVs that overscanned. Implement that by back porting the relevant patches from mainline (even if that does mean it scales instead of reducing the FB size) and you're complaining.
Sorry, those running older mainline kernels are in the minority vs those running latest or our kernel.

We could drop the margin properties should ```disable_overscan``` be set, but that's still going to need the user to set that option in the first place as it defaults to 1.

1920x1200 is an issue with the KMS driver as the pixel clock exceeds the defined max based on the HDMI state machine clock (```vc4_hdmi_encoder_mode_valid```, works out at 151.5MHz). I'd need to see what the firmware sets the HSM clock to for 1920x1200, as whilst it is technically beyond the spec, it generally works fine.
With Maxime's reworking of the driver it becomes an explicit pixel frequency limit rather than a computed number, and currently it's set at 148.5MHz for Pi0-3.

lategoodbye, 2020-03-15:
> 1920x1200 is an issue with the KMS driver as the pixel clock exceeds the defined max based on the HDMI state machine clock (`vc4_hdmi_encoder_mode_valid`, works out at 151.5MHz). I'd need to see [...]

Yes, all those (non-working) fallback mechanism made it hard to narrow my original issue. At first i need to find the major version and then bisected between 5.2 and 5.3. No fun at all.

The Firmware KMS driver operates with 154 MHz pixel clock (according to /sys/kernel/debug/clk/clk_summary) in my case (tested with rpi-4.19).

There are multiple issues which comes to my mind as non-VC4-expert:
- why does the firmware suggest a mode (1920x1200) which is actually out of spec
- why doesn't KMS driver fallback to 1920x1080 in case the provided video commandline parameter isn't "valid"

6by9, 2020-03-15:
Presumably that's running in DMT mode 68 then - https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/gpu/drm/drm_edid.c#L520. 1080p60 is 148.5MHz, so really is only just higher as it has reduced blanking.
The firmware has hdmi_pixel_freq_limit as an option, and that defaults to 162MHz on Pi0-3 (300MHz on Pi4). The firmware is therefore happy with 1920x1200@60.

It's the HDMI state machine clock that I'm interested in based on Eric having hard coded it to 163682864Hz in https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/gpu/drm/vc4/vc4_hdmi.c#L64, and then comments about it at https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/gpu/drm/vc4/vc4_hdmi.c#L723 being the limiting factor on pixel rate.
Either the fixed HSM clock he is using is too low and should be variable, or the comments are based on something that isn't true.

KMS fallback is rather bizarre/haphazard.
It takes the command line mode as gospel, and adds it as an additional mode should it not exist in the EDID or panel definition. In adding the mode it will be tested by the driver as to whether it is within supported parameters, but AFAIK it doesn't remove the request to make that the default mode should the driver report it invalid. That's what then leaves things in the odd situation.
NB This is the core KMS code, not the driver that doesn't behave in a wonderful manner.

lategoodbye, 2020-03-17:
Since there is currently nothing we can do from firmware side. Close this

closed 2020-03-17 by lategoodbye

6by9, 2020-03-17:
I will drop the margins if they are all 0 as it juts adds clutter.

I'm looking at the DRM code to see if the cmdline mode selection can be made more sane should it be rejected by the driver. That'll obviously only apply to later kernels though.