raspberrypi/firmware #1351: 256MiB SDcard in pizeroW does not boot bootcode.bin

BETLOG, 2020-03-16:
**Describe the bug**
Cannot boot via bootcode.bin on 256MiB sdcard to USB ssd

**To reproduce**
Copy bootcode.bin to freshly FAT16 or FAT32 formatted 256MiB SDcard
Plug in USB3 SSD drive
Attempt to boot
Failure.
Without plugging in HDMI I can't say for sure what the error is, but given the total absence of flashing LED I have to assume the firmware does a check for SDcard size prior to attempting boot.
I have tried two of the ten 256Mib SDcards I just bought, and neither work with FAT32 or FAT16, yet a 32GiB card works fine.

**Expected behaviour**
Booting via sdcard to OS on USB ssd should work regardeless of SDcard size.
Using a 32GiB card to boot a 51KiB bootloader is undesirable.

**System**

```
pi@pizero:~ $ sudo raspinfo
sudo: raspinfo: command not found
```
PiZeroW
pi@pizero:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2
pi@pizero:~ $ vcgencmd version
Feb 12 2020 12:38:08 
Copyright (c) 2012 Broadcom
version 53a54c770c493957d99bf49762dfabc4eee00e45 (clean) (release) (start)
pi@pizero:~ $ uname -a
Linux pizero 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux

pelwell, 2020-03-16:
I've not heard of (and I can't think of a reason for) any size check in the ROM or bootloader. It seems more likely that some aspect of the formatting is incompatible with the loaders.

Since it sounds like you have a reasonable supply of cards of various kinds, can you try using a tool like WinDiskImager (or dd, if you're a CLI type) to make a .img file of a programmed but not-working 256MB card. You can then write that image to a spare 32GB card, and also upload it to DropBox, Google Drive etc. so that we can examine it.

BETLOG, 2020-03-16:
I'm a linux user.
So... you want me to format a fat32 sdcard, copy bootcode.bin to it, and then make 256MiB and 32GiB images, and upload them?
Really? That sounds not only somewhat crazy, but also assumes I can even just casually upload a 32GiB file..... I can, but I'd really want to be sure that I understand and that you aren't asking me to do something ludicrous first.

pelwell, 2020-03-16:
> So... you want me to format a fat32 sdcard, copy bootcode.bin to it, and then make 256MiB and 32GiB images, and upload them?

No, that's not what I asked:
1. Format 256MB card.
2. Read the raw sectors into a .img file.
3. Write the .img to a 32GB card (or any larger card that is known to work) and test it.
4. Upload the .img somewhere.

If you don't have a spare card for step 3, just upload the .img.

timg236, 2020-03-16:
If you have a USB UART and the GPIO pins available you could try this to see if the ROM is able to run bootcode.bin
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/README.md

256MiB FAT partitions are pretty common, it's the default now on Raspbian. Unfortunately, it's possible that a  256MiB is an old SDv1 card and isn't supported by the 2835 ROM.

However, as @pelwell says it would be useful to see the FAT image in-case there's something odd there.

Details of the card would be useful

BETLOG, 2020-03-16:
> No, that's not what I asked:

ok, just the 256MiB version.. because: see at bottom.

https://www.dropbox.com/s/ekepal65ban8x2p/256MiB_sdcard.img?dl=0
https://www.dropbox.com/s/745s4kmb7n2xrr3/bootcode.bin_256MiB_sdcard.img?dl=0

Formatted fat32 with partitionmanager 3.3.1
Linux betlognuc 4.15.0-88-generic #88-Ubuntu SMP Tue Feb 11 20:11:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

downloaded from:
https://raw.githubusercontent.com/raspberrypi/firmware/master/boot/bootcode.bin
drag&drop copied with Dolphin
imaged with:

```
sudo dd if=/dev/sdg of=/tmp/bootcode.bin_256MiB_sdcard.img
[sudo] password for user: 
493568+0 records in
493568+0 records out
252706816 bytes (253 MB, 241 MiB) copied, 16.7503 s, 15.1 MB/s
user@betlognuc:~$ sudo dd if=/dev/sdg of=/tmp/256MiB_sdcard.img
493568+0 records in
493568+0 records out
252706816 bytes (253 MB, 241 MiB) copied, 16.5664 s, 15.3 MB/s
```

> SDv1 card and isn't supported by the 2835 ROM.

This is almost certainly the problem. They seem to work fine in all other respects.
https://www.aliexpress.com/item/33019390820.html?spm=a2g0s.9042311.0.0.714b4c4dPBpcdK

BETLOG, 2020-03-16:
Based on your 'SDv1' above it looks like SDHC | SDXC | SDUC are good, and SD (formatted FAT16) is not.
..... Now I'm pondering how can I reliably identify which cards to NOT buy on aliexpress (australias most convenient and cheapest option)

https://en.wikipedia.org/wiki/SD_card#Comparison
https://www.sdcard.org/developers/overview/speed_class/
https://superuser.com/questions/381490/how-to-tell-the-sd-card-version

timg236, 2020-03-17:
I'm not saying that the ROM explicitly does not support SDv1 but older standard capacity SD cards tend to be more awkward. SD-HC is very well tested and the cards tend to be much better.

FAT16 is supported, you can try creating a small FAT16 partition on a larger HC card to verify FAT support.

BETLOG, 2020-03-17:
I will test that eventually, but it's essentially moot. i'm only using small cards - that don't seem to exist in SDHC types - because they are cheap and only need to hold 52K. But reading 52k for a reboot or two every few months shouldn't tax a new $5 SDHC too heavily, so I guess I can afford to buy one or two  :)

closed 2020-03-17 by BETLOG